esphome/firmware #45: ESP32-S3-Box does not compile

BeardedTek, 2023-09-19:
I get the following log output:
```
INFO ESPHome 2023.10.0-dev
INFO Reading configuration /config/esphome/esp32-s3-box-assistant.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/esphome/esphome.git@pull/5229/head
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO45 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 403, in command_compile
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 190, in write_cpp
    return write_cpp_file()
  File "/esphome/esphome/__main__.py", line 208, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 342, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 295, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 592, in copy_files
    repo_dir, _ = git.clone_or_update(
  File "/esphome/esphome/git.py", line 114, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 32, in run_git_command
    raise cv.Invalid(err_str)
voluptuous.error.Invalid: fatal: Needed a single revision
Unable to find current revision in submodule path 'components/esp-sr'
```

BeardedTek, 2023-09-19:
It appears that it's trying to pull the v2.5 branch from https://github.com/espressif/esp-adf which no longer exists.

jesserockz, 2023-09-21:
v2.5 is a tag, not a branch: https://github.com/espressif/esp-adf/tree/v2.5

Someone else had seen this too and it seemed to fix itself automatically. Could be a network glitch cloning the repo.

sparkydave1981, 2023-10-16:
I had this same problem, tried again 2 days later and all was good.

genebean, 2023-11-07:
Still an issue for me in https://github.com/esphome/issues/issues/4952

jaymunro, 2024-01-05:
Have you tried cleaning your build files?

BjornPython, 2024-10-12:
> Have you tried cleaning your build files?

also having this same exact error, tried this, did not resolve it.

Feighery, 2025-01-08:
> Have you tried cleaning your build files?

This was the issue for me, was going around in circles with this issue for a bit.

closed 2025-02-10 by jesserockz