esphome/firmware #53: m5 echo bug: can't unpause, unmute. unresponsive.

elektrinis, 2023-10-05:
Using as HA integration via ESPhome. It mostly works, however some features are broken:
1. Power button does nothing
2. If muted, can't unmute. Have to restart the board.
3. If paused, can't unpause. Renders unusable
4. Starts ar full volume, even resets to full volume after unsuccessful stream.
5. Audio quality is really low, seems like 8bit, low frequencies are filtered out and does not sound good even with high end, sensitive speaker.
6. Klicks/pops when starting stream. Needs some fade-in and fade-out. 
7. Buffering issues, intermittent audio streaming (using HA radio browser).

I was hoping I have finally found a wifi amplifier that works with HA. Nice progress, but much work remaining.

KrOnAsK, 2023-12-20:
Anything new here? Also I cant use that installer on the website you sent because when I try to connect the M5 to wifi after it always fails even though my SSID and keyword are correct, could you export your yaml and post it here?

Nardol, 2023-12-20:
I can confirm at least one point:
> If muted, can't unmute. Have to restart the board.

Which is really annoying.

elektrinis, 2023-12-20:
Still not fixed. Using it as daily driver for radio listening, have to set volume to 0 to switch it off, otherwise have to disassemble my ceiling to reset the device.
Still bad sound quality at very low volumes.

closed 2025-05-14 by github-actions[bot]