esphome/firmware #60: Cannot play HA media on ESP Muse Luxe

rp3tya, 2023-10-13:
The device can play online radio streams but reboots when trying to play media from Home Assistant. This used to work with previous versions (2022.x) of ESPHome.

Config: https://github.com/esphome/firmware/blob/main/media-player/raspiaudio-muse-luxe.yaml
ESPHome: 2023.9.3
HA: 2023.10.2

Similar issue reported in the old repo: https://github.com/esphome/media-players/issues/29

```
...
[14:35:53][C][logger:357]: Logger:
[14:35:53][C][logger:358]:   Level: DEBUG
[14:35:53][C][logger:359]:   Log Baud Rate: 115200
[14:35:53][C][logger:361]:   Hardware UART: UART0
[14:35:53][C][i2c.arduino:053]: I2C Bus:
[14:35:53][C][i2c.arduino:054]:   SDA Pin: GPIO18
[14:35:53][C][i2c.arduino:055]:   SCL Pin: GPIO23
[14:35:53][C][i2c.arduino:056]:   Frequency: 50000 Hz
[14:35:53][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[14:35:53][I][i2c.arduino:069]: Results from i2c bus scan:
[14:35:53][I][i2c.arduino:075]: Found i2c device at address 0x10
[14:35:53][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Volume Up'
[14:35:53][C][gpio.binary_sensor:016]:   Pin: GPIO19
[14:35:54][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Volume Down'
[14:35:54][C][gpio.binary_sensor:016]:   Pin: GPIO32
[14:35:54][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Action'
[14:35:54][C][gpio.binary_sensor:016]:   Pin: GPIO12
[14:35:54][C][fastled:021]: FastLED light:
[14:35:54][C][fastled:022]:   Num LEDs: 1
[14:35:54][C][fastled:023]:   Max refresh rate: 400
[14:35:54][C][light:103]: Light 'Muse1'
[14:35:54][C][light:105]:   Default Transition Length: 1.0s
[14:35:54][C][light:106]:   Gamma Correct: 2.80
[14:35:54][C][adc:097]: ADC Sensor 'Battery'
[14:35:54][C][adc:097]:   Device Class: 'voltage'
[14:35:54][C][adc:097]:   State Class: 'measurement'
[14:35:54][C][adc:097]:   Unit of Measurement: 'V'
[14:35:54][C][adc:097]:   Accuracy Decimals: 3
[14:35:54][C][adc:097]:   Icon: 'mdi:battery-outline'
[14:35:54][C][adc:107]:   Pin: GPIO33
[14:35:54][C][adc:122]:  Attenuation: 11db
[14:35:54][C][adc:142]:   Update Interval: 15.0s
[14:35:54][C][captive_portal:088]: Captive Portal:
[14:35:54][C][mdns:115]: mDNS:
[14:35:54][C][mdns:116]:   Hostname: muse1
[14:35:54][C][ota:097]: Over-The-Air Updates:
[14:35:54][C][ota:098]:   Address: muse1.local:3232
[14:35:54][C][api:138]: API Server:
[14:35:54][C][api:139]:   Address: muse1.local:6053
[14:35:54][C][api:143]:   Using noise encryption: NO
[14:35:54][C][improv_serial:032]: Improv Serial:
[14:35:54][C][audio:203]: Audio:
[14:35:54][C][audio:225]:   External DAC channels: 2
[14:35:54][C][audio:226]:   I2S DOUT Pin: 26
[14:35:54][C][audio:227]:   Mute Pin: GPIO21
[14:36:17][D][media_player:059]: 'Muse1' - Setting
[14:36:17][D][media_player:066]:   Media URL: http://homeassistant.local:8123/api/tts_proxy/7b502c3a1f48c8609ae212cdfb639dee39673f5e_en-us_be0728c770_tts.piper.wav
WARNING muse1.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for muse1.local
WARNING Disconnected from API

INFO Successfully connected to muse1.local
[14:36:47][D][sensor:094]: 'Battery': Sending state 4.22671 V with 3 decimals of accuracy
[14:36:53][D][api:102]: Accepted 192.168.x.x
[14:36:53][D][api.connection:1031]: Home Assistant 2023.10.2 (192.168.x.x): Connected successfully
[14:37:04][D][media_player:059]: 'Muse1' - Setting
[14:37:04][D][media_player:069]:   Volume: 0.21
[14:37:38][D][media_player:059]: 'Muse1' - Setting
[14:37:38][D][media_player:066]:   Media URL: http://icestreaming.rai.it/3.mp3
[14:37:40][W][component:204]: Component i2s_audio.media_player took a long time for an operation (1.09 s).
[14:37:40][W][component:205]: Components should block for at most 20-30ms.
...
```

DotBowder, 2023-10-19:
I just purchased 4x ESP Muse Luxe's off of Amazon and I've been trying to get them to work with Home Assistant. Two of them remain in the box so if I've somehow bricked these others I can test.

Steps I took:
1. Go to the ESPHOME addon in Home Assistant
2. Add a new device over USB Serial
3. Recognize and flash device
4. Wait for reboot and see device is online
5. Update config for the device to match the corresponding options (apart from naming) in the [linked config](https://github.com/esphome/firmware/blob/main/media-player/raspiaudio-muse-luxe.yaml).
6. Wait for reboot and see the device is online
7. Import the device into Home Assistant

Now, the device exists in Home Assistant. I can click the 3 buttons and see the sensors light up in HA. I can turn on and off the LED from HA. I can see the media player in HA.

When I attempt to send any sort of audio to the media player (radio, mp3, tts) the device crashes and reboots.

If you go into ESPHome, find your muse, select "logs", then select "Plug into this computer", you can see some more details about the crash. When it crashes I get this dump. `The following tasks did not reset the watchdog in time`

```
E (78046) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (78046) task_wdt:  - loopTask (CPU 1)
E (78046) task_wdt: Tasks currently running:
E (78046) task_wdt: CPU 0: IDLE
E (78046) task_wdt: CPU 1: IDLE
E (78046) task_wdt: Aborting.
abort() was called at PC 0x4010fcf8 on core 0

Backtrace:0x4008425d:0x3ffbea9c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```

At this point, I was convinced it was my Home Assistant setup. I did find that I was using an HTTP Internal URL instead of HTTPS, but that didn't solve the problem.

---

You are the first person I have seen to make any sort of mention of this problem online. I am glad that I am not alone.

---

Before using the ESP Muse Luxe, I tried to get some M5Stack Core2 devices to work with audio, but I experienced the same behavior. Immediate crashing upon sending any kind of media to the media player. I burned a decent bit of my free time last week on the M5Stack Core2, and figured I would wait until the Muse Luxe's arrived, but they have the same problem.

I am suspicious that there is a bug in one of the libraries used by esphome.

When I go to compile the firmware I get:
```
INFO ESPHome 2023.10.0
INFO Reading configuration /config/esphome/f1-muse-officed-1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/3552/head
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing f1-muse-officed-1 (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|-- Wire @ 2.0.0
|-- Improv @ 1.2.3
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- ESP32-audioI2S @ 2.0.7
|-- FastLED @ 3.3.2
```
---

HA Info:
```
Home Assistant 2023.10.3
Supervisor 2023.10.0
Operating System 11.0
Frontend 20231005.0 - latest
```

ESP Home Info:
```
2023.10.0
```

There is an update for 2023.10.1. I may try that tomorrow.

rp3tya, 2023-11-16:
My Muse is working again as expected after 2023.11.1 update (although I do not see any relevant fixes in the changelog).

closed 2023-11-18 by rp3tya