esphome/firmware #61: Voice Assistant fails during Installation

AseKarlsson, 2023-10-13:
Hi,
Tried to follow the guide from: https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/

I get stuck with this error during installation:
```
INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/m5stack-atom-echo-8a1714.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1014, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 986, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 840, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 828, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 690, in validate_config
    substitutions.do_substitution_pass(config, command_line_substitutions)
  File "/esphome/esphome/components/substitutions/__init__.py", line 143, in do_substitution_pass
    config.move_to_end(CONF_SUBSTITUTIONS, False)
AttributeError: 'dict' object has no attribute 'move_to_end'
```

Anyone with ideas?

t34wrj, 2023-10-13:
I'm having the same issue. According to the yaml that is linked to in the Home Assistant installation link - github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main - the minimum required esphome version is 2023.10.0b1 whereas the version you and I are running is 2023.9.3 - which appears to be the latest version available in the Home Assistant add-ons at the time of writing. I wonder if this is causing the issue.

AseKarlsson, 2023-10-14:
[image]

AseKarlsson, 2023-10-14:
Any release date for 2023.10.0 of ESPHome?

thundergreen, 2023-10-14:
do you use addon in homeassistant? then just use BETA image iso "normal"

AseKarlsson, 2023-10-14:
@thundergreen , try to avoid Betas. Burned myself so many times even installing the .0 version

thundergreen, 2023-10-14:
@AseKarlsson  I agree but this won't kill your hoe assistant instance.. just your esphome and you can always switch back to stabble

AseKarlsson, 2023-10-14:
Just strange to release this https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/ and its still needs a beta.
It also was up in "Year of the voice chapter 4" but no info on that this is still in beta.

AseKarlsson, 2023-10-18:
Today upgraded ESP Home to 2023.10.

Dont get the same erroe anymore, now I get the below when installing the software to the Echo:

[image]

thundergreen, 2023-10-18:
You have to allow this entity to send commande to Home assistant

AseKarlsson, 2023-10-18:
> You have to allow this entity to send commande to Home assistant

Please advice, what should I do?

jesserockz, 2023-10-19:
> You have to allow this entity to send commande to Home assistant

@thundergreen this is incorrect advice and is not required for the voice assistant.

jesserockz, 2023-10-19:
> Today upgraded ESP Home to 2023.10.
> 
> Dont get the same error anymore, now I get the below when installing the software to the Echo:
> 
> [image]
> 
> and when checking in the logs after installation I get:
> 
> [18:45:54][D][voice_assistant:366]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD [18:45:54][D][voice_assistant:176]: Waiting for speech... [18:45:54][D][voice_assistant:366]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD [18:45:54][D][voice_assistant:189]: VAD detected speech [18:45:54][D][voice_assistant:366]: State changed from WAITING_FOR_VAD to START_PIPELINE [18:45:54][D][voice_assistant:372]: Desired state set to STREAMING_MICROPHONE [18:45:54][D][voice_assistant:206]: Requesting start... [18:45:54][W][voice_assistant:217]: Could not request start.

@AseKarlsson What Home Assistant Version are you running

AseKarlsson, 2023-10-19:
HA Version 2023.10.3