esphome/firmware #66: Wake Word Atom Freeze

CAVACO-PT, 2023-10-16:
I have the $13 Atom, it works fine for wake word detection right after turning on wake word switch on their device page. However, after 1 or 2 commands they stop responding ...

if i toogle the switch and wait a bit it comes back , but after saying 2 or 3 commands next to each other it freezes again , and seams to come back after 5 or more minutes ... tried with HA cloud and local , same behavior !!

What are some troubleshooting steps I can try?

CAVACO-PT, 2023-10-16:
I have HA in a Proxmox VM , with superviser … using an ATOM , i just bought 2 units , both have the same issue , the firmware in the atom is

atom-echo-voice-assistant
por m5stack
Firmware: 2023.10.0b1 (Oct 13 2023, 23:14:59)
Hardware: 1.0

I have tried and without wake word … by pressing the button , and the ATOM doesnt have this problems , so its related to the wake word , without it i can press multiple times the button and it responds all the time !!!

arhills, 2023-10-16:
I have HAOS in Proxmox VM on an HPt630 16gb ram

Home Assistant 2023.10.3
Supervisor 2023.10.0
Operating System 11.0
Frontend 20231005.0 - latest

Using the M5stack atom echo I get similar issues, works 2 or 3 times then just stops responding to wake word. Sometimes switching wake word of and on again sorts it for another 2 or so times sometimes you just have to wait 10 minutes or so and it starts working again. When it responds it does what I ask 99% of the time.

andrew-codechimp, 2023-10-16:
See also this issue on the wyoming openwakeword project.
Unclear if it's an ESP device issue or openwakeword
https://github.com/rhasspy/wyoming-openwakeword/issues/2

Photo64, 2023-10-16:
Home Assistant 2023.10.3
Supervisor 2023.10.0
Operating System 11.0
Frontend 20231005.0 - latest

Same symptoms, M5stack atom echo freezes after 2 or 3 times and no longer responds

dattas, 2023-10-16:
Home Assistant 2023.10.3
 Supervisor 2023.10.0 
Operating System 11.0 
Frontend 20231005.0 - latest 

Same issues here using an M5 ATOM echo, I also tried Porcupine1 for good measure and it has the same issue. I can see in the addon logs for openwakeword/porcupine1 , Whisper, then Piper that the flow goes all the way through all of them even showing a success on Piper for the response but no output on the device:
```
INFO:wyoming_faster_whisper.handler: What time is it?
```
```
DEBUG:wyoming_piper.handler:input: {'text': 'It is currently a test message.'}
DEBUG:wyoming_piper.handler:/tmp/tmp32gzdqbs/1697498762860832160.wav
DEBUG:wyoming_piper.handler:Completed request
DEBUG:wyoming_piper.handler:Sent info
```

Turning off Wakeword detection and using the button works every time for me as well.

The ESPHome logs for the ATOM are incomplete as this is all I can ever get them to display:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
I (30) boot: ESP-IDF 4.4.5 2nd stage bootloader
I (30) boot: compile time 23:16:27
I (30) boot: chip revision: v1.1
I (33) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (70) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (78) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (85) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (92) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2adc4h (175556) map
I (176) esp_image: segment 1: paddr=0003adec vaddr=3ffb0000 size=0300ch ( 12300) load
I (181) esp_image: segment 2: paddr=0003de00 vaddr=40080000 size=02218h (  8728) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=b6b84h (748420) map
I (461) esp_image: segment 4: paddr=000f6bac vaddr=40082218 size=1291ch ( 76060) load
I (503) boot: Loaded app from partition at offset 0x10000
I (503) boot: Disabling RNG early entropy source...
I (514) cpu_start: Pro cpu up.
I (515) cpu_start: Starting app cpu, entry point is 0x4008237c
I (0) cpu_start: App cpu up.
I (531) cpu_start: Pro cpu start user code
I (531) cpu_start: cpu freq: 160000000
I (531) cpu_start: Application information:
I (536) cpu_start: Project name:     m5stack-atom-echo
I (541) cpu_start: App version:      2023.10.0b1
I (547) cpu_start: Compile time:     Oct 13 2023 23:15:00
I (553) cpu_start: ELF file SHA256:  ab8d6202b50ea1e3...
I (559) cpu_start: ESP-IDF:          4.4.5
I (564) cpu_start: Min chip rev:     v0.0
I (568) cpu_start: Max chip rev:     v3.99 
I (573) cpu_start: Chip rev:         v1.1
I (578) heap_init: Initializing. RAM available for dynamic allocation:
I (585) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (591) heap_init: At 3FFB81B8 len 00027E48 (159 KiB): DRAM
[... 100 lines cut ...]
  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/m5stack-atom-echo-8a2640/.pioenvs/m5stack-atom-echo-8a2640/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_67dae && [1/2] Building C object CMakeFiles/cmTC_67dae.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_67dae
    FAILED: cmTC_67dae 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_67dae.dir/testCCompiler.c.obj  -o cmTC_67dae   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: cannot find CMakeFiles/cmTC_67dae.dir/testCCompiler.c.obj: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 22.42 seconds =========================
```

balloob, 2023-10-18:
This has been fixed in latest ESPHome beta 3.

closed 2023-10-18 by balloob

notownblues, 2023-10-18:
I can confirm Wake Word is now rock solid on ESPhome beta

CAVACO-PT, 2023-10-18:
was this already in the release that just got out of ESP 10.0 , or only on beta ?

shampeon, 2023-10-19:
> was this already in the release that just got out of ESP 10.0 , or only on beta ?

Assuming it's [this PR](https://github.com/esphome/esphome/pull/5547), it was in 10.0.

shampeon, 2023-10-19:
Spoke too soon. I'm on 2023.10.1 and my ATOM Echoes stop responding after a period of time. Killing power and restarting them has no effect.

```
INFO Successfully connected to 10.10.10.119
[11:59:14][D][voice_assistant:366]: State changed from IDLE to START_MICROPHONE
[11:59:14][D][voice_assistant:372]: Desired state set to WAIT_FOR_VAD
[11:59:14][D][voice_assistant:159]: Starting Microphone
[11:59:14][D][voice_assistant:366]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[11:59:14][D][esp-idf:000]: I (11441) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[11:59:14][D][voice_assistant:366]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[11:59:14][D][voice_assistant:176]: Waiting for speech...
[11:59:14][D][voice_assistant:366]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[11:59:14][D][voice_assistant:189]: VAD detected speech
[11:59:14][D][voice_assistant:366]: State changed from WAITING_FOR_VAD to START_PIPELINE
[11:59:14][D][voice_assistant:372]: Desired state set to STREAMING_MICROPHONE
[11:59:14][D][voice_assistant:206]: Requesting start...
[11:59:14][W][voice_assistant:217]: Could not request start.
[11:59:14][D][light:036]: 'M5Stack Atom Echo 888038' Setting:
[11:59:14][D][light:051]:   Brightness: 100%
[11:59:14][D][light:059]:   Red: 100%, Green: 0%, Blue: 0%
[11:59:14][D][voice_assistant:366]: State changed from START_PIPELINE to IDLE
[11:59:14][D][voice_assistant:372]: Desired state set to IDLE
[11:59:15][D][light:036]: 'M5Stack Atom Echo 888038' Setting:
[11:59:15][D][light:051]:   Brightness: 100%
[11:59:15][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[11:59:23][D][api:102]: Accepted 192.168.0.8
[11:59:23][D][api.connection:1032]: Home Assistant 2023.10.3 (192.168.0.8): Connected successfully
```

When it worked, these messages would repeat in a loop, but now the unit stays in this state indefinitely.

shampeon, 2023-10-19:
> When it worked, these messages would repeat in a loop, but now the unit stays in this state indefinitely.

This is apparently due to a deadlock, fixed in 1.8.0 of the OpenWakeWord Docker container. https://github.com/rhasspy/wyoming-openwakeword/issues/2