esphome/firmware #68: ESP32 Box Lite blank after instal

MathewJH1, 2023-10-17:
Completed the install process via ESPHome, ESP s3 Box Lite is left blank and no prompt to setup WIFI

yml I am using

---
esphome:
  name: esp32-s3-box-lite
  friendly_name: ESP32 S3 Box Lite
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "1.0"
  min_version: 2023.10.0b1
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: esp32s3box
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

api:
ota:
logger:
  hardware_uart: USB_SERIAL_JTAG
  logs:
    sensor: INFO

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/esp32-s3-box-lite.yaml@main

wifi:
  ap:

sensor:
  - platform: adc
    pin: GPIO1
    id: front_buttons
    update_interval: 16ms
    attenuation: 11db
    on_value:
      - lambda: |-
          // none: 3.121
          // left: 2.392
          // middle: 1.965
          // left+middle: 1.600
          // right: 0.794
          // left+right: 0.726
          // middle+right: 0.682
          // all: 0.632
          if (x > 3) {
            id(left).publish_state(false);
            id(middle).publish_state(false);
            id(right).publish_state(false);
          } else if (x > 2.3) {
            id(left).publish_state(true);
            id(middle).publish_state(false);
            id(right).publish_state(false);
          } else if (x > 1.9) {
            id(left).publish_state(false);
            id(middle).publish_state(true);
            id(right).publish_state(false);
          } else if (x > 1) {
            id(left).publish_state(true);
            id(middle).publish_state(true);
            id(right).publish_state(false);
          } else if (x > 0.73) {
            id(left).publish_state(false);
            id(middle).publish_state(false);
            id(right).publish_state(true);
          } else if (x > 0.7) {
            id(left).publish_state(true);
            id(middle).publish_state(false);
            id(right).publish_state(true);
          } else if (x > 0.65) {
            id(left).publish_state(false);
            id(middle).publish_state(true);
            id(right).publish_state(true);
          } else {
            id(left).publish_state(true);
            id(middle).publish_state(true);
            id(right).publish_state(true);
          }
binary_sensor:
  - platform: template
    id: left
    name: "Left"
  - platform: template
    id: middle
    name: "Middle"
  - platform: template
    id: right
    name: "Right"

  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
    name: Left Top Button
    disabled_by_default: true
    on_click:
      - if:
          condition:
            switch.is_off: use_wake_word
          then:
            - if:
                condition: voice_assistant.is_running
                then:
                  - voice_assistant.stop:
                  - script.execute: reset_led
                else:
                  - voice_assistant.start:
          else:
            - voice_assistant.stop
            - delay: 1s
            - script.execute: reset_led
            - script.wait: reset_led
            - voice_assistant.start_continuous:

output:
  - platform: ledc
    pin: GPIO45
    inverted: true
    id: backlight_output

light:
  - platform: monochromatic
    output: backlight_output
    name: LCD Backlight
    id: led
    restore_mode: ALWAYS_OFF
    disabled_by_default: true
    default_transition_length: 0s
    effects:
      - pulse:
          transition_length: 250ms
          update_interval: 250ms

esp_adf:
  board: esp32s3boxlite

microphone:
  - platform: esp_adf
    id: box_mic

speaker:
  - platform: esp_adf
    id: box_speaker

voice_assistant:
  id: va
  microphone: box_mic
  speaker: box_speaker
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3
  on_listening:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_tts_start:
    - light.turn_on:
        id: led
        blue: 0%
        red: 0%
        green: 100%
        brightness: 100%
        effect: pulse
  on_end:
    - delay: 100ms
    - wait_until:
        not:
          speaker.is_playing:
    - script.execute: reset_led
  on_error:
    - light.turn_on:
        id: led
        blue: 0%
        red: 100%
        green: 0%
        brightness: 100%
        effect: none
    - delay: 1s
    - script.execute: reset_led
    - script.wait: reset_led
    - lambda: |-
        if (code == "wake-provider-missing" || code == "wake-engine-missing") {
          id(use_wake_word).turn_off();
        }

script:
  - id: reset_led
    then:
      - if:
          condition:
            switch.is_on: use_wake_word
          then:
            - light.turn_on:
                id: led
                blue: 100%
                red: 100%
                green: 0%
                brightness: 100%
                effect: none
          else:
            - light.turn_off: led

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
      - script.execute: reset_led
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - script.execute: reset_led

external_components:
  - source: github://pr#5230
    components: esp_adf
    refresh: 0s

psram:
  mode: octal
  speed: 80MHz

MathewJH1, 2023-10-18:
Log files from esp

ESP-ROM:esp32s3-20210327
[D][binary_sensor:036]: 'Left Top Button': Sending state ON
[D][binary_sensor:036]: 'Left Top Button': Sending state OFF
[W][light:176]: 'LCD Backlight' - This color mode does not support setting RGB color!
[D][light:036]: 'LCD Backlight' Setting:
[D][light:051]:   Brightness: 100%
[E][voice_assistant:412]: No API client connected
[D][voice_assistant:366]: State changed from IDLE to IDLE
[D][voice_assistant:372]: Desired state set to IDLE
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420b7c6a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x162c
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f2c
entry 0x403c9950
I (24) boot: ESP-IDF 4.4.5 2nd stage bootloader
I (25) boot: compile time 19:44:55
I (25) boot: chip revision: v0.1
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 16MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (72) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (79) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (87) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (94) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=2de58h (187992) map
I (141) esp_image: segment 1: paddr=0003de80 vaddr=3fc97240 size=02198h (  8600) load
I (143) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=bdb38h (777016) map
I (286) esp_image: segment 3: paddr=000fdb60 vaddr=3fc993d8 size=01f7ch (  8060) load
I (288) esp_image: segment 4: paddr=000ffae4 vaddr=40374000 size=13238h ( 78392) load
I (316) boot: Loaded app from partition at offset 0x10000
I (317) boot: Disabling RNG early entropy source...
I (318) opi psram: vendor id : 0x0d (AP)
I (321) opi psram: dev id    : 0x02 (generation 3)
I (327) opi psram: density   : 0x03 (64 Mbit)
I (332) opi psram: good-die  : 0x01 (Pass)
I (337) opi psram: Latency   : 0x01 (Fixed)
I (341) opi psram: VCC       : 0x01 (3V)
I (346) opi psram: SRF       : 0x01 (Fast Refresh)
I (351) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (357) opi psram: BurstLen  : 0x01 (32 Byte)
I (362) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (368) opi psram: DriveStrength: 0x00 (1/1)
I (373) spiram: Found 64MBit SPI RAM device
I (378) spiram: SPI RAM mode: sram 80m
I (382) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (389) cpu_start: Pro cpu up.
I (393) cpu_start: Starting app cpu, entry point is 0x40376014
I (0) cpu_start: App cpu up.
I (694) spiram: SPI SRAM memory test OK
I (703) cpu_start: Pro cpu start user code
I (703) cpu_start: cpu freq: 240000000
I (704) cpu_start: Application information:
I (704) cpu_start: Project name:     esp32-s3-box-lite
I (705) cpu_start: App version:      2023.10.0b3
I (706) cpu_start: Compile time:     Oct 17 2023 19:41:22
I (706) cpu_start: ELF file SHA256:  2cd0a801a9403264...
I (707) cpu_start: ESP-IDF:          4.4.5
I (708) cpu_start: Min chip rev:     v0.0
[... 210 lines cut ...]
[C][wifi:386]:   BSSID: [redacted]
[C][wifi:387]:   Hostname: 'esp32-s3-box-lite'
[W][wifi_esp32:889]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_CONN
[C][wifi:389]:   Signal strength: 0 dB ▂▄▆█
[C][wifi:393]:   Channel: 1
[C][wifi:394]:   Subnet: 255.255.255.0
[C][wifi:395]:   Gateway: 192.168.4.1
[C][wifi:396]:   DNS1: 0.0.0.0
[C][wifi:397]:   DNS2: 0.0.0.0
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: esp32-s3-box-lite
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: esp32-s3-box-lite.local:3232
[W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 2 restarts
[C][api:138]: API Server:
[C][api:139]:   Address: esp32-s3-box-lite.local:6053
[C][api:143]:   Using noise encryption: NO
[D][binary_sensor:036]: 'Left Top Button': Sending state ON
[D][binary_sensor:036]: 'Left Top Button': Sending state OFF
[W][light:176]: 'LCD Backlight' - This color mode does not support setting RGB color!
[D][light:036]: 'LCD Backlight' Setting:
[D][light:051]:   Brightness: 100%
[E][voice_assistant:412]: No API client connected
[D][voice_assistant:366]: State changed from IDLE to IDLE
Terminal disconnected: NetworkError: The device has been lost.

closed 2023-10-24 by MathewJH1

bert269, 2023-12-16:
I used the same code as supplied on Github, with wifi credentials added, but it does not give any indication on the screen that it is working after the flash. It seems to go into a boot-loop, as I hear it connecting and dsconnection constantly form the USB-connection sound.
The default code does not prompt for the wifi credentials, on the box-lite.
I will try to flash one of the S3-boxes (non-lite) to see if that works better.