esphome/firmware #69: ESP32-S3-Box It looks like the microphone is not working at all.

FaintGhost, 2023-10-17:
Using the ESPHome normal installation with the good, HA entities are appearing, Assist Pipeline is configured, but no response whether holding down the left button to speak or using the wake word.

The logs are as follows:
```
INFO ESPHome 2023.10.0b3
INFO Reading configuration /config/esphome/esp32-s3-box.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO45 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from esp32-s3-box.local using esphome API
INFO Successfully connected to esp32-s3-box.local
[18:57:42][I][app:102]: ESPHome version 2023.10.0b3 compiled on Oct 17 2023, 18:50:30
[18:57:42][I][app:104]: Project esphome.voice-assistant version 1.0
[18:57:42][C][wifi:546]: WiFi:
[18:57:42][C][wifi:382]:   Local MAC: 7C:DF:A1:FF:41:28
[18:57:42][C][wifi:383]:   SSID: 'zzz'[redacted]
[18:57:42][C][wifi:384]:   IP Address: 192.168.50.142
[18:57:42][C][wifi:386]:   BSSID: 5C:E7:47:4F:64:BC[redacted]
[18:57:42][C][wifi:387]:   Hostname: 'esp32-s3-box'
[18:57:42][C][wifi:389]:   Signal strength: -28 dB ▂▄▆█
[18:57:42][C][wifi:393]:   Channel: 1
[18:57:42][C][wifi:394]:   Subnet: 255.255.255.0
[18:57:42][C][wifi:395]:   Gateway: 192.168.50.1
[18:57:42][C][wifi:396]:   DNS1: 192.168.50.1
[18:57:42][C][wifi:397]:   DNS2: 0.0.0.0
[18:57:42][C][logger:361]: Logger:
[18:57:42][C][logger:362]:   Level: DEBUG
[18:57:42][C][logger:363]:   Log Baud Rate: 115200
[18:57:42][C][logger:365]:   Hardware UART: USB_SERIAL_JTAG
[18:57:42][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Mute'
[18:57:42][C][gpio.binary_sensor:016]:   Pin: GPIO1
[18:57:42][C][ledc.output:164]: LEDC Output:
[18:57:42][C][ledc.output:165]:   Pin GPIO45
[18:57:42][C][ledc.output:166]:   LEDC Channel: 0
[18:57:42][C][ledc.output:167]:   PWM Frequency: 1000.0 Hz
[18:57:42][C][ledc.output:168]:   Bit depth: 14
[18:57:42][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Top Left Button'
[18:57:42][C][gpio.binary_sensor:016]:   Pin: GPIO0
[18:57:42][C][light:103]: Light 'LCD Backlight'
[18:57:42][C][light:105]:   Default Transition Length: 0.0s
[18:57:42][C][light:106]:   Gamma Correct: 2.80
[18:57:42][C][template.switch:068]: Template Switch 'Use wake word'
[18:57:42][C][template.switch:091]:   Restore Mode: restore defaults to ON
[18:57:42][C][template.switch:057]:   Optimistic: YES
[18:57:42][C][psram:020]: PSRAM:
[18:57:42][C][psram:021]:   Available: YES
[18:57:42][C][psram:024]:   Size: 8191 KB
[18:57:42][C][mdns:115]: mDNS:
[18:57:42][C][mdns:116]:   Hostname: esp32-s3-box
[18:57:42][C][ota:097]: Over-The-Air Updates:
[18:57:42][C][ota:098]:   Address: esp32-s3-box.local:3232
[18:57:42][C][api:138]: API Server:
[18:57:42][C][api:139]:   Address: esp32-s3-box.local:6053
[18:57:42][C][api:141]:   Using noise encryption: YES
[18:57:52][D][binary_sensor:036]: 'Top Left Button': Sending state ON
[18:57:54][D][binary_sensor:036]: 'Top Left Button': Sending state OFF
[18:57:57][D][binary_sensor:036]: 'Top Left Button': Sending state ON
[18:57:59][D][binary_sensor:036]: 'Top Left Button': Sending state OFF
[19:00:01][D][switch:012]: 'Use wake word' Turning ON.
[19:00:01][D][switch:055]: 'Use wake word': Sending state ON
[19:00:01][D][voice_assistant:366]: State changed from IDLE to START_MICROPHONE
[19:00:01][D][voice_assistant:372]: Desired state set to WAIT_FOR_VAD
[19:00:01][D][light:036]: 'LCD Backlight' Setting:
[19:00:01][D][light:047]:   State: ON
[19:00:01][D][light:051]:   Brightness: 25%
[19:00:01][D][voice_assistant:159]: Starting Microphone
[19:00:01][D][voice_assistant:366]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[19:00:01][D][esp-idf:000]: I (471276) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=8

[19:00:01][D][esp-idf:000]: I (471284) I2S: I2S0, MCLK output by GPIO2

[19:00:01][D][esp-idf:000]: I (471290) ESP32_S3_BOX: I2S0, MCLK output by GPIO0

[19:00:01][D][esp-idf:000]: I (471297) AUDIO_PIPELINE: link el->rb, el:0x3d81134c, tag:i2s, rb:0x3d811760

[19:00:01][D][esp-idf:000]: I (471304) AUDIO_PIPELINE: link el->rb, el:0x3d8114c0, tag:filter, rb:0x3d8137a0

[19:00:01][D][esp-idf:000]: I (471314) AUDIO_ELEMENT: [i2s-0x3d81134c] Element task created

[19:00:01][D][esp-idf:000]: I (471318) AUDIO_THREAD: The filter task allocate stack on external memory

[19:00:01][D][esp-idf:000]: I (471325) AUDIO_ELEMENT: [filter-0x3d8114c0] Element task created

[19:00:01][D][esp-idf:000]: I (471333) AUDIO_ELEMENT: [raw-0x3d8115f0] Element task created

[19:00:01][D][esp-idf:000]: I (471343) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8397763 Bytes, Inter:97172 Bytes, Dram:97172 Bytes

[19:00:01][D][esp-idf:000]: I (471350) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1

[19:00:01][D][esp-idf:000]: I (471358) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1

[19:00:01][D][esp-idf:000]: I (471364) RSP_FILTER: sample rate of source data : 16000, channel of source data : 2, sample rate of destination data : 16000, channel of destination data : 1

[19:00:01][D][esp-idf:000]: I (471370) AUDIO_PIPELINE: Pipeline started

[19:00:01][D][esp_adf.microphone:294]: Microphone started
[19:00:01][D][voice_assistant:366]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[19:00:01][D][voice_assistant:176]: Waiting for speech...
[19:00:01][D][voice_assistant:366]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[19:00:48][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:00:48][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed

```

hugobloem, 2023-10-18:
I got a bit further by lowering the VAD threshold to 1. Then it does seem to send its audio to Home Assistant. However, there are some issues with the connection on the HA side of things. PRs for this have been approved but I think we have to wait for a new HA release.

e.g.:
[Don't warn about unknown pipeline events in ESPHome](https://github.com/home-assistant/core/commit/474f4329bc7769ae2cda787309897ba34f7b99fe)
[Close existing UDP server for ESPHome voice assistant](https://github.com/home-assistant/core/commit/f0317f0d59ef193d826393d5cce50fe610a00010)

FaintGhost, 2023-10-19:
> I got a bit further by lowering the VAD threshold to 1. Then it does seem to send its audio to Home Assistant. However, there are some issues with the connection on the HA side of things. PRs for [...]

Thanks for the explanation, I tried for a long time and was frustrated that it wouldn't work, now I just have to be quiet and wait for the fix.

jjmuriel, 2024-06-02:
Same here. 

**Hardware: 
HA Yellow with 8GB RAM and 512 GB SSD.
ESP32-S3-BOX 3**

**Core 2024.5.5
Supervisor 2024.05.1
Operating System 12.3
Frontend 20240501.1**

**ESPHome Add On 2024.5.4**

Several warnings during compilation:

```
INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/esp32-s3-box-3-XXXyaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@None
INFO Compiling app...
Processing esp32-s3-box-3-XXX (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- Improv @ 1.2.3
|-- noise-c @ 0.1.4
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/api_connection.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/api_server.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/list_entities.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/proto.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/api/user_services.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/button/button.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/captive_portal/captive_portal.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/display/display.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/display/display_buffer.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/display/rect.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32/core.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble/ble.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble/ble_advertising.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble/ble_uuid.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble_server/ble_2901.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble_server/ble_2902.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble_server/ble_characteristic.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble_server/ble_descriptor.o
Compiling .pioenvs/esp32-s3-box-3-05b148/src/esphome/components/esp32_ble_server/ble_server.o
[... 2624 lines cut ...]
    - lambda: id(voice_assistant_phase) = ${voice_assist_not_ready_phase_id};
    - script.execute: draw_display

script:
  - id: draw_display
    then:
      - if:
          condition:
            lambda: return !id(init_in_progress);
          then:
            - if:
                condition:
                  wifi.connected:
                then:
                  - if:
                      condition:
                        api.connected:
                      then:
                        - lambda: |
                            switch(id(voice_assistant_phase)) {
                              case ${voice_assist_listening_phase_id}:
                                id(s3_box_lcd).show_page(listening_page);
                                id(s3_box_lcd).update();
                                break;
                              case ${voice_assist_thinking_phase_i

Windyo, 2024-06-12:
Are you guys compiling on Windows ? if yes it could be linked to https://github.com/esphome/issues/issues/5389

maxi1134, 2024-06-16:
> Are you guys compiling on Windows ? if yes it could be linked to [esphome/issues#5389](https://github.com/esphome/issues/issues/5389) of which the fix was simlply fire up linux and compile there.

I am compiling directly within HA and can't get the mic to work.

closed 2025-05-14 by github-actions[bot]