esphome/firmware #73: Muse Luxe: Reset after Reboot

mroxso, 2023-10-19:
Hey,
i have successfully installed voice-assistant on the Muse Luxe.
What I did:
1. I already had a media-player Muse Luxe on Home Assistant (ESPHome)
2. Then I copied the new voice-assistant code to the existing configuration file (I replaced encryption key and wifi access)
3. Then I flashed the new config to the Muse Luxe

Everything seemed fine and working.
However after I restarted the Muse Luxe, it resetted to it it's old firmware (media-player). How is this possible?
I also noticed that it's speaker cracks now a little bit. And I also noticed that it is getting warm where the buttons are at the top right.

Any suggestions?

mroxso, 2023-11-22:
I did not manage to solve this issue yet.
Any ideas?

oraad, 2023-12-27:
I had the same issue when I tried the voice-assistant, but I'm not using the voice assistant, so I didn't try to solve it.
Recently, I was reading about bluetooth proxy component on [ESPHome](https://esphome.io/components/bluetooth_proxy) , and I found this note in the "Warning" section

`The [ESP32 Platform](https://esphome.io/components/esp32) component should be configured to use the esp-idf framework, as the arduino framework uses significantly more memory and performs poorly with the Bluetooth proxy enabled. When switching from arduino to esp-idf, make sure to update the device with a serial cable as the partition table is different between the two frameworks as [OTA Update Component](https://esphome.io/components/ota) updates will not change the partition table.`

The media_player template uses arduino and the voice-assistant uses esp-idf, and I'm guessing the partitioning is causing the reset issue

mroxso, 2024-01-02:
> I had the same issue when I tried the voice-assistant, but I'm not using the voice assistant, so I didn't try to solve it. Recently, I was reading about bluetooth proxy component on [...]

Thank you for the information!

After tinkering around and installing the newest version of voice-assistant on it, it seems that this bug has been fixed for me.
Maybe because it uses the esp-idf framework now.

closed 2024-01-02 by mroxso