esphome/firmware #75: Wake Word and PTT - Best of both worlds

timmo001, 2023-10-21:
Is there a way to be able to use the push to talk whilst wake word is enabled? I've been experimenting but couldnt find a way other than disabling wake word and turning it back on after a 30 seconds or so

```
      - if:
          condition:
            switch.is_off: use_wake_word
          then:
            - if:
                condition: voice_assistant.is_running
                then:
                  - voice_assistant.stop:
                  - delay: 1s
                  - script.execute: reset_led
                  - script.wait: reset_led
            - voice_assistant.start:
          else:
            - voice_assistant.stop:
            - switch.turn_off: use_wake_word
            - delay: 1s
            - voice_assistant.start:
            - delay: 30s
            - switch.turn_on: use_wake_word

```

rwjack, 2023-12-01:
Desperately need this, and most importantly, the ability to toggle a sensor to have the microphone listen.

Eg. Press a button on my living room zigbee controller, so the atom echo that's 5ft away starts listening.

timmo001, 2023-12-01:
I'm devising a similar thing with a template button its not how I like it though. Be nice if there was a nicer way to override wake word

timmo001, 2025-05-07:
May be a little outdated but I still think this could be a good addition