esphome/firmware #77: Add media player to echo voice assistant

itsme9, 2023-10-23:
Hi, if possible could a media player entity be added to the atom echo voice assistant config
Thanks

riesi77, 2023-11-07:
Would be a great improvement and I guess appreciated by many people.

gigawebs, 2023-11-07:
This would be very helpfulv, would love to see this.

pdsccode, 2023-11-08:
Chiming in here as well. Not only for the echo but for the use in conjunction with esp-idf framework in general. This would enable us to build our own voice assistant and use the same device as target for announcements (and music as well with the right speakers).

All about sustainability 😉

mehhhhhhhhhhhhhhhhhhhhhhhhhh, 2023-11-11:
YES PLEASE!!!!

ikocijan, 2023-11-21:
This would be a great addition. My use case is that I would use the atom echo as a speaker for my alarm. I bought the device thinking it's both a speaker and a voice assistant, but I was surprised that it's not recognized as a media player.

kksligh, 2023-11-23:
agreed

morpheus8888, 2023-11-25:
please

riesi77, 2023-12-14:
@jesserockz : Is this the correct  feature request for getting a media player on esp-idf? https://github.com/esphome/feature-requests/issues/1864?

RaceNJason, 2024-04-01:
@jesserockz - if you are going to close this issue, as riesi77 put it...where should we place our feature request for getting a media player on esp-idf then? Even if there is no planned timeline, this issue shouldn't be closed unless there is an alternative for requesting this feature. I don't use GitHub much so I'm not sure if there is a specific place to do feature requests (so please don't burn me if there is) and for the moment...this is the only thread I've seen relating to this issue and to the ESPHome GitHub repository.

What I do know about the M5Stack Echo is that it obviously supports I2S speaker and microphone already and that HA can send TTS to it already. No one here seems to actually want a 'media player' per say to play music from their echo (if they did, that would be sad). What everyone really wants is to be able to do speech notifications or sounds (like doorbells and such). Especially since what is the point of having both a voice assistant and a separate speaker when they should be able to do it in one unit. Maybe HA needs another device type like just a basic speaker which can be linked to TTS and sound notifications instead of full blown media players (and that could be added to the M5Stack Echo). Anyway, off subject...so if someone can provide a link to the proper channel so future people visiting this page will have a place to go and add their vote that would be great. Or, somehow reopen this issue even if it is on the back burner with no timeline. At least ESPHome developers will know how many people want to have this feature added. I would think that is more important than closing an issue just because there isn't any current plan for a feature like this...

ChristophCaina, 2024-06-04:
Hi @jesserockz,
can you maybe have a look into this repository?
It seems, that this is a working example of VoiceAssistant + MediaPlayer on an S3-Box3

https://github.com/gnumpi/esphome_audio/blob/06718e7412175e481f111ee110160bfe042525ab/examples/esp32-s3-box-3-dplx.yaml

RaceNJason, 2024-06-04:
If I had a S3-Box3 I would test this immediately. Every time I've tried to acquire one they are either out of stock or some seller wants an exorbitant amount for it. Mouser says new inventory is suppose to be coming June 7 which is a few days away. The closest thing I have to an S3-Box3 is a K128 (Core S3 Box) which is similar, but not exact.

Along those lines though, I could've sworn I've seen a YouTube video where someone did exactly this (and possibly with the code you are looking at). Especially when I saw the Casita images section which I'm absolutely sure I saw on a YouTube video (i.e. the images in response to the voice commands). Of course I don't know if any of it will help to develop a MediaPlayer type interface to the echo (what this thread is about), but maybe it will help point it in the right direction??

Anyway, if there is someone with an S3-Box3 who wants to try this and report back their experience (along with if it could be applicable to an M5Stack Echo), it could be helpful to the rest of us. I'll assume ChristophCaina doesn't have an S3-Box3, otherwise the post would've reported the experience rather than a comment to look into it...

jarodwsams, 2024-09-07:
> Hi @jesserockz, can you maybe have a look into this repository? It seems, that this is a working example of VoiceAssistant + MediaPlayer on an S3-Box3
> 
> https://github.com/gnumpi/esphome_audio/blob/06718e7412175e481f111ee110160bfe042525ab/examples/esp32-s3-box-3-dplx.yaml

I've got an Atom Echo that I'm going to try this with. I recently updated my echo nodes to the newer [OpenWakeWord config from ESPHome](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml) and I really like how much more performant it is.

The gnumpi repository has an [Atom Echo example](https://github.com/gnumpi/esphome_audio/blob/06718e7412175e481f111ee110160bfe042525ab/examples/m5stack-atom-echo-adf.yaml). I'm going to try that bare, first, and then try to combine the OWW base with the changes from the gnumpi example file. I'll share results here, in case anyone else finds this thread even though it's closed.

BVerschaete, 2025-01-21:
@jarodwsams Did you ever get a working configuration for the combination of Gnumpi's esphome audio config and the esphome voice_assistant config?

SilentBob83, 2025-02-13:
hello, i am trying this constellation VA + Media player on the M5Stack Echo.

but it seems that i am having a syntax problem : 

adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: i2s_dplx
    i2s_dout_pin: GPIO22
    fixed_settings: false

is red marked, and the element after adf_pipeline are read faulty .

i copied the ymal content from the /examples/m5stack-atom-echo-mWW.yaml File. 

can anybody give me a direction ?

jesserockz, 2025-02-14:
The maintained configurations located in esphome/wake-word-voice-assistants will be updated after the ESPHome 2025.2.0 release next week. The voice assistant files in this repository can be considered archived and may be removed at some point in the future.

Jesse