esphome/firmware #87: Request: ESP32-S3-BOX-3

saya6k, 2023-11-02:
ESP32-S3-BOX-3 has been advertised by Home Assistant like new standard for esp32 voice assistant.

Today I got my package, and realized esp32-s3-box.yaml don't work with box-3.

Please make new firmware for ESP32-S3-BOX-3.

jarodwsams, 2023-11-06:
I'll second this. I just received my Box 3 this weekend.

ChristophCaina, 2023-11-08:
Hi there :)
I also received my Box3 today...

Right now, I am trying to figure out, what exactly might has changed, because my current configuration of the S3-Box (without VoiceAssist) does not work on the new S3-Box3...

Display Backlight and the Display seem to use a different GPIO than for the S3-Box...

In addition to that, it would be great, if the additional Hardware (Sensor-Board) could somehow be implemented, at least, with a documentation what GPIOs and so on needs to be configured.

As far as I got, the box3 comes with a built in Gyroscope, in addition to that, the additional Sensor Board will support a RADAR Sensor for Presence Detection, an IR emitter and receiver, Temp & Hum Sensor, Battery Charger...

niemyjski, 2023-11-09:
I was kind of shocked as well as this device is listed on the blog.

saya6k, 2023-12-08:
I will close this since there's working config of ESP32-S3-BOX-3