esphome/firmware #90: M5Stack Atom Echo V1.1 update

molly200, 2023-11-04:
I've  recently purchased three M5Atom Echo units on amazon.com and the seller was advertised as the M5stack store.

Spent quite some time to get the microphone working with the voice assistant code and accomplished nothing. 

Tested the factory echo record example "repeater.ino" and it did not work. However, it appears that the Atom library has been updated and the updated example "RecordPlay.ino" located at https://github.com/m5stack/M5Atom/blob/master/examples/Echo/RecordPlay/RecordPlay.ino works on these units. The updated code worked in my testing.

Opened the case and noticed that the board version is now "v1.1 Echo". The board seems to be slightly different from V1.0.  Every other test regarding the LED and speaker mode seems to work OK!  It leads me to believe that there are some changes in hardware, which have not yet been disclosed. It may also explain the problems which a number of users are having with these new units.

Wondering if the Atom Echo voice assistant project and code will be updated for the new boards?

Thanks

martin131, 2023-11-06:
I cannot confirm I have 1.1 version board but I ordered it recently a I cannot get it work too.

markusressel, 2023-11-06:
Same thing here. I cannot get the microphone to work, opened the unit because of this issue and it says V1.1 on the board. No idea what to do from here.

baerrs, 2023-11-12:
And here two... I have the 1.1 version as well

wlayher, 2023-11-12:
I also have a device with the revision v1.1 of the board. It executes one voice command, then it gets stuck and does not jump back into the program loop. After a reset of the device it executes one voice command and gets stuck again. Also the speaker does not give confirmation feedback of the wake word.

ernasipi, 2023-11-23:
Can also confirm same issue