esphome/firmware #91: Muse luxe loud static when assist is turned on

w1ll1am23, 2023-11-04:
Installed the firmware today and setup the full assist pipeline. When pressing the play button or toggling the assist switch in the HA UI I get full volume static blasting out of the speakers. 

Any idea what's going on?

w1ll1am23, 2023-11-05:
I think this has something to do with running on a vlan separate from home assistant. If I move this over to the HA vlan it seems to work.

mroxso, 2023-11-22:
I don't have this issue. I don't use VLAN (yet). So maybe this is the issue, yes.

fyhertz, 2023-11-26:
I have observed the same behavior when the muse luxe is not on the same network segment as HA. It would be nice if this was stated in the documentation :) A clean error message if possible would be even better :)

jesserockz, 2023-11-27:
I am unable to reproduce this with my luxe. 
Added, deleted, reflashed, tested on main vlan, NoT vlan. HA always connects, and its silently waiting for my wake word which works correctly.

closed 2025-05-13 by github-actions[bot]