esphome/firmware #93: GL-S10 and other diy proxies aren't available on the install page anymore

blakadder, 2023-11-06:
GL-S10 and other diy proxies aren't available on the install page anymore

LeiChat, 2023-11-29:
Slightly off-topic, sorry. Noticed you had requested a photo of the new version of the GL-S10 on an issue in the old repository, so thought I would share here. I am using your web flasher but not sure which button to press to put the GL-S10 v2.1 into flashing mode.
[image]

KSchmeeds, 2023-12-13:
I'm noticing this as well, it seems the advanced/DIY option is gone from the webpage.

jesserockz, 2023-12-22:
Yes, we felt that they did not "fit" well with the aim of the new ready made projects page. 

We still build the firmware files, and a better place to have a page for flashing specific devices could potentially be on https://devices.esphome.io

closed 2023-12-22 by jesserockz

blakadder, 2023-12-22:
I've put up firmware for both versions of GL-S10 on https://blakadder.github.io/bluetooth-proxies/