esphome/firmware #107: ESP32-S3-BOX-3

PhilippeP62, 2023-11-23:
Since yesterday HA/Esphome wants me to upgrade to 2023.11.3 but current GitHub version is .2

There is a typo in the yaml : 

  package_import_url: github://esphome/firmware/voice-assistant/esp32-s3-box.yaml@main

It should be 

  package_import_url: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main

Then when trying to update , it t also throw errors about the Casita PNG files not found , it works when installing through Esphome.io/USB .

PhilippeP62, 2023-11-23:
Update error log :

INFO ESPHome 2023.11.3 INFO Reading configuration /config/esphome/esp32-s3-box-3-5ab118.yaml... INFO Updating https://github.com/esphome/esphome.git@pull/5230/head WARNING GPIO0 is a strapping PIN and should only be used for I/O with care. Attaching external pullup/down resistors to strapping pins can cause unexpected failures. See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins Failed config image: [source /data/packages/c46f54c1/voice-assistant/esp32-s3-box-3.yaml:234] 
- Could not find file '/config/esphome/casita/Upset@2x.png'. Please make sure it exists (full path: /config/esphome/casita/Upset@2x.png). file: casita/Upset@2x.png id: casita_error resize: 200x200 type: RGB24 use_transparency: True 
- Could not find file '/config/esphome/casita/Sleeping@2x.png'. Please make sure it exists (full path: /config/esphome/casita/Sleeping@2x.png). file: casita/Sleeping@2x.png id: casita_idle resize: 200x200 type: RGB24 use_transparency: True
...

PhilippeP62, 2023-11-24:
2023.11.4 has resolved the issue 
But the yaml typo is still there.

jlpouffier, 2023-12-07:
This was fixed on : https://github.com/esphome/firmware/pull/106

closed 2023-12-08 by PhilippeP62