esphome/firmware #117: listen tone instead of listen light

viperjet, 2023-12-07:
It's a feature request. In case of not always have a look on the atom 5 echo, it would be very nice to play a listen tone, eg. a simple beep instead of the light. 

Is this possible?

Thanks,
Udo

RemakingEden, 2023-12-17:
I'd second this. It would be really helpful, if possible

Nardol, 2023-12-20:
It could also enhance accessibility for blind users.
I don't know how to generate raw audio, which is the only way to make a sound with speaker component. For media_player a sound can be played even if there would be some lag IMO.

umglurf, 2024-06-22:
There are some possible solutions in https://community.home-assistant.io/t/play-sound-when-wakeword-detected/653928

To generate audio suitable for `speaker.play`, you can use the following commands (where `INPUT_FILE` is any kind of audio file that ffmpeg supports):
```
ffmpeg -i INPUT_FILE -ac 1 -acodec pcm_u8 -ar 16000 out.wav
sox out.wav --bits 8 --encoding signed-integer --endian little out.raw
```

rwjack, 2024-08-14:
I saw something on the community forums that someone modified the firmware to play a sound. Does anyone have detailed instructions on how to do so?

umglurf, 2024-08-15:
> I saw something on the community forums that someone modified the firmware to play a sound. Does anyone have detailed instructions on how to do so?

Have a look at https://community.home-assistant.io/t/play-sound-when-wakeword-detected/653928/27?u=umglurf
How to generate the sound is described in the comment above and in https://community.home-assistant.io/t/play-sound-when-wakeword-detected/653928/16?u=umglurf

rwjack, 2024-08-15:
Awesome, thanks!