esphome/firmware #118: ESPHome BT Proxy crashes

hmakmur, 2023-12-09:
I have been trying to figure out what was going on with my LD2410 sensors which kept getting to 'Unavailable' state.  I use ESPHome BT proxy with it. After narrowing it down, it seems that I kept seeing the BT Proxy having  errors.

I have 3 LD2410 sensors and 4 BT Proxies. It seems to me that is plenty capacity for the proxies to handle 3 sensors.  My other wifi based ESP devices have no issues with any connections. 

Since there is not much is exposed in BTProxy for HA, I can't really monitor BT Proxy. There is no uptime, other diagnostic info I can watch in HD. Looking at HA logs I see something like below.Hopefully someone can shed some lights why the BT Proxy goes 'Unavailable' and how to fix it.  

Thanks.

The error log shows:
```
home-assistant.log:2023-12-09 11:54:48.121 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
home-assistant.log:bleak_retry_connector.BleakOutOfConnectionSlotsError: HLK-LD2410B_A4CB - 8A:5A:C4:15:A4:CB: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address 8A:5A:C4:15:A4:CB was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device
home-assistant.log:2023-12-09 11:58:50.050 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log:2023-12-09 11:58:53.525 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log:bleak_retry_connector.BleakOutOfConnectionSlotsError: HLK-LD2410B_BFBF - 19:96:5F:6E:BF:BF: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address 19:96:5F:6E:BF:BF was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device
home-assistant.log:2023-12-09 12:02:40.665 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log:2023-12-09 12:22:41.861 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log.1:2023-12-09 00:19:37.016 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
home-assistant.log.1:2023-12-09 00:19:42.619 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 00:27:37.359 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 00:27:39.137 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
home-assistant.log.1:2023-12-09 00:32:47.652 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
home-assistant.log.1:2023-12-09 00:44:14.759 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log.1:2023-12-09 00:45:38.096 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 00:50:38.243 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: [Errno 104] Connection reset by peer
home-assistant.log.1:2023-12-09 00:55:14.242 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log.1:2023-12-09 01:02:08.593 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log.1:2023-12-09 01:07:39.723 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
home-assistant.log.1:2023-12-09 01:07:48.828 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: [Errno 104] Connection reset by peer
home-assistant.log.1:2023-12-09 01:07:48.842 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Read failed (ReadFailedAPIError)
home-assistant.log.1:2023-12-09 01:08:52.906 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 01:19:38.251 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 01:26:18.159 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:bleak_retry_connector.BleakOutOfConnectionSlotsError: HLK-LD2410B_BFBF - 19:96:5F:6E:BF:BF: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address 19:96:5F:6E:BF:BF was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device
home-assistant.log.1:2023-12-09 01:31:52.768 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 01:44:42.871 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 01:58:29.205 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 01:59:50.674 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log.1:2023-12-09 02:02:52.228 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
home-assistant.log.1:2023-12-09 02:03:02.238 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Hello timed out (TimeoutAPIError)
home-assistant.log.1:2023-12-09 02:03:02.238 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection lost
home-assistant.log.1:bleak_retry_connector.BleakOutOfConnectionSlotsError: HLK-LD2410B_A4CB - 8A:5A:C4:15:A4:CB: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address 8A:5A:C4:15 [... cut ...]
home-assistant.log.1:2023-12-09 03:12:52.510 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-d01390 @ 192.168.11.159: Connection error occurred: esp32-bluetooth-proxy-d01390 @ 192.168.11.159: EOF received
home-assistant.log.1:2023-12-09 03:43:32.343 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: Connection error occurred: esp32-bluetooth-proxy-6a1604 @ 192.168.11.179: EOF received
home-assistant.log.1:2023-12-09 07:39:06.610 WARNING (MainThread) [aioesphomeapi.connection] esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: Connection error occurred: esp32-bluetooth-proxy-67eb80 @ 192.168.11.183: EOF received
```

oywino, 2024-01-31:
What version of ESPHome have you flashed the GL-S10 with?
Which yaml file are you using?

hmakmur, 2024-01-31:
I dont have GL-S10. I use [BT Proxy of ESP32](https://esphome.io/components/bluetooth_proxy.html). At the time, I had most recent one as of dec 9. Firmware: 2023.10.6 (Nov 14 2023, 05:22:16) 
Each ESP BT Proxy can only take 3 connections I now place 1BT proxy next to the LD2410B hoping that will reduce the "No backend with an available connection slot" error.  This and many fixes in HA had reduced the error message. I had also replaced a few of the ESP32 board as they seem to be dead so quickly.

My current setup still have issues. During the day, for the BT proxy lost the internet for whatever reasons, likely wifi connection problem and the LD2410B goes to 'unavailable'.  

Sometimes reloading the LD2410B works which i do automatically after it goes unavailable for 20 seconds via HA. Often, reloading LD2410B wont work and I think this is likely HA issue. The only fix after that is to restart HA. 

I love the idea of BT Proxy but 3 connections limit and network reliability could easily put LD2410B in to unavailable state. Since my setup is not reliable, when I get time, I plan to go away from BTproxy and go to Wifi.

closed 2025-05-13 by github-actions[bot]