esphome/firmware #128: Cannot connect after flashing to S3-box.

bert269, 2023-12-16:
Ok, I do not know what I am missing here.
I first tried the S3-BOX-lite - it does not prompt for the password. Just display the screen "No WiFi" after a few seconds. My smartphone conencts to the wifi ap, but it never asks for the wifi-credentials.
Then I downloaded the github code, add the wifi credentials to the code, flash that and the devices goes into a boot-loop, Screen stay black all the time.

Then I tried a S3-BOX by flashing the code from ESPHOME.io/projects. Selected the dorrect device, but after flashing the wifi-credentials problem is the same. Never asks for the credentials and screen turn to "no Wi-fi" display.
It was picked up by HA integrations, but clicking on "configure" does not do anything, it jsut tries to connect without any luck.
Do I have to connect the device into the Yellow that runs HA or into any PC connected to the newwork?

kevireilly, 2023-12-27:
If you were able to install the firmware from https://esphome.io/projects then you've got a connection to the device and shouldn't need to plug it in to Yellow. Think I experienced something similar upon initial install, but then I was able to just select "Change wifi" on the pop up from the projects site to configure wifi again (after being fairly confident I just had done so after initial installation).

bert269, 2024-03-06:
The problem was that I connected to the ESP-BOX3 in "download mode" to flash. That is wrong. Just need to power on and connect to USB to flash. Then it worked.
I used Willow before and it required it to be in "download mode", therefore that's how I was used to doing it.

closed 2025-05-12 by github-actions[bot]