esphome/firmware #129: S3 Box 3 working in integration but will not adopt in addon

ardenking, 2023-12-16:
MY S3 box appeared to be corrupt
tried to reload from addon and got the following 

    Could not download from https://github.com/esphome/firmware/raw/main/voice-assistant/casita/Focusing@2x.png: 404 Client Error: Not Found for url: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/Focusing@2x.png.
    file: |-
      https://github.com/esphome/firmware/raw/main/voice-assistant/casita/Focusing@2x.png
    id: casita_thinking
    resize: 200x200
    type: RGB24
    use_transparency: True
  - 
    Could not download from https://github.com/esphome/firmware/raw/main/voice-assistant/casita/Sleeping@2x.png: 404 Client Error: Not Found for url: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/Sleeping@2x.png.
    file: |-
      https://github.com/esphome/firmware/raw/main/voice-assistant/casita/Sleeping@2x.png
    id: casita_vad_off
    resize: 200x200
    type: RGB24
    use_transparency: True  

So I deleted it from addon and installed from https://esphome.io/projects/index.html

Came online and working in integration

Was Discovered by addon  but on trying adopt get the same error and config is the same as before delete 

substitutions:
  name: esp32-s3-box-3-5ab994
  friendly_name: ESP32 S3 Box 3 5ab994
packages:
  esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

Any ideas ?

ChristophCaina, 2023-12-21:
try to remove all build files from the device, delete it completely and let esphome autodetect it again.
I also had some issues with installing the box via the adopt option in the past... but then, at some point it just worked...

kevireilly, 2023-12-27:
I've also had trouble getting Adopt to work, but the device still appeared in ESPHome. From there I was able to successfully Edit > Install or otherwise select the Install option to get it in to ESPHome to observe logs and what not.

closed 2025-05-12 by github-actions[bot]