esphome/firmware #134: Atom M5 echo "Speaker buffer full" causes lock up

HarvsG, 2023-12-21:
https://github.com/esphome/firmware/blob/d320371317b0196055e4414b15acd6a6abc24096/voice-assistant/m5stack-atom-echo.yaml#L1

[logs_m5stack-atom-echo-b836b0_logs (1).txt](https://github.com/esphome/firmware/files/13742200/logs_m5stack-atom-echo-b836b0_logs.1.txt)

As per the attached logs, occasionally I get a speaker buffer full error followed by a full 1 minute lockup if the voice assistant.

Nardol, 2023-12-25:
Look I encounter the same, and TTS response is cut before its end.

Nardol, 2023-12-26:
Things are really better after removing `esp_adf` component.

horga83, 2023-12-30:
removing esp_adf just hangs my M5 and it certainly doesn't work.  Did you change something else, could you post your changes?

horga83, 2023-12-30:
Actually I found that just by commenting out the logger: and improv_serial: it seems much better

Nardol, 2023-12-31:
> removing esp_adf just hangs my M5 and it certainly doesn't work. Did you change something else, could you post your changes?

Strange, it was not so broken after removing esp ADF but finally I put it back but removed logger, improv serial and esp32 improv.

It seams this device has to only keep the minimum necessary for voice assistant, nothing else.

I also removed light but will not recommend it for anybody who is not blind :slightly_smiling_face: even with a so minimum firmware I have to restart the device after some days, which is better.

I really look forward to be able to find somewhere where there are ESP32S3 box in stock because IMO M5 atom echo speaker is not really reliable.

lordbah, 2024-01-02:
I got errors about ble.enable/disable that way so commented those out as well.
```
wifi:
  on_connect:
    - delay: 5s # Gives time for improv results to be transmitted
  #   - ble.disable:
  # on_disconnect:
  #   - ble.enable:
```

sniesler08, 2024-02-20:
I am having the same problem with speaker buffering for responses with voice cutting in and out. Home Assistant did a video showing their M5Stack Atom Echo working with no buffering or anything, I can only conclude they used the stock esphome configuration so I am unsure why my Echo has issue with buffering. 

Did anyone really solve this issue? Any assistant would be amazing...

As I have tried commenting logger, improv_serial and esp32 improv which helped a tiny bit but not much better.

casrou, 2024-03-26:
I actually think it is working now.

After flashing the newest voice assistant from https://esphome.io/projects/ (Voice Assistant 2024.3.0) to the M5Stack Atom Echo and updating all relevant docker containers (esphome, homeassistant, mosquitto, openwakeword, zigbee2mqtt) to their latest version, I'm now able to do multiple voice commands in a row 🎉

gilcu3, 2024-04-05:
> I actually think it is working now.
>
> After flashing the newest voice assistant from https://esphome.io/projects/ (Voice Assistant 2024.3.0) to the M5Stack Atom Echo and updating all relevant docker [...]

Could you share versions of tools? After updating to homeassistant 2024.4.0 I cannot get it to work, everything work except that the device does not output audio and enters in an infinite loop for this reason.

benji2020, 2024-04-29:
I have had issues with it not responding - someone mentions a fix here if it's any help...
https://community.home-assistant.io/t/a-silly-simple-question-about-esphome-specifically-m5stack-atom-echo-firmware/674850/5

closed 2025-05-12 by github-actions[bot]