esphome/firmware #139: ESP-S3-BOX-3 Voice Assistant speaker volume too low

stalakerob, 2024-01-04:
The speaker volume of the Box-3 Voice Assistant is very low. Even in a quiet room you can barely hear it. Increasing the volume_multiplier config setting does not change anything.

nickrout, 2024-01-04:
Esp-s3-box (ie the non -3) is the same.

jaymunro, 2024-01-05:
Volume on mine is quite acceptable for me.
For comparison, I did an audio dB measurement at 50cm using the `Alan` TTS model in piper. Result at 50cm was 71dB max.

letsautomatenet, 2024-02-29:
Interestingly this is the same issue you get with the Sonoff TX Ultimate light switch when you flash that with ESPHome and use is as a media player / TTS.

Hopefully the ESPHome brains will figure it out.

sammcj, 2024-03-04:
I have a esp-s3-box-3 with a very quiet speaker (using it with home assistant/esphome) - is there a standard way to set the volume on these?

cfrudolphy, 2024-03-04:
I have the same issue with both the esp-s3-box-3 and the atom m5 echo device.  Even with my hearing aids in I can't hear the response.  Hopefully someone is working on this.

sammcj, 2024-03-04:
@cfrudolphy so I've got a bit of a dodgy hack for my specific use case with esphome and the esp-s3-box-3, not sure if you can adapt for your usage - https://github.com/sammcj/esphome-esp-s3-box-3-volume

cfrudolphy, 2024-03-05:
Thanks for this.  But first I have to get it adopted into the ESPHome
add-on.  Try tomorrow.

Chuck Rudolphy
346.383.6920
Mobile from my Pixel 6

dmakovec, 2024-03-08:
@sammcj thanks for hacking this out. I'm a bit unfamiliar with how types get cast, but reading your notes and inserting the output.set_level call into the esphome's `on_boot` sequence, I'm seeing the following. Any tips on what I might have missed? TIA!

```
 on_boot: 
    - priority: 600.0
      then: 
        - output.set_level: 
            
            ID 'speaker_volume' of type esp_box_volume::ESPBoxVolume doesn't inherit from output::FloatOutput. Please double check your ID is pointing to the correct value.
            id: speaker_volume
            level: 0.85
        - script.execute: 
        ..
```

mihsu81, 2024-04-02:
@sammcj I have the same issue even after the PR has been merged.

```yaml
  on_boot: 
     - priority: 600.0
       then: 
          - output.set_level: 
            
              ID 'speaker_volume' of type esp_box_volume::ESPBoxVolume doesn't inherit from output::FloatOutput. Please double check your ID is pointing to the correct value.
              id: speaker_volume
              level: 0.85
```

jaymunro, 2024-04-03:
> I have the same issue even after the PR has been merged.

@mihsu81 are you using release 2024.3.1 or dev of the ESPHome addon? The PR has not yet made its way there. I think you'd have to compile a plugin using Jessie's repo to use this or possibly link it in as an external component or something. Not sure how though. I'm sure I'll be one of many eagerly reviewing release notes on the next versions of ESPHome addon.

mihsu81, 2024-04-03:
I'm having the same issues with ESPHome 2024.4.0-dev. I've opened a new issue in @sammcj's repo https://github.com/sammcj/esphome-esp-s3-box-3-volume/issues/1.

cl0ud6uru, 2024-05-02:
Any updates on this? So far everything works great but the speaker volume is almost impossible to hear. As mentioned above, the stock firmware out of the box was plenty loud. I've got the radar and battery sensor working, but I'm at a loss on volume control.

fire219, 2024-05-10:
Confirming it's still an issue. ESPHome 2024.4.2, have added the `external_component` for `esp_box_volume` and even switched the board component source to sammcj's:

```
components:
      - name: esp32_s3_box_3_board
        source: github://sammcj/esphome-esp-s3-box-3-volume@main
        refresh: always
```

Same applies on jesserocks' repo and/or without the `external_component`. Any permutation still gets the `ID 'speaker_volume' of type esp_box_volume::ESPBoxVolume doesn't inherit from output::FloatOutput. Please double check your ID is pointing to the correct value.` error.

After digging through the commits to jesserockz/esp32-s3-box-3-board#4, it looks like the syntax for setting the volume should be a bit different (eg no need for the `esp_box_volume` declaration and id), but I'm not smart enough to figure out how to use it. Trying to apply the volume level to the speaker (`box_speaker` in the HA voice assistant script) gives a similar error as before, but for `esp_adf::ESPADFSpeaker`.

hargcore, 2024-05-21:
Same thing here I can't hear the spoken words on the speaker after the click - I am using Piper (Alan-Low) as my TTS response on my HA Voice Assistant but tried others. If I use the same TTS set up in the Services Dev Tools out of a speaker it plays the full audio end to end at normal volume. Latest version of HA and all add-ons installed and a fresh S3 install as of today.

cl0ud6uru, 2024-05-21:
> Same thing here I can't hear the spoken words on the speaker after the click - I am using Piper (Alan-Low) as my TTS response on my HA Voice Assistant but tried others. If I use the same TTS set up [...]

Mine is no longer playing any sound after the last update. Are you getting sound at all?

hargcore, 2024-05-21:
I discovered the "no sound at all" issue is related to the last update. The fix is here. So unrelated to this muted sound issue:
https://github.com/esphome/issues/issues/5791#issue-2297438479

styphonthal, 2024-06-04:
anyone have luck adjusting the sound?

sammcj, 2024-06-11:
Here's my esp-s3-box-3 esphome config if it helps, it gives me 3 volume settings in HA that I can select from.

[image]

```yaml
---
#------------------------------------------------------------------------------------#
#     PIN Schematics                                                                 #
#                                                                                    #
#       GPIO-00 MCU-BOOT                                                             #
#       GPIO-01 Speaker Mute-Status                                                  #
#       GPIO-02 I2S MCLK                                                             #
#       GPIO-03 Touch-Screen TT21100 Interrupt Pin                                   #
#       GPIO-04 ILI92xxx Display DC-Pin (SPI: CLK-Pin)                               #
#       GPIO-05 ILI92xxx Display CS-Pin (SPI: MOSI-Pin)                              #
#       GPIO-06 ILI92xxx Display SDA                                                 #
#       GPIO-07 ILI92xxx Display SCK                                                 #
#       GPIO-17 I2S_SCLK                                                             #
#       GPIO-40 I2C_SCL (Temp & Hum) -- SENSOR v1.1 [AHT30]                          #
#       GPIO-41 I2C_SDA (Temp & Hum) -- SENSOR v1.1 [AHT30]                          #

substitutions:
  name: esp32-s3-box-3-5acf94
  friendly_name: "ESP32 S3 Box 3 5acf94"

  micro_wake_word_model: alexa
  # micro_wake_word_model: /config/hey_mycroft.json
  wake_word_engine_location: "On device"
  voice_assist_idle_phase_id: "1"
  voice_assist_listening_phase_id: "2"
  voice_assist_thinking_phase_id: "3"
  voice_assist_replying_phase_id: "4"
  voice_assist_not_ready_phase_id: "10"
  voice_assist_error_phase_id: "11"
  voice_assist_muted_phase_id: "12"

  # Request text parameters
  request_text_font_size: "14"
  request_text_start_x: "50"
  request_text_start_y: "25"
  request_text_max_width: "200"
  request_text_line_height: "20"
  request_text_max_lines: "2"

  # Response text parameters
  response_text_font_size: "16"
  response_text_start_x: "5"
  response_text_start_y: "160"
  response_text_max_line_length: "280"
  response_text_line_height: "20"
  response_text_max_lines: "4"

  # 320 x 240
  loading_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/loading.png
  idle_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/idle.png
  listening_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/listening.png
  thinking_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/thinking.png
  replying_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/replying.png
  error_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/error.png
  loading_illustration_background_color: "000000"
  idle_illustration_background_color: "000000"
  listening_illustration_background_color: "000000"
  thinking_illustration_background_color: "000000"
  replying_illustration_background_color: "000000"
  error_illustration_background_color: "000000"

  # These unqiue characters have been extracted from every test file of every language available on https://github.com/home-assistant/intents (14 March 2024)
  allowed_characters: " !#%'()+,-./0123456789:;<>?@ABCDEFGHIJKLMNOPQRSTUVWYZ[]_abcdefghijklmnopqrstuvwxyz{|}°²³µ¿ÁÂÄÅÉÖÚßàáâãäåæçèéêëìíîðñòóôõöøùúûüýþāăąćčďĐđēėęěğĮįıļľŁłńňőřśšťũūůűųźŻżŽžơưșțΆΈΌΐΑΒΓΔΕΖΗΘΚΜΝΠΡΣΤΥΦάέήίαβγδεζηθικλμνξοπρςστυφχψωϊόύώАБВГДЕЖЗИКЛМНОПРСТУХЦЧШЪЭЮЯабвгдежзийклмнопрстуфхцчшщъыьэюяёђєіїјљњћאבגדהוזחטיכלםמןנסעפץצקרשת،ءآأإئابةتجحخدذرزسشصضطظعغفقكلمنهوىيٹپچڈکگںھہیےংকচতধনফবযরলশষস়ািু্చయలిెొ్ംഅആഇഈഉഎഓകഗങചജഞടഡണതദധനപഫബഭമയരറലളവശസഹാിീുൂെേൈ്ൺൻർൽൾაბგდევზთილმნოპრსტუფქყშჩცძჭხạảấầẩậắặẹẽếềểệỉịọỏốồổỗộớờởợụủứừửữựỳ—、一上不个中为主乾了些亮人任低佔何作供依侧係個側偵充光入全关冇冷几切到制前動區卧厅厨及口另右吊后吗启吸呀咗哪唔問啟嗎嘅嘛器圍在场執場外多大始安定客室家密寵对將小少左已帘常幫幾库度庫廊廚廳开式後恆感態成我戲戶户房所扇手打执把拔换掉控插摄整斯新明是景暗更最會有未本模機檯櫃欄次正氏水沒没洗活派温測源溫漏潮激濕灯為無煙照熱燈燥物狀玄现現瓦用發的盞目着睡私空窗立笛管節簾籬紅線红罐置聚聲脚腦腳臥色节著行衣解設調請謝警设调走路車车运連遊運過道邊部都量鎖锁門閂閉開關门闭除隱離電震霧面音頂題顏颜風风食餅餵가간감갔강개거게겨결경고공과관그금급기길깥꺼껐꼽나난내네놀누는능니다닫담대더데도동됐되된됨둡드든등디때떤뜨라래러렇렌려로료른를리림링마많명몇모무문물뭐바밝방배변보부불블빨뽑사산상색서설성세센션소쇼수스습시신실싱아안않알았애야어얼업없었에여연열옆오온완외왼요운움워원위으은을음의이인일임입있작잠장재전절정제져조족종주줄중줘지직진짐쪽차창천최추출충치침커컴켜켰쿠크키탁탄태탬터텔통트튼티파팬퍼폰표퓨플핑한함해했행혀현화활후휴힘，？"

external_components:
[... 770 lines cut ...]
          }
      - id: error_page
        lambda: |-
          it.fill(id(error_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(ollama_error), ImageAlign::CENTER);
      - id: no_ha_page
        lambda: |-
          it.image((it.get_width() / 2), (it.get_height() / 2), id(error_no_ha), ImageAlign::CENTER);
          // it.printf(10, 10, id(font_request), Color::WHITE, "IP: %s", id(wifi_info).state.c_str());

      - id: no_wifi_page
        lambda: |-
          it.image((it.get_width() / 2), (it.get_height() / 2), id(error_no_wifi), ImageAlign::CENTER);
      - id: initializing_page
        lambda: |-
          it.fill(id(loading_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(ollama_initializing), ImageAlign::CENTER);
      - id: muted_page
        lambda: |-
          it.fill(Color::BLACK);
          // if (id(display_conversation).state) {
          //   it.printf(10, 10, id(font_request), Color::WHITE, "Muted");
          //   it.printf(10, 30, id(font_request), Color::WHITE, "Press the button to talk");
          // }
```

Joakim-T, 2024-06-15:
> Sorry, way too many Github notifications to deal with.
> 
> Here's my esp-s3-box-3 esphome config if it helps, it gives me 3 volume settings in HA that I can select from.
> 
> <img alt="image" width="355" src="https://private-user-images.githubusercontent.com/862951/338434540-c8dcabdb-9324-481b-a64e-86b1ce2e9714.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg0Mzk0NjYsIm5iZiI6MTcxODQzOTE2NiwicGF0aCI6Ii84NjI5NTEvMzM4NDM0NTQwLWM4ZGNhYmRiLTkzMjQtNDgxYi1hNjRlLTg2YjFjZTJlOTcxNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYxNVQwODEyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NzdkODA2ZDFjMjE3ZTAzMmU3YTU2N2UxMWYyZTQxN2NkY2EwNmYwYjdlN2M0ZWVhMDgxN2NkNThkYmJkNGU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qWEttJoCGp1Dr9KiKYpKfQOMFvbv5tdtP3W5yDAiVFs">
> ```yaml
> ---
> #------------------------------------------------------------------------------------#
> #     PIN Schematics                                                                 #
> #                                                                                    #
> #       GPIO-00 MCU-BOOT                                                             #
> #       GPIO-01 Speaker Mute-Status                                                  #
> #       GPIO-02 I2S MCLK                                                             #
> #       GPIO-03 Touch-Screen TT21100 Interrupt Pin                                   #
> #       GPIO-04 ILI92xxx Display DC-Pin (SPI: CLK-Pin)                               #
> #       GPIO-05 ILI92xxx Display CS-Pin (SPI: MOSI-Pin)                              #
> #       GPIO-06 ILI92xxx Display SDA                                                 #
> #       GPIO-07 ILI92xxx Display SCK                                                 #
> #       GPIO-17 I2S_SCLK                                                             #
> #       GPIO-40 I2C_SCL (Temp & Hum) -- SENSOR v1.1 [AHT30]                          #
> #       GPIO-41 I2C_SDA (Temp & Hum) -- SENSOR v1.1 [AHT30]                          #
> 
> substitutions:
>   name: esp32-s3-box-3-5acf94
>   friendly_name: "ESP32 S3 Box 3 5acf94"
> 
>   micro_wake_word_model: alexa
>   # micro_wake_word_model: /config/hey_mycroft.json
>   wake_word_engine_location: "On device"
>   voice_assist_idle_phase_id: "1"
>   voice_assist_listening_phase_id: "2"
>   voice_assist_thinking_phase_id: "3"
>   voice_assist_replying_phase_id: "4"
>   voice_assist_not_ready_phase_id: "10"
>   voice_assist_error_phase_id: "11"
>   voice_assist_muted_phase_id: "12"
> 
>   # Request text parameters
>   request_text_font_size: "14"
>   request_text_start_x: "50"
>   request_text_start_y: "25"
>   request_text_max_width: "200"
>   request_text_line_height: "20"
>   request_text_max_lines: "2"
> 
>   # Response text parameters
>   response_text_font_size: "16"
>   response_text_start_x: "5"
>   response_text_start_y: "160"
>   response_text_max_line_length: "280"
>   response_text_line_height: "20"
>   response_text_max_lines: "4"
> 
>   # 320 x 240
>   loading_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/loading.png
>   idle_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/idle.png
>   listening_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/listening.png
>   thinking_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/thinking.png
>   replying_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/replying.png
>   error_illustration_file: https://github.com/sammcj/home-assistant-s3-box-community-illustrations/raw/main/sammcj/illustrations/error.png
>   loading_illustration_background_color: "000000"
>   idle_illustration_background_color: "000000"
>   listening_illustration_background_color: "000000"
>   thinking_illustration_background_color: "000000"
>   replying_illustration_background_color: "000000"
>   error_illustration_background_color: "000000"
> 
>   # These unqiue characters have been extracted from every test file of every language available on https://github.com/home-assistant/intents (14 March 2024)
>   allowed_characters: " !#%'()+,-./0123456789:;<>?@ABCDEFGHIJKLMNOPQRSTUVWYZ[]_abcdefghijklmnopqrstuvwxyz{|}°²³µ¿ÁÂÄÅÉÖÚßàáâãäåæçèéêëìíîðñòóôõöøùúûüýþāăąćčďĐđēėęěğĮįıļľŁłńňőřśšťũūůűųźŻżŽžơưșțΆΈΌΐΑΒΓΔΕΖΗΘΚΜΝΠΡΣΤΥΦάέήίαβγδεζηθικλμνξοπρςστυφχψωϊόύώАБВГДЕЖЗИКЛМНОПРСТУХЦЧШЪЭЮЯабвгдежзийклмнопрстуфхцчшщъыьэюяёђєіїјљњћאבגדהוזחטיכלםמןנסעפץצקרשת،ءآأإئابةتجحخدذرزسشصضطظعغفقكلمنهوىيٹپچڈکگںھہیےংকচতধনফবযরলশষস়ািু্చయలిెొ్ംഅആഇഈഉഎഓകഗങചജഞടഡണതദധനപഫബഭമയരറലളവശസഹാിീുൂെേൈ്ൺൻർൽൾაბგდევზთილმნოპრსტუფქყშჩცძჭხạảấầẩậắặẹẽếềểệỉịọỏốồổỗộớờởợụủứừửữựỳ—、一上不个中为主乾了些亮人任低佔何作供依侧係個側偵充光入全关冇冷几切到制前動區卧厅厨及口另右吊后吗启吸呀咗哪唔問啟嗎嘅嘛器圍在场執場外多大始安定客室家密寵对將小少左已帘常幫幾库度庫廊廚廳开式後恆感態成我戲戶户房所扇手打执把拔换掉控插摄整斯新明是景暗更最會有未本模機檯櫃欄次正氏水沒没洗活派温測源溫漏潮激濕灯為無煙照熱燈燥物狀玄现現瓦用發的盞目着睡私空窗立笛管節簾籬紅線红罐置聚聲脚腦腳臥色节著行衣解設調請謝警设调走路車车运連遊運過道邊部都量鎖锁門閂閉開關门闭除隱離電震霧面音頂題顏颜風风食餅餵가간감갔강개거게겨결경고공과관그금급기길깥꺼껐꼽나난내네놀누는능니다닫담대더데도동됐되된됨둡드든등디때떤뜨라래러렇렌려로료른를리림링마많명몇모무문물뭐바밝방배변보부불블빨뽑사산상색서설성세센션소쇼수스습시신실싱 [... cut ...]
>           it.fill(Color::BLACK);
>           // if (id(display_conversation).state) {
>           //   it.printf(10, 10, id(font_request), Color::WHITE, "Muted");
>           //   it.printf(10, 30, id(font_request), Color::WHITE, "Press the button to talk");
>           // }
> ```

This worked great for me but i would have liked it to be even louder. But perhaps too much to ask from a device this small.
Also i wish the mic would be a bit more sensitive, very common that it does not pickup on the wake word "hey jarvis".

qJake, 2024-08-04:
Still having this issue, tried the ESP Volume custom component but it isn't compiling for me right now, opened an issue here to track it: https://github.com/sammcj/esphome-esp-s3-box-3-volume/issues/2

Anyone else found *any other way at all* to control the volume on these things? I can't believe it's been months and we still only have one way to do it...

closed 2025-05-12 by github-actions[bot]