esphome/firmware #148: M5Stack Atom Echo compilation error

KixMan28, 2024-01-21:
Hi!

I have an M5Stack Atom Echo on which I try to run the following YAML code:

```
substitutions:
  name: "atom-echo"
  friendly_name: "Atom Echo"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: false
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"
  min_version: 2023.11.1
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

logger:

api:

ota:
  safe_mode: true
  num_attempts: 5
  password: !secret ota_password

web_server:

improv_serial:

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_pwd
  manual_ip:
    static_ip: 192.168.1.139
    gateway: !secret esphome_gateway
    subnet: !secret esphome_subnet
    dns1: !secret esphome_dns1
    dns2: !secret esphome_dns2
  fast_connect: true

text_sensor:
  - platform: wifi_info
    mac_address:
      name:  "${friendly_name} Mac Address"

i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

microphone:
[... 222 lines cut ...]
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1326:
    project_codemodel = get_cmake_code_model(
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 222:
    codemodel = json.load(fp)
  File "/usr/lib/python3.11/json/__init__.py", line 293:
    return loads(fp.read(),
  File "/usr/lib/python3.11/json/__init__.py", line 346:
    return _default_decoder.decode(s)
  File "/usr/lib/python3.11/json/decoder.py", line 337:
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.11/json/decoder.py", line 355:
    raise JSONDecodeError("Expecting value", s, err.value) from None
========================== [FAILED] Took 6.36 seconds ==========================
```

What is causing this?

oleander, 2024-01-22:
Skipping the last commit seems to work: (`@main^1`)

```yaml
dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main^1
```

KixMan28, 2024-01-22:
@oleander Hi! Thank you for the hint, but I still get the same error message. Any other hint?

oleander, 2024-01-28:
Try this (don't forget to replace the missing fields):

```yaml
substitutions:
  name: <name>
  friendly_name: <friendly_name>

packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: m5stack.atom-echo-voice-assistant
    version: recommended

esp32:
  board: m5stack-atom
  framework:
    platform_version: 5.4.0
    type: esp-idf
    version: recommended

logger:
  level: DEBUG

api:
  encryption:
    key: <key>

wifi:
  ssid: <name>
  password: <password>

  ap:
    ssid: <name>
    password: <password>

captive_portal:
ota:
```

KixMan28, 2024-01-29:
@oleander Thank you, but still the same result:

Here is my new program:

```
substitutions:
  name: "atom-echo"
  friendly_name: "Atom Echo"

packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main 

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: false
  project:
    name: m5stack.atom-echo-voice-assistant
    version: recomended
  min_version: 2023.11.1
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: m5stack-atom
  framework:
    platform_version: 5.4.0
    type: esp-idf
    version: recommended

logger:
  level: DEBUG

api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxWow1kkKVursgAe0eNYC5sAlGJ8g=

ota:
  safe_mode: true
  num_attempts: 5
  password: !secret ota_password

web_server:

captive_portal:

improv_serial:

esp32_improv:
  authorizer: none
  
wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_pwd
  manual_ip:
    static_ip: 192.168.1.139
    gateway: !secret esphome_gateway
    subnet: !secret esphome_subnet
    dns1: !secret esphome_dns1
    dns2: !secret esphome_dns2
#  use_address: 192.168.1.67
  fast_connect: true
[... 134 lines cut ...]
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1326:
    project_codemodel = get_cmake_code_model(
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 222:
    codemodel = json.load(fp)
  File "/usr/lib/python3.11/json/__init__.py", line 293:
    return loads(fp.read(),
  File "/usr/lib/python3.11/json/__init__.py", line 346:
    return _default_decoder.decode(s)
  File "/usr/lib/python3.11/json/decoder.py", line 337:
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.11/json/decoder.py", line 355:
    raise JSONDecodeError("Expecting value", s, err.value) from None
========================= [FAILED] Took 38.85 seconds =========================
```

By the way, at the first run of the installation with the new program, during compilation, my HA  restarted automatically.
The error message above is the result of the 2nd installation run.

Any other solution?

oleander, 2024-02-01:
It happens, It seems, when the esp idf package for platformio doesn't get along with the underlying version of esp idf. Play around with the numbers of a bit.

KixMan28, 2024-02-01:
With which numbers I should play?

KixMan28, 2024-02-01:
> It happens, It seems, when the esp idf package for platformio doesn't get along with the underlying version of esp idf. Play around with the numbers of a bit. 

Hi @oleander !

Finally I succeeded to install my firmware. I have changed the platform from 5.4.0 to 5.3.0. When I tried to install the firmare I received several errors. Then I switched back to 5.4.0 and it worked.

Only one small warning at the and of compilation, see below:

'''
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: missing --end-group; added as last command line option
'''

What is causing this?

closed 2024-03-04 by KixMan28