esphome/firmware #181: Alexa Speaker Output

jtwhitlow, 2024-03-21:
I have been trying to send my voice assistant output to my alexa echo after the m5stack echo takes in the wake word and my gpt processes it.  I am new to esphome and i have been able to get it to work on my google speaker, but was wondering what I am doing wrong on my alexa speaker.  I have tried several variations on passing to alexa.  My "testing" message works, but I cant get the return x; to pass on.  Thanks in advance to anyone that can help.

#this is what works for google
 #       service: media_player.play_media
 #       data:
 #         entity_id: media_player.basement_echo
 #         media_content_id: !lambda 'return x;'
 #         media_content_type: music
 #         announce: "true"

### This is the testing message that works, but I havent been able to make that dynamic using the !lambda return function
- homeassistant.service:
        service: notify.alexa_media
        data:
          target: media_player.basement_echo
          message: "testing"
        data_template:
          data: |
            {{{"type": "tts"}}}

danktankk, 2024-03-25:
I have been trying the same thing for about 2 weeks.  Im surprised more people havent chimed in on this.   You are further along than I am.

sajov, 2024-04-10:
Hi there, not sure if it helps, but in this video it was done for a google device
https://youtu.be/o3yZWD_sFIE?si=yZcuiIifcfnqSrru

jtwhitlow, 2024-04-10:
Ya I went through that video originally... it works for google speakers but
couldnt get it to work with the alexa media player (amazon echo).

danktankk, 2024-04-16:
yep same..  a real bummer too, I liked these little atoms..  they arent even big enough to be a paperweight.  so in the bin they go

cl0ud6uru, 2024-05-10:
This is what is working for me on my VLC Telnet speaker. 
Need to use Media player, and not notify service. Just make sure your Alexa is a media player. 

I also commented out a few of the speaker things to stop the speaker from playing on the S3-Box-3 device

`microphone:
  - platform: esp_adf
    id: box_mic

# speaker: !remove
#   # - id: !remove box_speaker

voice_assistant:
  id: va
  microphone: box_mic
  #speaker: box_speaker
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3
  on_listening:
    - lambda: id(voice_assistant_phase) = ${voice_assist_listening_phase_id};
    - text_sensor.template.publish:
        id: text_request
        state: "..."
    - text_sensor.template.publish:
        id: text_response
        state: "..."
    - script.execute: draw_display
  on_stt_vad_end:
    - lambda: id(voice_assistant_phase) = ${voice_assist_thinking_phase_id};
    - script.execute: draw_display
  on_stt_end:
    - text_sensor.template.publish:
        id: text_request
        state: !lambda return x;
    - script.execute: draw_display
  on_tts_start:
    - text_sensor.template.publish:
        id: text_response
        state: !lambda return x;
    - lambda: id(voice_assistant_phase) = ${voice_assist_replying_phase_id};
    - script.execute: draw_display

  on_tts_end:
    - homeassistant.service:
        service: media_player.play_media
        data:
          entity_id: media_player.vlc_telnet
          media_content_id: !lambda 'return x;'
          media_content_type: music
          announce: "true"
    - delay: 5s  # Adjust the delay as needed                 
    - lambda: id(voice_assistant_phase) = ${voice_assist_idle_phase_id}; 
    - script.execute: draw_display

  # on_tts_stream_start: !remove

  # on_tts_stream_end: !remove

  on_error:
    - if:
        condition:
          lambda: return !id(init_in_progress);
        then:
          - lambda: id(voice_assistant_phase) = ${voice_assist_error_phase_id};
          - script.execute: draw_display
          - delay: 1s
          - if:
              condition:
                switch.is_off: mute
              then:
                - lambda: id(voice_assistant_phase) = ${voice_assist_idle_phase_id};
              else:
                - lambda: id(voice_assistant_phase) = ${voice_assist_muted_phase_id};
          - script.execute: draw_display
  on_client_connected:
    - if:
        condition:
          switch.is_off: mute
        then:
          - wait_until:
              not: ble.enabled
          - voice_assistant.start_continuous:
          - lambda: id(voice_assistant_phase) = ${voice_assist_idle_phase_id};
        else:
          - lambda: id(voice_assistant_phase) = ${voice_assist_muted_phase_id};
    - lambda: id(init_in_progress) = false;
    - script.execute: draw_display
  on_client_disconnected:
    - lambda: id(voice_assistant_phase) = ${voice_assist_not_ready_phase_id};
    - script.execute: draw_display
  `

danktankk, 2024-06-01:
> This is what is working for me on my VLC Telnet speaker. Need to use Media player, and not notify service. Just make sure your Alexa is a media player.
>
> I also commented out a few of the speaker [...]

this isnt for the amazon echo, which is by far the most popular speaker owned.  I am hoping that someone has figured it out.  If it works for the google brew of smart speakers, then I would hope it would work for the amazon echos

cl0ud6uru, 2024-06-05:
> Thanks, but this isnt for the amazon echo, which is by far the most popular speaker owned, and this also doesnt work for the echo either.   I am hoping that someone has figured it out though.  If it works for the google brew of smart speakers, then I would hope it would work for the amazon echos

Did you try using the Alexa? If shows as a media player in home assistant. You should just be able to change the config I posted to use your Alexa device speaker. As long as it shows as a media player entity in home assistant. Unfortunately, I don't have one to test with.

john500y2k, 2024-07-12:
Have been struggling with this a lot too but finally had success!

```
  on_tts_start:
    - homeassistant.service:
        service: notify.alexa_media_your_echo_device
        data:
          message: !lambda return x;
```

Even I am looking at it now and saying, that's it... that's all that was needed...

I have seen references where they place this piece within `on_tts_end` however I opted for start as thought it would save a tiny bit of time not waiting for the tt_end to then send the alexa notification.

Really hope this helps everyone out.

danktankk, 2024-07-12:
> Have been struggling with this a lot too but finally had success!
> 
> ```
>   on_tts_start:
>     - homeassistant.service:
>         service: notify.alexa_media_your_echo_device
>         data:
>           message: !lambda return x;
> ```
> 
> Even I am looking at it now and saying, that's it... that's all that was needed...
> 
> I have seen references where they place this piece within `on_tts_end` however I opted for start as thought it would save a tiny bit of time not waiting for the `on_tts_start` to finish to then send the Alexa notification when `on_tts_end` is triggered.
> 
> Really hope this helps everyone out.

could you provide the full `on_tts_start:` and `on_tts_end:`?

i am going to give this a whirl as soon as humanly possible

closed 2025-05-11 by github-actions[bot]