esphome/firmware #182: fatal error when compiling

mph070770, 2024-03-23:
I've used the instructions from here:

`https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/`

and get the following failure.  I've tried cleaning build files and it fails in the same way.  What is wrong?

```
INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/m5stack-atom-echo-109354.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing m5stack-atom-echo-109354 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40406.240122 (4.4.6) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/build/m5stack-atom-echo-109354/.pioenvs/m5stack-atom-echo-109354/CMakeFiles/CMakeOutput.log".
See also "/data/build/m5stack-atom-echo-109354/.pioenvs/m5stack-atom-echo-109354/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/m5stack-atom-echo-109354/.pioenvs/m5stack-atom-echo-109354/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_28285 && [1/2] Building C object CMakeFiles/cmTC_28285.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_28285
    FAILED: cmTC_28285 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_28285.dir/testCCompiler.c.obj  -o cmTC_28285   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:381 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 11.52 seconds =========================
```

mph070770, 2024-03-27:
No dev's have any comment on this?  Still broken for me and I'm just following the instructions.

GrumpySlav, 2025-01-09:
I've got the same with ESP32-S3-BOX-3:
```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-web-2e0310.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing esp32-wake1 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esp32-wake1/.pioenvs/esp32-wake1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.75 seconds ==========================
```

GrumpySlav, 2025-01-13:
I've got solution. Add this into esp32 section, line 55:
```
framework:
    type: esp-idf 
    version: 4.4.8
    platform_version: 5.4.0
```

bazcurtis, 2025-03-27:
I hit another error have using the above. 

src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope

In the end I used this new firmware - [New Firmware](https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box-3/esp32-s3-box-3.yaml)

jesserockz, 2025-04-17:
This is the wrong repository for this issue and it has been solved a long time ago

closed 2025-04-17 by jesserockz