esphome/firmware #193: Simplified Chinese or Traditional Chinese and pronunciation cannot be displayed correctly

zminde, 2024-05-18:
Thanks a lot for this profile, it works smoothly on box3 to control smart home devices using voice. After a few days of testing, it runs very stably, but this only happens when using the English voice agent. When using Simplified Chinese or Traditional Chinese as tts and stt, the Chinese characters cannot be displayed correctly on the box3 screen, and there is no Chinese voice feedback. I see that there are some Chinese character sets in the config file, but I don't know if the problem is directly related to the character set, it would be great if shei has tips for using Simplified Chinese or Traditional Chinese agents.

kyutov, 2024-05-20:
This comportment is caused by the font, which is used in the firmware. I have had exactly the same issue with the Cyrillic characters #185. I have created a PR #188 , which will allow you to change the default font of the firmware to Simplified Chinese or Traditional Chinese, but the PR is still waiting for approval ...

zminde, 2024-05-20:
> This comportment is caused by the font, which is used in the firmware. I had exactly the same issue with the Cyrillic characters #185. I have created a PR #188 , which will allow you to change the default font of the firmware to Simplified Chinese or Traditional Chinese, but the PR is still waiting for approval ...

Yes, the problem is indeed caused by missing fonts. Thank you very much for your work on resolving this issue. Hopefully this will be approved soon or the project owner will add wider font support.

jesserockz, 2024-06-18:
You can override the fonts in your adopted yaml using `!extend`
Example:
```yaml
font:
  - id: !extend font_request
    file:
      family: Roboto
```

You are not even restricted to gfonts, you can completely override the `file` variable to whatever you want as long as it is valid

closed 2025-05-10 by github-actions[bot]