esphome/firmware #194: ESP32 S3 Box and Box 3 eventually stop responding to wake words

demartinoaj, 2024-05-18:
Hello,

I'm having an issue with both the ESP32 S3 Box and Box 3 firmware where after a few hours they will eventually stop responding to wake words. This issue occurs when using both local wake word detection on the esp32 box or home assistant wake word detection, so I assume its an issue with the box. It seems to be resolved by toggling the box mute in home assistant, but then will inevitable break again in a few hours. I've managed to capture the error in the log below, which I've attached. Has anyone encountered this before?

[logs_esp32-s3-box-3_no_wake.txt](https://github.com/esphome/firmware/files/15366610/logs_esp32-s3-box-3_no_wake.txt)

simonjowett, 2024-06-28:
Same Box 3 hardware and I am experiencing wake word issues similar to your own however, mine seems to lock up after only 1 or 2 responses.  When it does respond, the screen stays on showing the text response and does not revert to the 'grey house'.  If the mic is muted and unmuted the screen does return so the ESP is not 'locked up', it's still online and responding to esphome config changes.   Pretty much useless at this point. 

[logs_esp32-s3-box-3-059f1c_logs.txt](https://github.com/user-attachments/files/16024235/logs_esp32-s3-box-3-059f1c_logs.txt)

lstull, 2024-09-22:
> Same Box 3 hardware and I am experiencing wake word issues similar to your own however, mine seems to lock up after only 1 or 2 responses. When it does respond, the screen stays on showing the text [...]

Did you ever get this worked out I am having the exact same thing.

simonjowett, 2024-09-22:
> Did you ever get this worked out I am having the exact same thing.

Not really,  I have sort of set this project/experiment aside as it was too frustrating and moved on to other things.  I have however got it on the bench in the kitchen and I have been updating it on ESPhome as the usual updates roll through.  I occasionally challenge it and it does seem a bit better but not fully useful at this stage (not mature enough tech software wise possibly)?

lstull, 2024-09-22:
Thanks for the reply. Yeah I hear you about frustrating immaturity

simonjowett, 2024-09-22:
The hardware is also a bit interesting - I have the green power light that flickers at times.  Also how bright is that green LED - I might have to dig in and change to a much bigger resistor!!

closed 2025-05-10 by github-actions[bot]