esphome/firmware #195: ESP32-S3-BOX-3 homeassistant voice-assist No-Wifi

huuub, 2024-05-22:
I bought 2 devices and following ESP32-S3-BOX voice assistant -[ Home Assistant (home-assistant.io)](https://www.home-assistant.io/voice_control/s3_box_voice_assistant/) the step to add the wifi never comes. I tried :

SYMPTOM
The installation wizard never shows the dialog to connect to the Wi-Fi, but directly returns to the screen with Install Voice Assistant.

REMEDY
Disconnect the USB cable connecting the ESP32-S3-BOX-3 and connect it again.

this 20 times but nothing changes. Resetting etc etc...

So unless someone knows a solution is there an alternative way to install this? Copy the [yamlfile ](https://github.com/esphome/firmware/blob/main/voice-assistant/esp32-s3-box-3.yaml)or something? I cant find a good tutorial for that?
I would appreciate some help...

bruberg, 2024-05-22:
Same here, the unit enters a reboot loop. It seems to give up after a while and set up a local AP instead.

```
[C][wifi:038]: Setting up WiFi...
[D][esp-idf:000]: I (374) wifi:
[D][esp-idf:000]: wifi driver task: 3ffd9650, prio:23, stack:6656, core=0
[D][esp-idf:000]: 

[D][esp-idf:000][wifi]: I (1605) system_api: Base MAC address is not set

[D][esp-idf:000][wifi]: I (1615) system_api: read default base MAC address from EFUSE

[D][esp-idf:000][wifi]: I (1628) wifi:
[D][esp-idf:000][wifi]: wifi firmware version: 1fd20f4
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1645) wifi:
[D][esp-idf:000][wifi]: wifi certification version: v7.0
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1656) wifi:
[D][esp-idf:000][wifi]: config NVS flash: enabled
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1676) wifi:
[D][esp-idf:000][wifi]: config nano formating: disabled
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1687) wifi:
[D][esp-idf:000][wifi]: Init data frame dynamic rx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1707) wifi:
[D][esp-idf:000][wifi]: Init static rx mgmt buffer num: 5
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1727) wifi:
[D][esp-idf:000][wifi]: Init management short buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1738) wifi:
[D][esp-idf:000][wifi]: Init dynamic tx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1758) wifi:
[D][esp-idf:000][wifi]: Init static rx buffer size: 1600
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1779) wifi:
[D][esp-idf:000][wifi]: Init static rx buffer num: 10
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1799) wifi:
[D][esp-idf:000][wifi]: Init dynamic rx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000]: I (1810) wifi_init: rx ba win: 6

[D][esp-idf:000]: I (1820) wifi_init: tcpip mbox: 32

[D][esp-idf:000]: I (1830) wifi_init: udp mbox: 6

[D][esp-idf:000]: I (1830) wifi_init: tcp mbox: 6

[D][esp-idf:000]: I (1840) wifi_init: tcp tx win: 5760

[D][esp-idf:000]: I (1840) wifi_init: tcp rx win: 5760

[D][esp-idf:000]: I (1850) wifi_init: tcp mss: 1440

[D][esp-idf:000]: I (1861) wifi_init: WiFi IRAM OP enabled

[D][esp-idf:000]: I (1861) wifi_init: WiFi RX IRAM OP enabled

[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: 64:B7:08:A0:78:98
[D][esp-idf:000][wifi]: I (1883) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06

[D][esp-idf:000][wifi]: I (1966) wifi:
[D][esp-idf:000][wifi]: mode : softAP (64:b7:08:a0:78:99)
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1968) wifi:
[D][esp-idf:000][wifi]: Total power save buffer number: 16
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1972) wifi:
[D][esp-idf:000][wifi]: Init max length of beacon: 752/752
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1992) wifi:
[D][esp-idf:000][wifi]: Init max length of beacon: 752/752
[D][esp-idf:000][wifi]: 

[C][wifi:253]: Setting up AP...
[C][wifi:255]:   AP SSID: 'm5stack-atom-echo-a07898'
[C][wifi:256]:   AP Password: ''
[D][esp-idf:000][wifi]: I (2025) wifi:
[D][esp-idf:000][wifi]: Total power save buffer number: 16
[D][esp-idf:000][wifi]: 

[C][wifi:265]:   IP Address: 192.168.4.1
[D][esp-idf:000][wifi]: I (2046) wifi:
[D][esp-idf:000][wifi]: mode : sta (64:b7:08:a0:78:98) + softAP (64:b7:08:a0:78:99)
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (2064) wifi:
[D][esp-idf:000][wifi]: enable tsf
[D][esp-idf:000][wifi]: 
```

I suspect this is related to the [many reported issues with ESPHome 2024.5.x](https://github.com/esphome/issues/issues), since the unit gets installed with 2024.5.0:

```
I (739) cpu_start: Project name:     m5stack-atom-echo
I (745) cpu_start: App version:      2024.5.0
I (750) cpu_start: Compile time:     May 20 2024 05:18:45
```

olo81, 2024-05-23:
the same occurs on Atom

D3SOX, 2024-05-23:
I needed to [pin esp-idf to v4.4.6](https://community.home-assistant.io/t/esp32-s3-box3/638287/345?u=d3sox). Then the microphone didn't work.
I’m unsure what change fixed that but I think the revert to ESPHome 2024.4.x. I now have a working versionwith on-device hot word working, mic, Wi-Fi, everything working. [See my repository](https://github.com/D3SOX/firmware)

Flashed via web esp home on the first click on Configure Wi-Fi it might give an error but try again it always worked the second time for me.

davidrpfarinha, 2024-05-23:
> the same occurs on Atom

I was going to reply the same thing. I tried to manually installed the YAML with ESPHome 2024.5.2 but it failed as well.

pwsh, 2024-06-06:
This may be fixed with the latest 2024.5.5 release. At least it is fixed for me after a month of trying workarounds.

Specifically [https://github.com/esphome/esphome/pull/6855](https://github.com/esphome/esphome/pull/6855)

OlwinFroon, 2024-06-09:
Same problem with a new S3 Box 3 here, as told in 1st post.

After the first WiFi fail, I tried manual install, which ended up with the infamous "See what went wrong".

The following installation dump went on for a while and then the UI lost connection, because my HA host rebooted without further notice.

The second attempt also ended up with my machine rebooting.
Scary!

I will not touch this thing again any time soon.

pwsh, 2024-06-09:
It is working on my esp32-s3-box3 with a manual IP address assignment. It wasn't working before.

I did clean build files before this. I'll grab my config file and post it here in a bit.

OlwinFroon, 2024-06-11:
Meh. That S3 Box sitting on my desk was burning a hole in my eyes, so I installed it again with the prefab image provided [HERE](https://www.home-assistant.io/voice_control/s3_box_voice_assistant/).

Instead of the promised WiFi dialog, I ended up with the box showing a sad red house and a "No WiFi" message 

So I tried [Improv Wi-Fi](https://www.improv-wifi.com/) to set the WiFi credentials via BLE, and it worked.

Then I had a box showing a sad red house saying "HA not found", indicating that it shoudl be adpoted in HA.

Lo and behold: it actually showed up in HA ESPhome dashboard.

Unfortunately, adopting requires recompiling. I gave it another try, same result: my HA host rebooted.

The reason might be equipment/performance related, perhaps my ODroid is just to weak for this kind of work.

So I ended up installing Python and ESPHome on my Windows PC, which should be a lot more powerful than the little ODroid.

Didn't work either, just another [obstacle](https://github.com/esphome/issues/issues/5896)...

pwsh, 2024-06-13:
Here's my ESP32-S3-BOX3 config that is working for voice assistant
```

---
substitutions:
  name: esp32-s3-box-3
  friendly_name: ESP32 S3 Box 3
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png
  micro_wake_word_model: hey_jarvis

  loading_illustration_background_color: "000000"
  idle_illustration_background_color: "000000"
  listening_illustration_background_color: "FFFFFF"
  thinking_illustration_background_color: "FFFFFF"
  replying_illustration_background_color: "FFFFFF"
  error_illustration_background_color: "000000"

  voice_assist_idle_phase_id: "1"
  voice_assist_listening_phase_id: "2"
  voice_assist_thinking_phase_id: "3"
  voice_assist_replying_phase_id: "4"
  voice_assist_not_ready_phase_id: "10"
  voice_assist_error_phase_id: "11"
  voice_assist_muted_phase_id: "12"

  # These unqiue characters have been extracted from every test file of every language available on https://github.com/home-assistant/intents (14 March 2024)
  allowed_characters: " !#%'()+,-./0123456789:;<>?@ABCDEFGHIJKLMNOPQRSTUVWYZ[]_abcdefghijklmnopqrstuvwxyz{|}°²³µ¿ÁÂÄÅÉÖÚßàáâãäåæçèéêëìíîðñòóôõöøùúûüýþāăąćčďĐđēėęěğĮįıļľŁłńňőřśšťũūůűųźŻżŽžơưșțΆΈΌΐΑΒΓΔΕΖΗΘΚΜΝΠΡΣΤΥΦάέήίαβγδεζηθικλμνξοπρςστυφχψωϊόύώАБВГДЕЖЗИКЛМНОПРСТУХЦЧШЪЭЮЯабвгдежзийклмнопрстуфхцчшщъыьэюяёђєіїјљњћאבגדהוזחטיכלםמןנסעפץצקרשת،ءآأإئابةتجحخدذرزسشصضطظعغفقكلمنهوىيٹپچڈکگںھہیےংকচতধনফবযরলশষস়ািু্చయలిెొ్ംഅആഇഈഉഎഓകഗങചജഞടഡണതദധനപഫബഭമയരറലളവശസഹാിീുൂെേൈ്ൺൻർൽൾაბგდევზთილმნოპრსტუფქყშჩცძჭხạảấầẩậắặẹẽếềểệỉịọỏốồổỗộớờởợụủứừửữựỳ—、一上不个中为主乾了些亮人任低佔何作供依侧係個側偵充光入全关冇冷几切到制前動區卧厅厨及口另右吊后吗启吸呀咗哪唔問啟嗎嘅嘛器圍在场執場外多大始安定客室家密寵对將小少左已帘常幫幾库度庫廊廚廳开式後恆感態成我戲戶户房所扇手打执把拔换掉控插摄整斯新明是景暗更最會有未本模機檯櫃欄次正氏水沒没洗活派温測源溫漏潮激濕灯為無煙照熱燈燥物狀玄现現瓦用發的盞目着睡私空窗立笛管節簾籬紅線红罐置聚聲脚腦腳臥色节著行衣解設調請謝警设调走路車车运連遊運過道邊部都量鎖锁門閂閉開關门闭除隱離電震霧面音頂題顏颜風风食餅餵가간감갔강개거게겨결경고공과관그금급기길깥꺼껐꼽나난내네놀누는능니다닫담대더데도동됐되된됨둡드든등디때떤뜨라래러렇렌려로료른를리림링마많명몇모무문물뭐바밝방배변보부불블빨뽑사산상색서설성세센션소쇼수스습시신실싱아안않알았애야어얼업없었에여연열옆오온완외왼요운움워원위으은을음의이인일임입있작잠장재전절정제져조족종주줄중줘지직진짐쪽차창천최추출충치침커컴켜켰쿠크키탁탄태탬터텔통트튼티파팬퍼폰표퓨플핑한함해했행혀현화활후휴힘，？"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "1.0"
  min_version: 2023.11.5
  on_boot:
    priority: 600
    then:
      - script.execute: draw_display
      - delay: 30s
      - if:
          condition:
            lambda: return id(init_in_progress);
          then:
            - lambda: id(init_in_progress) = false;
            - script.execute: draw_display

esp32:
  board: esp32s3box
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_AUDIO_BOARD_CUSTOM: "y"
      CONFIG_ESP32_S3_BOX_3_BOARD: "y"
    components:
      - name: esp32_s3_box_3_board
        source: github://jesserockz/esp32-s3-box-3-board@main
        refresh: 0s
[... 408 lines cut ...]
            it.filled_rectangle(20 , 20 , 280 , 30 , Color::WHITE );
            it.rectangle(20 , 20 , 280 , 30 , Color::BLACK );
            it.filled_rectangle(20 , 190 , 280 , 30 , Color::WHITE );
            it.rectangle(20 , 190 , 280 , 30 , Color::BLACK );
            it.printf(30, 25, id(font_request), Color::BLACK, "%s", id(text_request).state.c_str());
            it.printf(30, 195, id(font_response), Color::BLACK, "%s", id(text_response).state.c_str());
          }
      - id: error_page
        lambda: |-
          it.fill(id(error_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(casita_error), ImageAlign::CENTER);
      - id: no_ha_page
        lambda: |-
          it.image((it.get_width() / 2), (it.get_height() / 2), id(error_no_ha), ImageAlign::CENTER);
      - id: no_wifi_page
        lambda: |-
          it.image((it.get_width() / 2), (it.get_height() / 2), id(error_no_wifi), ImageAlign::CENTER);
      - id: initializing_page
        lambda: |-
          it.fill(id(loading_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(casita_initializing), ImageAlign::CENTER);
      - id: muted_page
        lambda: |-
          it.fill(Color::BLACK);
```

OlwinFroon, 2024-06-13:
Thanks!
Didn't work, though. Same as before.

The No-WiFi thing seems to boil down to a bug in Improv-WiFi.

The serial variant doesn't seem to work at all, but I can use the BT variant after flashing the prefab binary.

OlwinFroon, 2024-06-24:
I seem to have fixed the crash problem.

After adding `compile_process_limit: 1` to the `esphome:` section, I could successfully update my S3 Box 3.

The process took quite a while, but it worked.

That should go in the FAQ.

closed 2025-05-10 by github-actions[bot]