esphome/firmware #200: ESPhome update 2024.5.5 broke wakeword detection on M5 Atom Echo

crissmil, 2024-06-05:
After updating HA ESPHome add-on to latest 2024.5.5 followed by updating my 3 devices M5 Atom Echo, none of them could use wake work anymore. 
I did restart HA, restart Pi5, compile all M5 devices again without success. They just do not react at all to wake-word, even if the log says Pipeline running. Using Assist with my laptop bult-in microphone is reacting normally to wake-word.

I downgraded ESPhome back to 2024.5.4, recompile all my devices and my satellites are running again.

pwsh, 2024-06-06:
Did you manually pin an IDF version? The latest fix was for IDF 4.4.7, so I'm wondering if you had pinned to IDF 4.4.6 like several people had previously.

[https://github.com/esphome/esphome/pull/6855](https://github.com/esphome/esphome/pull/6855)

crissmil, 2024-06-06:
I did not manually pin any IDF version. 

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

Should I?

pwsh, 2024-06-07:
I wish I had an m5 to test with, so this is all guesswork based off issues I had with the esp32-s3-box-3

I'd say try one of these 2 configurations. I think recommended will pull in 4.4.7 per [https://esphome.io/components/esp32.html](https://esphome.io/components/esp32.html) and is the same as leaving blank for default, so no change from what you have. 

```
  framework:
    type: esp-idf
    version: recommended
```
    
  or
 
```
  framework:
    type: esp-idf
    version: 4.4.6
```

But it also could do with the change in ringbuffer size added to 2024.5.5
[https://github.com/esphome/esphome/pull/6830/commits/ba250521228e971ff221d1ddbe2559e0bf61e00c](https://github.com/esphome/esphome/pull/6830/commits/ba250521228e971ff221d1ddbe2559e0bf61e00c)

closed 2025-05-10 by github-actions[bot]