esphome/firmware #204: ESPHome 2024.06 broke GL.iNet GL-S10 OTA updates

denix0, 2024-06-19:
ESPHome 2024.06 has some OTA changes which are listed under the "Breaking changes" section.

After ESPHome upgrade, trying to update GL.iNet GL-S10 firmware fails with:

```
Failed config
At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality
```

I'm guessing the corresponding firmware yaml file here:
https://github.com/esphome/firmware/blob/main/bluetooth-proxy/gl-s10.yaml

needs updating to add "platform: esphome" under ota per this description on the corresponding PR:
https://github.com/esphome/esphome/pull/6459

randybb, 2024-06-19:
these configs are not updated yet (and examples in devices subdomain). If you have it imported in your dashboard, just change `ota:` part based on release notes.

denix0, 2024-06-19:
@randybb sorry, I'm still a noob when it comes to ESPHome... My device-specific yaml does not have any `ota:` sections, it just points to the main `gl-s10.yaml` here with this:

```
packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/gl-s10.yaml@main
```

Not entirely sure how to fix it locally - would appreciate any help here. Thanks.

randybb, 2024-06-19:
Then just wait couple of days with the update / installation - 2024.6 was just released, so this breaking change needs to find its way to all configs.

Unfortunately I never used packages, not sure if just by adding this part to your config it will override it
```yaml
ota:
  - platform: esphome
```

randybb, 2024-06-19:
will be done soon: https://github.com/esphome/firmware/pull/202

closed 2024-07-01 by denix0