esphome/firmware #206: ESP32 S3 Box can't be updated

ChristophCaina, 2024-06-20:
I've seen, that there's already an OTA fix available for the VoiceAssistant Firmware - but somehow, the S3-Box still can't be updated...

I am getting the following error:
[image]

WarC0zes, 2024-06-21:
Hi,
add in your code:
```
ota:
  - platform: esphome
```
despite declaring the package, it does not add the OTA configuration. I had to add it to the ESP code to be able to compile otherwise I got an error.

ChristophCaina, 2024-06-21:
I've tried this with 2024.6.0
Since 2024.6.1 this hast been added to the Firmware, but the Box ist still Not getting the Update. 

For all other devices, adding the Plattform was working.

ChristophCaina, 2024-06-21:
it seems that this is due to the "micro_wake_word_model" in the substitutions...
If I remove this from the configuration, compiling / uploading the new firmware is working as expected.

closed 2024-07-31 by ChristophCaina