esphome/firmware #216: Can't update S3 Box because of duplicate OTA instance

TheZoker, 2024-06-28:
Hi there,

This is my yaml config:
```yaml
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box.yaml@main

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "ESPHome Fallback"
    password: !secret api_password

ota:
  - platform: esphome
    password: !secret api_password
```

Every time I try to compile this, I get this error:
```
Failed config

ota.esphome: [source /data/packages/c46f54c1/wake-word-voice-assistant/esp32-s3-box.yaml:85]
  
  Only one instance of the esphome ota platform is allowed per port. Note that this error may result from OTA specified in packages.
  platform: esphome
  id: ota_esphome
  version: 2
  port: 3232
```

Since all my devices use a password for OTA I'm not sure how to solve this. Can somebody help me fix this? Thanks!

DeveshwarH1996, 2024-06-30:
I've had the same issue too, for some reason even when I change the hostname it seems to always add the id after the name. I had to use a static IP to get it to update or listen to the logs

jwillikers, 2024-07-17:
I've hit the same issue now too. Seems like there ought to be a way to add a password to an existing `esphome` `ota` instance, that way there isn't an `esphome` `ota` instance available without a password.

jwillikers, 2024-07-17:
I think the [extend](https://esphome.io/guides/configuration-types#extend) keyword can be used to modify the default `esphome` `ota` instance and add a password to it like so:

```
ota:
  - id: !extend ota_esphome
    password: !secret esp32_s3_box_3_01_ota_esphome_password
```

jwillikers, 2024-07-17:
I guess that may not even be necessary due to esphome/esphome#7001 being merged.

closed 2025-05-10 by github-actions[bot]