esphome/firmware #217: Cant update m5stack-atom-echo after add to HA

userosos, 2024-06-28:
INFO ESPHome 2024.6.4
INFO Reading configuration /config/esphome/m5stack-atom-echo-0dba90.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Cloning https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1066, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 484, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 194, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 344, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 297, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 684, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 67, in clone_or_update
    run_git_command(cmd)
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: fetch-pack: invalid index-pack output

michaelsage, 2024-07-01:
Did you manage to resolve this? I am having the same issue, but some slightly different line numbers

``---
esphome:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  min_version: 2024.6.0
  name_add_mac_suffix: true
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:
api:

ota:
  - platform: esphome
    id: ota_esphome

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

wifi:
  on_connect:
    - delay: 5s  # Gives time for improv results to be transmitted
    - ble.disable:
  on_disconnect:
    - ble.enable:
  ap:

improv_serial:

esp32_improv:
  authorizer: none

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    mode: mono

voice_assistant:
  id: va
  microphone: echo_microphone
  speaker: echo_speaker
  noise_suppression_level: 2
  auto_gain: 31dBFS
[... 199 lines cut ...]
    on_turn_off:
      - script.execute: reset_led
  - platform: template
    id: timer_ringing
    optimistic: true
    internal: true
    restore_mode: ALWAYS_OFF
    on_turn_on:
      - delay: 15min
      - switch.turn_off: timer_ringing

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s
  - source: github://jesserockz/esphome-components
    components: [file]
    refresh: 0s

esp_adf:

file:
  - id: timer_finished_wave_file
    file: https://github.com/esphome/firmware/raw/main/voice-assistant/sounds/timer_finished.wav``

macfr76, 2024-07-06:
Similar issue for me :

```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/m5stack-atom-echo-23be74.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1066, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 484, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 194, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 344, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 297, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 684, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 111, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: Unable to find current revision in submodule path 'components/esp-adf-libs'
```

asyikin1090, 2024-07-26:
similar issue with me:

INFO ESPHome 2024.7.2
INFO Reading configuration /config/esphome/m5stack-atom-echo-869a64.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Cloning https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1080, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1067, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 433, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 194, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 344, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 297, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 684, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 67, in clone_or_update
    run_git_command(cmd)
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: fetch-pack: invalid index-pack output

userosos, 2024-07-27:
After upgrade ESP it compile ok BUT

```
INFO ESPHome 2024.7.2
INFO Reading configuration /config/esphome/m5stack-atom-echo-0dba90.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:04:44][;2I(9 piag:sget4 adr01b98d=0854 size=19a70h (105072) load
[15:04:44]E (823) esp_image: Image length 1836048 doesn't fit in partition length 1835008
[15:04:44]E (823) boot: OTA app partition slot 0 is not bootable
[15:04:44]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[15:04:44]E (834) boot: OTA app partition slot 1 is not bootable
[15:04:44]E (839) boot: No bootable app partitions in the partition table
[15:04:44]ets Jun  8 2016 00:22:57
[15:04:44]
[15:04:44]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:04:44]configsip: 188777542, SPIWP:0xee
[15:04:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:04:44]mode:DIO, clock div:2
[15:04:44]load:0x3fff0030,len:6652
[15:04:44]ho 0 tail 12 room 4
[15:04:44]load:0x40078000,len:15052
[15:04:44]load:0x40080400,len:3836
[15:04:44]entry 0x4008069c
[15:04:44]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[15:04:44]I (57) boot: compile time 10:16:21
[15:04:44]I (58) boot: Multicore bootloader
[15:04:44]I (61) boot: chip revision: v1.1
[15:04:44]I (65) boot.esp32: SPI Speed      : 40MHz
[15:04:44]I (70) boot.esp32: SPI Mode       : DIO
[15:04:44]I (74) boot.esp32: SPI Flash Size : 4MB
[15:04:44]I (79) boot: Enabling RNG early entropy source...
[15:04:44]I (84) boot: Partition Table:
[15:04:44]I (88) boot: ## Label            Usage          Type ST Offset   Length
[15:04:44]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[15:04:44]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[15:04:44]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[15:04:44]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[15:04:44]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[15:04:44]I (133) boot: End of partition table
[15:04:44]I (137) boot: No factory image, trying OTA 0
[15:04:44]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=65918h (416024) map
[15:04:45]I (301) esp_image: segment 1: paddr=00075940 vaddr=3ffbdb60 size=051bch ( 20924) load
[15:04:45]I (310) esp_image: segment 2: paddr=0007ab04 vaddr=40080000 size=05514h ( 21780) load
[15:04:45]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=136950h (1272144) map
[15:04:45]I (779) esp_image: segment 4: paddr=001b6978 vaddr=40085514 size=19a70h (105072) load
[15:04:45]E (823) esp_image: Image length 1836048 doesn't fit in partition length 1835008
[15:04:45]E (823) boot: OTA app partition slot 0 is not bootable
[15:04:45]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[15:04:45]E (834) boot: OTA app partition slot 1 is not bootable
[15:04:45]E (839) boot: No bootable app partitions in the partition table
[15:04:45]ets Jun  8 2016 00:22:57
[15:04:45]
```

userosos, 2024-07-27:
> similar issue with me:
> 
> INFO ESPHome 2024.7.2 INFO Reading configuration /config/esphome/m5stack-atom-echo-869a64.yaml... INFO Updating https://github.com/esphome/esphome.git@pull/5230/head INFO Updating https://github.com/jesserockz/esphome-components.git@None INFO Generating C++ source... INFO Cloning https://github.com/espressif/esp-adf@v2.5 Traceback (most recent call last): File "/usr/local/bin/esphome", line 8, in sys.exit(main()) ^^^^^^ File "/esphome/esphome/**main**.py", line 1080, in main return run_esphome(sys.argv) ^^^^^^^^^^^^^^^^^^^^^ File "/esphome/esphome/**main**.py", line 1067, in run_esphome rc = POST_CONFIG_ACTIONS[args.command](args, config) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/esphome/esphome/**main**.py", line 433, in command_compile exit_code = write_cpp(config) ^^^^^^^^^^^^^^^^^ File "/esphome/esphome/**main**.py", line 194, in write_cpp return write_cpp_file() ^^^^^^^^^^^^^^^^ File "/esphome/esphome/**main**.py", line 212, in write_cpp_file writer.write_cpp(code_s) File "/esphome/esphome/writer.py", line 344, in write_cpp copy_src_tree() File "/esphome/esphome/writer.py", line 297, in copy_src_tree copy_files() File "/esphome/esphome/components/esp32/**init**.py", line 684, in copy_files repo_dir, _ = git.clone_or_update( ^^^^^^^^^^^^^^^^^^^^ File "/esphome/esphome/git.py", line 67, in clone_or_update run_git_command(cmd) File "/esphome/esphome/git.py", line 31, in run_git_command raise cv.Invalid(lines[-1][len("fatal: ") :]) voluptuous.error.Invalid: fetch-pack: invalid index-pack output

Can you clean build files may be it will work?

Evanito, 2024-07-29:
Suggest folks experiencing this issue update their esphome configs for the device to use the new adopted config. (Change `xxx.yaml@main`  to `xxx.adopted.yaml@main`) and that should do the trick

userosos, 2024-09-07:
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/m5stack-atom-echo-0dba90.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
Failed config
Can`t use the box today.
external_components: [source /data/packages/c46f54c1/voice-assistant/m5stack-atom-echo.yaml:294]
  - 
    fetch-pack: invalid index-pack output.
    source: github://pr#5230
    components: 
      - esp_adf
    refresh: 0s
  - source: github://jesserockz/esphome-components
    components: 
      - file
    refresh: 0s

userosos, 2024-09-07:
Bultin config:

```
substitutions:
  name: m5stack-atom-echo-0dba90
  friendly_name: M5Stack Atom Echo 0dba90
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: VqVVZlHXxXEy2EgTo4o2+q3wb08SbyV5ZpHBpzRfxwI=
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

closed 2025-05-10 by github-actions[bot]