esphome/firmware #219: on device wake word only works for first command - s3box3 wake word config

BigBobbas, 2024-06-29:
https://github.com/esphome/firmware/blob/main/wake-word-voice-assistant/esp32-s3-box-3.yaml
micro wakeword does not start again after first command when using 'on device' wakeword.

removing line 223 - fixes the issue
`- lambda: return id(voice_assistant_phase) != ${voice_assist_timer_finished_phase_id}; `

DeveshwarH1996, 2024-06-30:
Hi, there is a little more happening for me. I belive the on device wake word recognition triggers something that eventually puts the device in WAITING_FOR_VAD state instead of just starting the audio pipeline. This means I have to use the wake word that is configured with openWakeWord after using the wake word that is used to trigger locally. For example if I use okay nabu on device and hey jarvis on the openWakeWord, for some reason after I say 'okay nabu' it waits for me to say 'hey jarvis' before it starts listening to any commands

HLandstrom, 2024-06-30:
I actually never thought of that. I have the same wake word on device and in openWakeWord.
It goes back to WAITING_FOR_VAD on the first try as you said but the second try it takes it and accepts voice input.

Here is logs for the first wake word so it goes back to WAITING_FOR VAD: https://pastebin.com/v7a7iCN0
and here is the logs for the second wake word that actually works: https://pastebin.com/iTwPSMtC

connelhooley, 2024-06-30:
Hi, I had a similar issue except the device wasn't responding to any of my wake words. I'm an ESP Home newb, just sharing what happened in-case this is useful.

- I stopped my wake word docker container as I was planning on using on device wake word dectection.
- Created the voice-assistant YAML below
- Ran the install and downloaded the .bin file from ESP Home web portal on my server.
- I went to https://web.esphome.io/ and installed the bin file over a USB cable.
- I then added the device to Home Assistant. I could see it there and I could toggle the LCD Backlight so the connection seemed good.
- The device was not responding to the wake word on the device. Below are the logs where I rebooted the device and then tried to use the wake word.
- I then started the wake word container again (it uses the same wakw word as the device)
- Set the "wake word engine location" selector for the device to "In Home Assistant"
- Set it back to "On device"
- And then the device started listening to me fairly consistly for around 20 mins.
- The next day however it responded once and then failed afterwards.

It's been quite unstable but the fact that switching the wake word engine location from and to on-device seemed to do something gives me hope that #220 might fix it.

voice-assistant.yml:

```yaml
substitutions:
  name: voice-assistant
  friendly_name: Voice Assistant
  micro_wake_word_model: hey_jarvis
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: some_secret_value

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Logs:
[device.log](https://github.com/user-attachments/files/16044788/device.log)

connelhooley, 2024-07-06:
I've updated to the latest version after #220 was merged and it seems to be working fine after some initial testing

HLandstrom, 2024-07-07:
My problems are fixed too after #220 was merged

Anto79-ops, 2024-07-18:
hi, was this PR in for 2024.7.0 of ESPHome? I still cannot get wakeword to work (even for my Atom5 Echo's)

BigBobbas, 2024-07-18:
@Anto79-ops there is possibly a new issue with 2024.7.0 if you view the device logs... do they just sit at 'waiting for VAD '?

Anto79-ops, 2024-07-19:
@BigBobbas just checked the logs and I don't the 'waiting for VAD'. Its just simply not responding, to wake word on device or HA. Im not sure what is going on, i'm wondering if I just should factory default the device.

Anto79-ops, 2024-08-02:
this is now working for me. I did not have to reset the device, so I think the last HW update 24.8.1 and FW upate 2024.7.3 did it. 

thanks all!

closed 2025-05-10 by github-actions[bot]