esphome/firmware #227: M5 Stack Atom Echo VA firmware does not install locally

HarvsG, 2024-07-06:
Related: https://github.com/esphome/issues/issues/5714#issuecomment-2211542063
Another user with the same issue: https://community.home-assistant.io/t/m5-stack-atom-echo-installation/746641/2

I can install the VA config with [Ready-made projects](https://esphome.io/projects/index.html) however when I try and adopt the device locally, or even manually add the device and install the VA config
```yaml
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
api:
  encryption:
    key: someAPIkey=
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

I get the following on loop.
```
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[13:02:07]8069c
[13:02:07]I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
[13:02:07]I (30) boot: compile time 12:57:16
[13:02:07]I (30) boot: Multicore bootloader
[13:02:07]I (33) boot: chip revision: v1.1
[13:02:07]I (36) boot.esp32: SPI Speed      : 40MHz
[13:02:07]I (41) boot.esp32: SPI Mode       : DIO
[13:02:07]I (46) boot.esp32: SPI Flash Size : 4MB
[13:02:07]I (50) boot: Enabling RNG early entropy source...
[13:02:07]I (56) boot: Partition Table:
[13:02:07]I (59) boot: ## Label            Usage          Type ST Offset   Length
[13:02:07]I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
[13:02:07]I (74) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[13:02:07]I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
[13:02:07]I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[13:02:07]I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[13:02:07]I (104) boot: End of partition table
[13:02:07]I (108) boot: No factory image, trying OTA 0
[13:02:07]I (113) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=65984h (416132) map
[13:02:07]I (272) esp_image: segment 1: paddr=000759ac vaddr=3ffbdb60 size=051bch ( 20924) load
[13:02:07]I (281) esp_image: segment 2: paddr=0007ab70 vaddr=40080000 size=054a8h ( 21672) load
[13:02:07]I (290) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=1369a0h (1272224) map
[13:02:07]I (750) esp_image: segment 4: paddr=001b69c8 vaddr=400854a8 size=19adch (105180) load
[13:02:07]E (794) esp_image: Image length 1836240 doesn't fit in partition length 1835008
[13:02:07]E (794) boot: OTA app partition slot 0 is not bootable
[13:02:07]E (796) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[13:02:07]E (805) boot: OTA app partition slot 1 is not bootable
[13:02:07]E (810) boot: No bootable app partitions in the partition table
[13:02:07]ets Jun  8 2016 00:22:57
[13:02:07]
[13:02:07]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:02:07]configsip: 188777542, SPIWP:0xee
[13:02:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:02:07]mode:DIO, clock div:2
[13:02:07]load:0x3fff0030,len:6652
[13:02:07]ho 0 tail 12 room 4
[13:02:07]load:0x40078000,len:15052
[13:02:07]load:0x40080400,len:3836
[13:02:07]entry 0x4008069c
[13:02:07]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[13:02:07]I (57) boot: compile time 12:57:16
[13:02:07]I (58) boot: Multicore bootloader
[13:02:07]I (61) boot: chip revision: v1.1
[13:02:07]I (65) boot.esp32: SPI Speed      : 40MHz
[13:02:07]I (70) boot.esp32: SPI Mode       : DIO
[13:02:07]I (74) boot.esp32: SPI Flash Size : 4MB
[13:02:07]I (79) boot: Enabling RNG early entropy source...
[13:02:07]I (84) boot: Partition Table:
[13:02:07]I (88) boot: ## Label            Usage          Type ST Offset   Length
[13:02:07]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[13:02:07]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[13:02:07]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[13:02:07]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[13:02:07]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[13:02:07]I (133) boot: End of partition table
[13:02:07]I (137) boot: No factory image, trying OTA 0
[13:02:07]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=65984h (416132) map
[13:02:08]I (301) esp_image: segment 1: paddr=000759ac vaddr=3ffbdb60 size=051bch ( 20924) load
[13:02:08]I (310) esp_image: segment 2: paddr=0007ab70 vaddr=40080000 size=054a8h ( 21672) load
[13:02:08]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=1369a0h (1272224) map
[13:02:08]I (779) esp_image: segment 4: paddr=001b69c8 vaddr=400854a8 size=19adch (105180) load
[13:02:08]E (823) esp_image: Image length 1836240 doesn't fit in partition length 1835008
[13:02:08]E (823) boot: OTA app partition slot 0 is not bootable
[13:02:08]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[13:02:08]E (834) boot: OTA app partition slot 1 is not bootable
[13:02:08]E (839) boot: No bootable app partitions in the partition table
[13:02:08]ets Jun  8 2016 00:22:57
[13:02:08]
[13:02:08]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:02:08]configsip: 188777542, SPIWP:0xee
[13:02:08]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:02:08]mode:DIO, clock div:2
[13:02:08]load:0x3fff0030,len:6652
[13:02:08]ho 0 tail 12 room 4
[13:02:08]load:0x40078000,len:15052
[13:02:08]load:0x40080400,len:3836
[13:02:08]entry 0x4008069c
```
[logs_m5-atom-echo_run (1).txt](https://github.com/user-attachments/files/16115937/logs_m5-atom-echo_run.1.txt)

I can install other configs e.g a plain ESP Home web which is what makes me think it is an issue with this firmware.

HarvsG, 2024-07-06:
I think the issue might be https://github.com/esphome/firmware/commit/a79c9fa96fe0fdd24c17e7571de3ef37755d54a1 because rolling back worked
```yaml
substitutions:
  name: m5stack-atom-echo-b836b0
  friendly_name: M5Stack Atom Echo b836b0
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@cd57ca6f951d44cc5bf61de124a15b349ef1f9a4
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: +myAPIKEY=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

TxHarleyDude, 2024-07-06:
I followed the suggestion from HarvsG and it got me back working as well.

lmaso90, 2024-07-06:
The suggestion from HarvsG also worked for me

molusk, 2024-07-08:
Thx for the workaround @HarvsG 👍

SDPismo, 2024-07-09:
Rollback worked for me. Thx! Wasted a lot of time working on this before giving up and rolling back :/
Looking forward to seeing what happened and a solution other than the workaround.

HarvsG, 2024-07-09:
Tagging @jesserockz as he made the commit and it seems this error is very reproduceable.
Apologies for the @.

HarvsG, 2024-07-10:
So I have done a bit of work to narrow down the issue and this also works as a workaround

```yml
substitutions:
  name: m5stack-atom-echo-b836b0
  friendly_name: M5Stack Atom Echo b836b0
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: +10OldPHYLsCRJ3HzpwFEvjYWcoOIRDaohlL8rdenvU=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# HarvsG's customisations, will likely break UI updating in HomeAssistant and 
# can be deleted once https://github.com/esphome/firmware/issues/227 is merged
update:
  - platform: http_request
    id: !remove update_http_request
```

almoney, 2024-07-10:
Also how do we do real regression testing on these releases.  I am going to skip all updates until I need something or a security fix is needed based on the last 2 updates...

Kaldek, 2024-07-11:
> Also how do we do real regression testing on these releases. I am going to skip all updates until I need something or a security fix is needed based on the last 2 updates...

Amen to that.  Burned a week on this darned issue, right when I was doing some heavy lifting on voice assistants around the house.

HarvsG, 2024-07-11:
> Amen to that. Burned a week on this darned issue, right when I was doing some heavy lifting on voice assistants around the house.

I have no idea what testing is done. But to be fair to the team the fact that the ready-made-project install works suggests that this error may be environment specific.

It would be helpful if everyone who has experienced this issue comments (or ideally edits their comment above) to explain their environment and ESPhome version.

HarvsG, 2024-07-11:
Ok, I might have another work-around.
So simply pasting the yaml from https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml or more specifically https://raw.githubusercontent.com/esphome/firmware/main/voice-assistant/m5stack-atom-echo.yaml seems to work. 
This is weird because it should be exactly the same as using the package import method. The only thing I can think of is that the package import results in an extra module being installed which tips the image into being too big.

jolbi, 2024-07-12:
I am experiencing the same problem as HarvsG. I managed to flash it via esphome cli on Debian trough USB port.
But after compilation is succesfull I am getting this error no matter the config I use:
```
ERROR Running command failed: Unable to verify flash chip connection (No serial data received.).
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 460800 --port /dev/ttyUSB0 --chip esp32 write_flash -z --flash_size detect 0x10000 /home/timg/esp_compiles/atom_echo_mostlycris/.esphome/build/m5stack-atom-echo/.pioenvs/m5stack-atom-echo/firmware.bin 0x1000 /home/timg/esp_compiles/atom_echo_mostlycris/.esphome/build/m5stack-atom-echo/.pioenvs/m5stack-atom-echo/bootloader.bin 0x8000 /home/timg/esp_compiles/atom_echo_mostlycris/.esphome/build/m5stack-atom-echo/.pioenvs/m5stack-atom-echo/partitions.bin 0xe000 /home/timg/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin locally.
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
```
With baud rate 115200 upload is successful via USB, but:
My Atom Echo is no longer recognised on USB port on a Windows PC, so I cannot/don't know how to flash it via ready-made projects. The device is also not listed in the esphome dashboard on HA anymore. But it is available as a device in HA and it is working properly.

Any ideas what could I do to bring it to initial state? Should I try to erase flash via esptool.py?

freeDom-, 2024-07-13:
I am experiencing the same issue.

Details of my environment:
    Home Assistant OS as Virtual Machine on a Proxmox server. 6GB RAM assigned, 4 cores of Intel N100
    ESPHome version 2024.6.6

Rolling back the mentioned commit did work for me as workaround.

Christoph-Wagner, 2024-07-15:
Same issue

* HA OS in a Proxmox VM, all current versions. 
  * 2 GB RAM, 2 cores of J4125
* ESPHome version 2024.6.6
* Originally installed via web, later adopted to ESPHome, which is when the error occurred.

Notably, `id: !remove update_http_request` did not work for me (error about `update_http_request` missing or something), and I had to instead used the rollbacked commit, which worked fine.

orywig, 2024-07-19:
I am seeing this same error after I try to update my atom from ESPHome add-on to HA. Flashing the device from the ESPHome ready-made projects page works, but once I adopt it again and try to install the config from ESPHome add-on, I get the message that the OTA partition is not big enough for the package and that I need to flash it over USB instead of wirelessly. I get the same error denoted in the original post above in the logs.

CoolSaet, 2024-07-20:
> Same issue
>
>
>
> * HA OS in a Proxmox VM, all current versions.
>
>   * 2 GB RAM, 2 cores of J4125
>
> * ESPHome version 2024.6.6
>
> * Originally installed via web, later adopted to ESPHome, which is when the [...]

I had the same error about update_http_request missing. Cleaning the build files resolved that.

EnumC, 2024-07-20:
Anyone know why NVS has size of 0006d000 (436kb)? If we reduce the partition by 4kb and increase each OTA app partition by 2kb, that would give plenty of space for the app to load.

EnumC, 2024-07-20:
Actually it looks like there's unutilized space between end of phy-init (0xc) to start of ota-1 (0x10).

mithomas, 2024-07-22:
@Christoph-Wagner: I had the same issue with version 2024.7.0; with the following I could get past the dependency issue:

```
ota:
  - platform: http_request
    id: !remove ota_http_request

update:
  - platform: http_request
    id: !remove update_http_request
```

Image is still too large for installing, though.

SDPismo, 2024-07-22:
Does anyone know if any developer is looking into this issue yet? Seems like it's wide spread and still occurring. As mentioned in another thread, I didn't see anything that appeared to address this in the latest esphome release (2024.7.1).

HarvsG, 2024-07-22:
> Does anyone know if any developer is looking into this issue yet? Seems like it's wide spread and still occurring. As mentioned in another thread, I didn't see anything that appeared to address this in the latest esphome release (2024.7.1).

No, it doesn't look as if it has got any traction

In the meantime, I've opened a PR #242  that supports optional on-device wake words, you can install use this config

```yaml
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu # hey_jarvis and hey_mycroft are also supported
packages:
  m5stack.atom-echo-wake-word-voice-assistant: github://HarvsG/firmware/wake-word-voice-assistant/m5stack-atom-echo.yaml@patch-3
esphome:
  name: ${name}
  name_add_mac_suffix: True
  friendly_name: ${friendly_name}
api:
  encryption:
    key: SOMEAPIKEY
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
Enjoy!

KEV-github, 2024-07-22:
> No, it doesn't look as if it has got any traction
>
> In the meantime, I've opened a PR #242 that supports optional on-device wake words, you can install use this config
>
> ```yaml
> substitutions:
>   name: [...]

Agree. Bit disappointed in the developers response to this issue. This solution worked for me. Thanks

austingreisman, 2024-07-22:
I've also gotten this problem

nem0nic, 2024-07-23:
Same issue since the previous update.  Skipped the last update because I didn't have time to be screwing around with it hoping the latest version would have the fix but same issue with the latest version as well.

HAOS on Rpi5 8GB

ERROR Error binary size: Error: The OTA partition on the ESP is too small. ESPHome needs to resize this partition, please flash over USB

freeDom-, 2024-07-23:
> Actually it looks like there's unutilized space between end of phy-init (0xc000) to start of ota-0 (0x10000). That seems to be an extra 16kb we can use for our app.

Thabks for your investigation. Are you willing to open a PR for this?

nem0nic, 2024-07-23:
It'd already part of [https://github.com/esphome/firmware/pull/242](url) I believe

HarvsG, 2024-07-23:
> It'd already part of [https://github.com/esphome/firmware/pull/242](url) I believe

Nope #242 is just a different firmware. I had to remove `improv ble` to make the space.

freeDom-, 2024-07-23:
I think rolling back the changes from @jesserockz is not a real solution, nor removing other features. If NVS is too big and there is unused space left, redistributing space / resizing partitions seems to be the better option!

almoney, 2024-07-24:
> Agree. Bit disappointed in the developers response to this issue. This solution worked for me. Thanks

I really appreciate this config help!  I am up and running with the local wake word!  Thank you!

jarrah31, 2024-07-24:
> ```yaml
> substitutions:
>   name: m5stack-atom-echo
>   friendly_name: M5Stack Atom Echo
>   micro_wake_word_model: okay_nabu # hey_jarvis and hey_mycroft are also supported
> packages: [...]

Is there any way to fully reset/wipe the Atom Echo because I'm unfortunately still ending up with a boot loop when I install this config via my ESPHome dashboard USB method.
```
[20:22:26]E (823) esp_image: Image length 1835616 doesn't fit in partition length 1835008
[20:22:26]E (823) boot: OTA app partition slot 0 is not bootable
[20:22:26]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[20:22:26]E (833) boot: OTA app partition slot 1 is not bootable
[20:22:26]E (839) boot: No bootable app partitions in the partition table
```

nem0nic, 2024-07-24:
> Is there any way to fully reset/wipe the Atom Echo because I'm unfortunately still ending up with a boot loop when I install this config via my ESPHome dashboard USB method.

Go into ESPHome and find your device and hit the 3 dots on the right on the device's card and select "Clean Build Files" and try the install again.

jarrah31, 2024-07-24:
> Go into ESPHome and find your device and hit the 3 dots on the right on the device's card and select "Clean Build Files" and try the install again.

ahh thank you so much, that did the trick!

JonSnow88, 2024-07-25:
When will we actually see a proper fix

jesserockz, 2024-07-25:
I merged #244 yesterday which is what people need to change their local adopted config to. Basically it removed a bunch of unnecessary components once the user has decided to adopt. This frees up a significant amount of flash.

Basically in your local yaml change from this:
```yaml
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
```
to this:
```yaml
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.adopted.yaml@main
```

The on device wake word stuff from @HarvsG is unrelated to this issue so please do not discuss it here. Testing and reports from that can be made on that PR itself.

closed 2024-07-25 by jesserockz