esphome/firmware #228: Bluetooth Proxy Ready-Made Project aborting

izenn, 2024-07-06:
using the web interface i uploaded the bluetooth-proxy to my device and it goes into a boot/abort loop.  I've used this device and cable before for this project, i was going to update it to the current version before i put it back in service.  I've also downloaded the YAML and compiled/uploaded it with esphome on the cli via serial as well as the esphome dashboard as an OTA update and got the same results

```
[13:48:22]abort() was called at PC 0x401372c4 on core 0
[13:48:22]
[13:48:22]
[13:48:22]Backtrace: 0x40082bfa:0x3ffbe3f0 0x40091ec9:0x3ffbe410 0x40097ed6:0x3ffbe430 0x401372c4:0x3ffbe4a0 0x400835f5:0x3ffbe4c0 0x401c644b:0x3ffbc780 0x40137a15:0x3ffbc7a0 0x400932a0:0x3ffbc7c0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40082bfa: panic_abort at /Users/izenn/.platformio/packages/framework-espidf/components/esp_system/panic.c:408
WARNING Decoded 0x40091ec9: esp_system_abort at /Users/izenn/.platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
WARNING Decoded 0x40097ed6: abort at /Users/izenn/.platformio/packages/framework-espidf/components/newlib/abort.c:46
WARNING Decoded 0x401372c4: task_wdt_isr at /Users/izenn/.platformio/packages/framework-espidf/components/esp_system/task_wdt.c:176 (discriminator 3)
WARNING Decoded 0x400835f5: _xt_lowint1 at /Users/izenn/.platformio/packages/framework-espidf/components/freertos/port/xtensa/xtensa_vectors.S:1118
WARNING Decoded 0x401c644b: cpu_ll_waiti at /Users/izenn/.platformio/packages/framework-espidf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /Users/izenn/.platformio/packages/framework-espidf/components/esp_pm/pm_impl.c:853
WARNING Decoded 0x40137a15: esp_vApplicationIdleHook at /Users/izenn/.platformio/packages/framework-espidf/components/esp_system/freertos_hooks.c:63
WARNING Decoded 0x400932a0: prvIdleTask at /Users/izenn/.platformio/packages/framework-espidf/components/freertos/tasks.c:4099
```

yaml:
```
substitutions:
  name: esp32-bluetooth-proxy
  friendly_name: Bluetooth Proxy

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: "24.7.4.1"

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ssid: Internal
  password: "password"

api:
  encryption:
    key: key

logger:
improv_serial:

ota:
  - platform: esphome
    id: ota_esphome
  - platform: http_request
    id: ota_http_request

update:
  - platform: http_request
    id: update_http_request
    name: Firmware
    source: https://firmware.esphome.io/bluetooth-proxy/esp32-generic/manifest.json

http_request:

dashboard_import:
  package_import_url: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
```

tomasdeml, 2024-07-07:
I've the same issue with the Voice Assistant project after updating to the latest version containing [this PR](https://github.com/esphome/firmware/pull/223). Not sure if it's important but my IoT VLAN has no access to Internet...

izenn, 2024-07-07:
i just stripped the OTA stuff out of the yaml and it is working now.  Here's my yaml that is working:

```
substitutions:
  name: esp32-bluetooth-proxy
  friendly_name: Bluetooth Proxy

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: "24.7.4.1"

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ssid: Internal
  password: "password"

api:
  encryption:
    key: key

logger:
improv_serial:

ota:
  - platform: esphome
    id: ota_esphome

dashboard_import:
  package_import_url: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
```

izenn, 2024-07-07:
just as a note -- i use this to do the initial flash, but then when i add to the esphome web interface rather than clicking install, i click skip then copy/paste the yaml into the config so that future updates will use the correct settings

Floriszz, 2024-07-16:
When validate it, it says:
```
Component ota.http_request requires component http_request.
No quick fixes available
```
screenshot:
[image]

Maybe this has to do with the new way of listing in one of the latest updates, but this time I can't get it right.

izenn, 2024-07-16:
remove 
```
    - platform: http_request
      id: ota_http_request
```
also, you're going to want to change your password since you just posted it publically

Floriszz, 2024-07-16:
Hi @izenn, does http_request not have a function for the 'Update:' part? Or is that the mistake that it is actually copied double while it is already in the 'update:' part?
btw. I already changed last characters in the password to show it here. But thank you for the tip😉

Floriszz, 2024-07-16:
Ok, found the right format!
Indeed it has to do with the new 'list requirements' in yaml for esphome, which is more stringent with nesting/indends.
Since 'Update:' is reliant on 'ota:' you should not have a different list indent. So if Ota: lines start at character 0, so needs Update: lines!

```
ota:
- platform: esphome
  id: ota_esphome
  password: ba5214c7c802ea8c5ad44ff374dcf362
- platform: http_request
  id: ota_http_request

update:
  - platform: http_request
    id: update_http_request
    name: Firmware Update
    source: https://firmware.esphome.io/bluetooth-proxy/esp32-generic/manifest.json
```

izenn, 2024-07-16:
what it seems is that if you try to deploy on a device that does not have internet access but does have network access it will abort.  if the device has full internet access then the ota/http_request portion is fine to be in there since it is able to hit the source url (and therefore not timing out) and the system will not go into the abort loop

jazzmonger, 2025-02-02:
I also get this message, even trying @Floriszz exact code and formatting...

  Component ota.http_request requires component http_request.
  platform: http_request
  id: ota_http_request

```
ota:
- platform: esphome
  id: ota_esphome
  password: xxxxx
- platform: http_request
  id: ota_http_request

update:
  - platform: http_request
    id: update_http_request
    name: Firmware Update
    source: https://firmware.esphome.io/bluetooth-proxy/esp32-generic/manifest.json
```

Has anyone cracked this?

jesserockz, 2025-04-17:
If you are taking control and writing and compiling your own configurations, you should not have the http_request/update components in there as that means if there is an update it will override your locally compiled firmware. 

The take control process for the latest bluetooth-proxies configurations has this sorted and does not include any http_request code if you choose to take control.

closed 2025-04-17 by jesserockz