esphome/firmware #234: M5Stack ATOM Echo enters a reboot loop after installing latest firmware

DetunizedGravity, 2024-07-12:
Hello,

I would like to report that both my 2 Atom Echoes (v1.0 hardware)  are stuck into a reboot loop after installing the latest voice assistant (home assistant) firmware as per:
[https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml](https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml)

The only thing I modified was to add wifi parameters to allow them to connect to my SSID. But the behavior is the same with or without this change.

The error that triggers the abort is as follows:
```
E (11355) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[08:58:24]E (11355) task_wdt:  - loopTask (CPU 1)
[08:58:24]E (11355) task_wdt: Tasks currently running:
[08:58:24]E (11355) task_wdt: CPU 0: IDLE0
[08:58:24]E (11355) task_wdt: CPU 1: IDLE1
[08:58:24]E (11355) task_wdt: Aborting.
[08:58:24]
[08:58:24]abort() was called at PC 0x400fff24 on core 0
[08:58:24]
[08:58:24]
[08:58:24]Backtrace: 0x40082f2a:0x3ffb0870 0x40089bfd:0x3ffb0890 0x4009043a:0x3ffb08b0 0x400fff24:0x3ffb0920 0x400838ad:0x3ffb0940 0x4019891f:0x3ffbbe10 0x40100609:0x3ffbbe30 0x4008b000:0x3ffbbe50
[08:58:24]
[08:58:24]
[08:58:24]
[08:58:24]
[08:58:24]ELF file SHA256: 48a493995ea962a3
[08:58:24]
[08:58:24]Rebooting...
```
Since I'm totally new to this ecosystem I am completely unable to understand (and fix) what is happening. If someone would be kind enough to help...

Full logs are attached here: [echo-0f7ba4_logs.txt](https://github.com/user-attachments/files/16189843/echo-0f7ba4_logs.txt)

mochman, 2024-07-12:
Same issue for me as well.

mochman, 2024-07-12:
I was able to fix it by removing all the `http_request` references in the  yaml config.
I removed it from the `ota:` section, the `update:` section, and the `http_request:` section.

Hope this helps.

lexiismadd, 2024-07-16:
I've been having the same issue..  Its to do with insufficient space on the device, the flash partition is too small for the image..  I removed the file section and references to it (given it doesn't work anyway), and that fixed it for me..

goobags, 2024-07-16:
I have the same issue on a ESP32-S3-Box. I tried to remove the http_request sections but it made it worse, probably did something wrong.

tipofthesowrd, 2024-07-17:
I'm facing the issue on my ATOM Echo. I had to remove the http section and references for it to boot normally

Taomyn, 2024-07-18:
I've been unable to get my Atom Echo to properly work since the release of ESPHome 2024.7.0 on Home Assistant

Initially when I tried to just update it as normal it said that a partition was too small and to use USB - so I did and it just came back boot looping

I then used the ESPHome to recover the device via USB, which worked, and Home Assistant could see but when it adopts the device and tries to update it, I get the same issue - cannot use WiFi and USB leaves me with it boot looping again.

My yaml is pretty basic, and none of the changes mentioned above to fix it work because they don't seem to be relevant:

```
substitutions:
  name: m5stack-atom-echo-31b338
  friendly_name: M5Stack Atom Echo 31b338
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <redacted>
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .mydomain.com

```

Anto79-ops, 2024-07-18:
I had the same problem, it was because I had extra things in my yaml which made too large of bin that it cannot fit on the device anymore, so it boot loops, what I did was remove encrpytion and that bad a small enoug bin to reflash (still at 96% space usage), so space is getting tight on the device. Here is my yaml, that is working for me.

```
substitutions:
  name: m5stack-atom-echo-a61210
  friendly_name: M5Stack Atom Echo a61210
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

HarvsG, 2024-07-18:
Could be related to #227?

HarlemSquirrel, 2024-07-22:
I got 7.0 to compile and install adding the following to the bottom of my config Yaml but so far it's not waking up with "OK Nabu" any more. Nothing stands out in the logs.

```yml
# Overrides to get smaller image
ota:
  platform: esphome
  id: ota_esphome

http_request: !remove

update: !remove
```

q20, 2024-07-22:
Changing the following:

```
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@cd57ca6f951d44cc5bf61de124a15b349ef1f9a4
```

then cleaning build files, now allows me to install successfully.

listlesslife, 2024-07-23:
> Changing the following:
> 
> ```
> packages:
>   m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@cd57ca6f951d44cc5bf61de124a15b349ef1f9a4
> ```
> 
> then cleaning build files, now allows me to install successfully.

I've been having problems with this since 2024.7.0. This worked for me. Thank you.

DetunizedGravity, 2024-07-25:
Fixed in [HA addon 2024.7.2](https://github.com/esphome/home-assistant-addon/releases/tag/2024.7.2) by shortening http_request default timeout to 4.5s. 5s was long enough to have the watchdog trigger. Updating the HA addon also means removing the 5s timer set in the yaml file, to let the default timer take precedence.

closed 2024-07-25 by DetunizedGravity