esphome/firmware #235: ModuleNotFoundError: No module named 'pkg_resources'

Floriszz, 2024-07-16:
I can't use ESPhome HomeAssistant add-in, while my RPi 3b, does not have enough memory.
I want to setup a Bluetooth-proxy. I can with the ready made package through https://esphome.io/projects/?type=bluetooth, but then I can't change any of the yaml config later on.
So I want to do it through the command line interface on Windows PC. This is working for Arduino framework.
But bluetooth_proxy requires esp-idf. I do read some requirements on https://esphome.github.io/esp-web-tools/ about "Preparing your firmware' with a version 4 or v3. But I don't understand what is meant with that.

the error I get is:

```
INFO ESPHome 2024.6.6
INFO Reading configuration D:\Docs\Werkset\ESP\esphome-web-75a2f0.yaml-v2.yml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing esphome-web-75a2f0-ble (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40407.240606 (4.4.7)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
ModuleNotFoundError: No module named 'pkg_resources':
  File "C:\Python\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Floris\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Python\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Python\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Python\Lib\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 676:
    return method(*args, **kw)
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Floris\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Floris\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 29:
    import pkg_resources
============================================= [FAILED] Took 10.36 seconds =============================================
```

joachimtingvold, 2024-07-30:
`pip install setuptools` should get you past that error.

whitty, 2024-08-11:
> `pip install setuptools` should get you past that error.

That seems to work for me, however that means that there's a missing dependency in some package - right?

Floriszz, 2024-08-17:
@joachimtingvold , thank you.
That did it! 
I saw same comments like mine on other boards and agree, that this is a dependency if you want to manually install the esp-idf platform.

rdndnl, 2024-09-17:
Same issue on a macOS using esphome from homebrew, resolved following @joachimtingvold suggestion but used pip as module since it is not available as command in esphome venv:

```
➜  /opt/homebrew/Cellar/esphome/2024.8.3/libexec/bin/python -m pip install setuptools
Collecting setuptools
  Downloading setuptools-75.1.0-py3-none-any.whl.metadata (6.9 kB)
Downloading setuptools-75.1.0-py3-none-any.whl (1.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 14.0 MB/s eta 0:00:00
Installing collected packages: setuptools
Successfully installed setuptools-75.1.0
```

closed 2025-05-09 by github-actions[bot]