esphome/firmware #240: (Atom Echo) Image length 1835728 doesn't fit in partition length 1835008

axwell, 2024-07-18:
Hello, 
i have a basic config for Atom Echo in my home asssitant, after updating esp home to 2024.7.0 and updated my atom echo device enters in a boot loop.

Config:
`
substitutions:
  name: m5stack-atom-echo-9d9400
  friendly_name: M5Stack Atom Echo 9d9400
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <redacted>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`

Logs:
`[10:26:28]I (779) esp_image: segment 4: paddr=001b6848 vaddr=40085524 size=19a60h (105056) load
[10:26:29]E (823) esp_image: ImaI (779) esp_image: segment 4: paddr=001b6848 vaddr=40085524 size=19a60h (105056) load
[10:26:29]E (823) esp_image: Image length 1835728 doesn't fit in partition length 1835008
[10:26:29]E (823) boot: OTA app partition slot 0 is not bootable
[10:26:29]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[10:26:29]E (833) boot: OTA app partition slot 1 is not bootable
[10:26:29]E (839) boot: No bootable app partitions in the partition table
[10:26:29]ets Jun  8 2016 00:22:57
[10:26:29]
[10:26:29]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:26:29]configsip: 188777542, SPIWP:0xee
[10:26:29]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:26:29]mode:DIO, clock div:2
[10:26:29]load:0x3fff0030,len:6652
[10:26:29]ho 0 tail 12 room 4
[10:26:29]load:0x40078000,len:15052
[10:26:29]load:0x40080400,len:3836
[10:26:29]entry 0x4008069c`

lindehoff, 2024-07-18:
This is related to existing issue https://github.com/esphome/firmware/issues/227

closed 2024-07-18 by axwell

HarvsG, 2024-07-18:
@axwell, please post your environment and logs etc in #227