esphome/firmware #241: After a few commands, Atom echo hangs.

draggero, 2024-07-21:
Hi, I have two problems with Atom Echo:

When I use it multiple times to issue a command using the wake word, it can hang for some time or until it is restarted.
When I try to upload m5stack-atom-echo.yaml, it stops properly capturing voice. I have to bring the device close to my mouth for it to capture anything. When I upload the original firmware, it works fine, but the problem described in point 1 occurs.
Any ideas?

draggero, 2024-07-22:
It works for me https://github.com/esphome/issues/issues/6059#issuecomment-2241205451
But still sometimes it hangs
`[10:07:00][E][voice_assistant:806]: Cannot receive audio, buffer is full`

Evanito, 2024-07-29:
Fixed if you use the new wake word v2 firmware. The existing "stream" mode is where this behavior is happening. Was able to reproduce it myself.

Taomyn, 2024-07-31:
I was having issues as well since the latest adopted firmware was released, so I switched to:

`m5stack.atom-echo-voice-assistant: github://esphome/firmware/wake-word-voice-assistant/m5stack-atom-echo.adopted.yaml@main`

And except with a ton of yellow warning during compilation, more than i have ever seen before, it seems to be working but only time will tell as it happens quite randomly and after a long time for me.

draggero, 2024-08-04:
> I was having issues as well since the latest adopted firmware was released, so I switched to:
>
> `m5stack.atom-echo-voice-assistant: [...]

Where I should insert it?

Taomyn, 2024-08-05:
> > I was having issues as well since the latest adopted firmware was released, so I switched to:
> > `m5stack.atom-echo-voice-assistant: github://esphome/firmware/wake-word-voice-assistant/m5stack-atom-echo.adopted.yaml@main`
> > And except with a ton of yellow warning during compilation, more than i have ever seen before, it seems to be working but only time will tell as it happens quite randomly and after a long time for me.
> 
> Where I should change it?

Replacing the current line under "Packages:" in your yaml config file for the device.