esphome/firmware #246: Update Frequency

TobiasPSP, 2024-07-24:
firstly, thx for all of your great work!

With the very high frequency of updates both for EspHome and HA core, and no separation for basic update category, i.e. dev, prod, security, there are today many parts stirring the pudding, and when they change all the time, it’s making things very hard to maintain.

I ran in quite a few issues lately that were solely *caused* by updates (breaking changes, unknown altered behavior, etc), these updates had no particular *benefit* for my particular scenario (as far as I can tell), so I could have probably ignored them for the next 15 version updates, then deal with the sum of them all sometime later this year with dedicated time.

Is it ok to defer most updates, or does that jeopardize security? The intrusive marks in HA suggest that updates are important.  

Would it be possible to at minimum start a beta/dev branch so that anyone can still get nightly builds, yet people with production systems can safely stick to well-tested updates twice a year or so unless security mandates otherwise?

HA and esphome are awesome. They both have outgrown their hobbyist roots, though, and I feel update scheme needs to grow with it.

many thx for your consideration 
Tobias

nem0nic, 2024-07-26:
I'd definitely vote for this.  I'm now 3 versions behind on my Atom Echo's because the updates broke them 3 versions ago and each new release introduces a new problem.  I wouldn't have even purchased them if they weren't listed on the ready made project pages so I was under the impression the support for them would be top tier.

jesserockz, 2025-04-28:
If a device works on its current firmware, you need to ask yourself, why am I going to update it. 
To answer that, you need to read the release notes yourself and figure out if there is something relevant to your specific device that is going to make it better for *you*. 
We put most big changes as their own section in the changelog, and so if nothing pops out at you, then just leave the device be. I personally have devices with 2-3 year old ESPHome firmware on them because they work and there is no point in updating them.

closed 2025-04-28 by jesserockz