esphome/firmware #247: ESPHome Firmware v24.7.24 breaks Echo Atom Further

nem0nic, 2024-07-25:
Previously we were dealing with OTA partition not being large enough for the last 2 updates under issue #240.  Our issue was closed out yesterday and a new version captioned above was released but that also breaks the atom when trying to update to v.24.7.24.  Reverting back to @HarvsG github repo gets it back to a working state but previous issue probably shouldn't have been closed until we could validate that it actually works?:

```
WARNING m5stack-atom-echo-236a78 @ 172.24.1.55: Connection error occurred: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2).
WARNING Can't connect to ESPHome API for m5stack-atom-echo-236a78 @ 172.24.1.55: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2). (HandshakeAPIError)
INFO Trying to connect to m5stack-atom-echo-236a78 @ 172.24.1.55 in the background
INFO Successfully connected to m5stack-atom-echo-236a78 @ 172.24.1.55 in 0.018s
WARNING m5stack-atom-echo-236a78 @ 172.24.1.55: Connection error occurred: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2).
INFO Successfully connected to m5stack-atom-echo-236a78 @ 172.24.1.55 in 0.043s
WARNING m5stack-atom-echo-236a78 @ 172.24.1.55: Connection error occurred: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2).
INFO Successfully connected to m5stack-atom-echo-236a78 @ 172.24.1.55 in 7.286s
WARNING m5stack-atom-echo-236a78 @ 172.24.1.55: Connection error occurred: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2).
INFO Successfully connected to m5stack-atom-echo-236a78 @ 172.24.1.55 in 0.051s
WARNING m5stack-atom-echo-236a78 @ 172.24.1.55: Connection error occurred: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2).
INFO Successfully connected to m5stack-atom-echo-236a78 @ 172.24.1.55 in 0.052s
WARNING m5stack-atom-echo-236a78 @ 172.24.1.55: Connection error occurred: m5stack-atom-echo-236a78 @ 172.24.1.55: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.2).
```

Evanito, 2024-07-29:
Heads up, should be fixed as of today if you change your esphome config to use the new adopted version. Edit the config, and change the url to include `blahblah.adopted.yaml@main` instead of `blahblah.yaml@main`

nem0nic, 2024-07-29:
Confirmed.  Set the URL back and made your change and my dev Atom Echo is working.  Time to push to prod :-D

closed 2025-03-21 by nem0nic