esphome/firmware #250: 2024.7.3 Breaks Atom Echo

nem0nic, 2024-08-01:
```shell
[17:44:57][D][esp-idf:000][speaker_task]: I (510281) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=8

[17:44:57][D][i2s_audio.speaker:206]: Started I2S Audio Speaker
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
[17:44:58][E][voice_assistant:806]: Cannot receive audio, buffer is full
``` 

After update, audio stops working which I believe was another issue we had in the past.

JasonDVD, 2025-01-25:
Has there still not been a fix for this? i'm on 2025.1.4

jesserockz, 2025-04-17:
There have been significant improvements to the configuration and components used for the atom-echo recently. Please try installing the latest firmware provided at https://esphome.io/projects/?type=voice

closed 2025-04-17 by jesserockz

nem0nic, 2025-04-21:
Yea unfortunately this made it worse because now I can't add it as a home assistant device.  Even though the new pop-up wizard asks me if I want to add it and it runs the same routine as the "Set up voice assistant" button does, it never adds it.  The "Set up voice assistant" button never goes away and now because I can't adopt it into ESPHome Web to roll it back to working firmware they are really useless now.