esphome/firmware #252: New install on S3-Box-3 ends up in boot loop

goobags, 2024-08-04:
I bought a new S3-Box-3 and have not been able to get it to load the voice assistant firmware for the last few weeks. Currently running HAOS 2024.7.4 and ESPHome 2024.7.3

Seems to be getting stuck in a boot loop as per below log output

`[13:00:59][C][api:025]: Setting up Home Assistant API server...
[13:00:59][C][voice_assistant:072]: Setting up Voice Assistant...
[13:00:59][C][esp_adf.speaker:024]: Setting up ESP ADF Speaker...
[13:00:59][I][app:062]: setup() finished successfully!
[13:01:04]E (12771) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[13:01:04]E (12771) task_wdt:  - loopTask (CPU 1)
[13:01:04]E (12771) task_wdt: Tasks currently running:
[13:01:04]E (12771) task_wdt: CPU 0: IDLE0
[13:01:04]E (12771) task_wdt: CPU 1: IDLE1
[13:01:04]E (12771) task_wdt: Aborting.
[13:01:04]
[13:01:04]abort() was called at PC 0x420785f8 on core 0
[13:01:04]
[13:01:04]
[13:01:04]Backtrace: 0x403779d2:0x3fc9d440 0x40380f71:0x3fc9d460 0x40388dba:0x3fc9d480 0x420785f8:0x3fc9d4f0 0x403783b5:0x3fc9d510 0x4212deb3:0x3fcaf110 0x42078dd9:0x3fcaf130 0x403824d0:0x3fcaf150
WARNING Found stack trace! Trying to decode it
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line: '/data/build/esp32-s3-box-3-059f88/.pioenvs/esp32-s3-box-3-059f88/firmware.elf': No such file
[13:01:04]
[13:01:04]
[13:01:04]
[13:01:04]
[13:01:04]ELF file SHA256: 03e08a82b34a8edd
[13:01:04]
[13:01:04]Rebooting...`

yaml file is as per below

`substitutions:
  name: esp32-s3-box-3-059f88
  friendly_name: ESP32 S3 Box 3 059f88
packages:
  esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 9xuFYZ2WUQalroebq6HH+/LmVsi5m0p/AvvlObchLx4=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password`

kuligs2, 2024-08-07:
try again, dont install it via esphome integration, use the website.. esphome is garbage.. it bricks devices from my experience..

https://www.home-assistant.io/voice_control/s3_box_voice_assistant/

Open this in chrome or edge only otherwise you wont be able to conenct from the website..  select the box 3 option 
[image]

press connect and install it that way, once it installs, press connect on the page again and in the pop up there should be add wifi option. if not then try again, keep pressing, try rebooting device, at some point it will show up.. it took me 15 min to get it..

goobags, 2024-08-13:
That's what I tried initially, and although ESPHome recognised it once therefore it must have connected to the network (which I can see in the logs), it never actually worked. I have tried again through the website you linked and I still get the same error ultimately ended up in a boot loop.

`E (13070) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13070) task_wdt:  - loopTask (CPU 1)
E (13070) task_wdt: Tasks currently running:
E (13070) task_wdt: CPU 0: IDLE0
E (13070) task_wdt: CPU 1: IDLE1
E (13070) task_wdt: Aborting.`

aschieweck, 2024-08-16:
I have the same issue: Installed to a new ESP32-S3-BOX-3 via the HA voice assistant website. Installation, Wi-Fi setup and pairing to HA worked without issue. The box then works, until I reboot it, e.g. hit reset button or unplug and plug it in again. It's then showing the loading, three dot image, thinks for a moment, flickers, and shows the loading screen again, ... By checking the logs, again though the website, I can see it's rebooting over and over with the same error message as in @goobags's last comment.