esphome/firmware #255: Ready Made Projects - BT Proxy "Failed to download manifest"

hamsando, 2024-08-05:
Attempting to go with a "Ready Made Project" from "https://esphome.io/projects/" - specifically, "Bluetooth Proxy" fails immediately with: "Failed to download manifest" immediately after connecting to the COM port.

This worked a couple of days ago. It seems that it is attempting to download:

https://firmware.esphome.io/bluetooth-proxies/olimex-esp32-poe-iso/manifest.json

but that file doesn't exist.

mhalano, 2024-08-05:
I also have this problem, but with the page https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/

closed 2024-08-06 by hamsando

Darkangeel-hd, 2024-08-07:
> Attempting to go with a "Ready Made Project" from "https://esphome.io/projects/" - specifically, "Bluetooth Proxy" fails immediately with: "Failed to download manifest" immediately after connecting [...]

This one should be fixed by esphome/bluetooth-proxies#97