esphome/firmware #257: Atom echo boot loop problem.

kuligs2, 2024-08-07:
After install its looping..
Logs:
```
[20:27:52]ets Jun  8 2016 00:22:57
[20:27:52]
[20:27:52]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:27:52]configsip: 188777542, SPIWP:0xee
[20:27:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:27:52]mode:DIO, clock div:2
[20:27:52]load:0x3fff0030,len:6652
[20:27:52]ho 0 tail 12 room 4
[20:27:52]load:0x40078000,len:15052
[20:27:52]load:0x40080400,len:3836
[20:27:52]entry 0x4008069c
[20:27:52]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[20:27:52]I (57) boot: compile time 20:20:34
[20:27:52]I (58) boot: Multicore bootloader
[20:27:52]I (61) boot: chip revision: v1.1
[20:27:52]I (65) boot.esp32: SPI Speed      : 40MHz
[20:27:52]I (70) boot.esp32: SPI Mode       : DIO
[20:27:52]I (74) boot.esp32: SPI Flash Size : 4MB
[20:27:52]I (79) boot: Enabling RNG early entropy source...
[20:27:52]I (84) boot: Partition Table:
[20:27:52]I (88) boot: ## Label            Usage          Type ST Offset   Length
[20:27:52]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[20:27:52]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[20:27:52]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[20:27:52]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[20:27:52]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[20:27:52]I (133) boot: End of partition table
[20:27:52]I (137) boot: No factory image, trying OTA 0
[20:27:52]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=658b4h (415924) map
[20:27:52]I (301) esp_image: segment 1: paddr=000758dc vaddr=3ffbdb60 size=051bch ( 20924) load
[20:27:52]I (310) esp_image: segment 2: paddr=0007aaa0 vaddr=40080000 size=05578h ( 21880) load
[20:27:52]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=1369a0h (1272224) map
[20:27:52]I (779) esp_image: segment 4: paddr=001b69c8 vaddr=40085578 size=19a0ch (104972) load
[20:27:52]E (823) esp_image: Image length 1836032 doesn't fit in partition length 1835008
[20:27:52]E (823) boot: OTA app partition slot 0 is not bootable
[20:27:52]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[20:27:52]E (834) boot: OTA app partition slot 1 is not bootable
[20:27:52]E (839) boot: No bootable app partitions in the partition table
[20:27:52]ets Jun  8 2016 00:22:57
[20:27:52]
[20:27:52]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:27:52]configsip: 188777542, SPIWP:0xee
[20:27:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:27:52]mode:DIO, clock div:2
[20:27:52]load:0x3fff0030,len:6652
[20:27:52]ho 0 tail 12 room 4
[20:27:52]load:0x40078000,len:15052
[20:27:52]load:0x40080400,len:3836
[20:27:52]entry 0x4008069c
[20:27:52]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[20:27:52]I (57) boot: compile time 20:20:34
[20:27:52]I (58) boot: Multicore bootloader
[20:27:52]I (61) boot: chip revision: v1.1
[20:27:52]I (65) boot.esp32: SPI Speed      : 40MHz
[20:27:52]I (70) boot.esp32: SPI Mode       : DIO
[20:27:52]I (74) boot.esp32: SPI Flash Size : 4MB
[20:27:52]I (79) boot: Enabling RNG early entropy source...
[20:27:52]I (84) boot: Partition Table:
[20:27:52]I (88) boot: ## Label            Usage          Type ST Offset   Length
[20:27:52]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[20:27:52]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[20:27:52]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[20:27:53]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[20:27:53]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[20:27:53]I (133) boot: End of partition table
[20:27:53]I (137) boot: No factory image, trying OTA 0
[20:27:53]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=658b4h (415924) map
[... 370 lines cut ...]
      - script.execute: reset_led
  - platform: template
    id: timer_ringing
    optimistic: true
    internal: true
    restore_mode: ALWAYS_OFF
    on_turn_on:
      - delay: 15min
      - switch.turn_off: timer_ringing

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s
  - source: github://jesserockz/esphome-components
    components: [file]
    refresh: 0s

esp_adf:

file:
  - id: timer_finished_wave_file
    file: https://github.com/esphome/firmware/raw/main/voice-assistant/sounds/timer_finished.wav
```

RobertoSchneiders, 2024-08-07:
I'm having the same issue on a brand-new device. My settings are identical to https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml (except for api and wifi settings ofc)

logs:
```
[19:57:42]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[19:57:42]I (57) boot: compile time 18:44:21
[19:57:42]I (58) boot: Multicore bootloader
[19:57:42]I (61) boot: chip revision: v1.1
[19:57:42]I (65) boot.esp32: SPI Speed      : 40MHz
[19:57:42]I (70) boot.esp32: SPI Mode       : DIO
[19:57:42]I (74) boot.esp32: SPI Flash Size : 4MB
[19:57:42]I (79) boot: Enabling RNG early entropy source...
[19:57:42]I (84) boot: Partition Table:
[19:57:42]I (88) boot: ## Label            Usage          Type ST Offset   Length
[19:57:42]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[19:57:42]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[19:57:42]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[19:57:42]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[19:57:42]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[19:57:42]I (133) boot: End of partition table
[19:57:42]I (137) boot: No factory image, trying OTA 0
[19:57:42]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=65914h (416020) map
[19:57:42]I (301) esp_image: segment 1: paddr=0007593c vaddr=3ffbdb60 size=051bch ( 20924) load
[19:57:42]I (310) esp_image: segment 2: paddr=0007ab00 vaddr=40080000 size=05518h ( 21784) load
[19:57:42]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=1369dch (1272284) map
[19:57:42]I (779) esp_image: segment 4: paddr=001b6a04 vaddr=40085518 size=19a6ch (105068) load
[19:57:42]E (823) esp_image: Image length 1836192 doesn't fit in partition length 1835008
[19:57:42]E (823) boot: OTA app partition slot 0 is not bootable
[19:57:42]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[19:57:42]E (834) boot: OTA app partition slot 1 is not bootable
[19:57:42]E (839) boot: No bootable app partitions in the partition table
[19:57:42]ets Jun  8 2016 00:22:57
[19:57:42]
[19:57:42]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:57:42]configsip: 188777542, SPIWP:0xee
[19:57:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:57:42]mode:DIO, clock div:2
[19:57:42]load:0x3fff0030,len:6652
[19:57:42]ho 0 tail 12 room 4
[19:57:42]load:0x40078000,len:15052
[19:57:42]load:0x40080400,len:3836
[19:57:42]entry 0x4008069c
[19:57:42]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[19:57:42]I (57) boot: compile time 18:44:21
[19:57:42]I (57) boot: Multicore bootloader
[19:57:42]I (61) boot: chip revision: v1.1
[19:57:42]I (65) boot.esp32: SPI Speed      : 40MHz
[19:57:42]I (70) boot.esp32: SPI Mode       : DIO
[19:57:42]I (74) boot.esp32: SPI Flash Size : 4MB
[19:57:42]I (79) boot: Enabling RNG early entropy source...
[19:57:42]I (84) boot: Partition Table:
[19:57:42]I (88) boot: ## Label            Usage          Type ST Offset   Length
[19:57:42]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[19:57:42]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[19:57:42]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[19:57:42]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[19:57:42]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[19:57:42]I (133) boot: End of partition table
[19:57:42]I (137) boot: No factory image, trying OTA 0
[19:57:42]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=65914h (416020) map
[19:57:42]I (301) esp_image: segment 1: paddr=0007593c vaddr=3ffbdb60 size=051bch ( 20924) load
[19:57:42]I (310) esp_image: segment 2: paddr=0007ab00 vaddr=40080000 size=05518h ( 21784) load
[19:57:42]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=1369dch (1272284) map
[19:57:42]I (779) esp_image: segment 4: paddr=001b6a04 vaddr=40085518 size=19a6ch (105068) load
[19:57:43]E (823) esp_image: Image length 1836192 doesn't fit in partition length 1835008
[19:57:43]E (823) boot: OTA app partition slot 0 is not bootable
[19:57:43]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[19:57:43]E (834) boot: OTA app partition slot 1 is not bootable
[19:57:43]E (839) boot: No bootable app partitions in the partition table
[19:57:43]ets Jun  8 2016 00:22:57
[19:57:43]
[19:57:43]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:57:43]configsip: 188777542, SPIWP:0xee
[19:57:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:57:43]mode:DIO, clock div:2
[19:57:43]load:0x3fff0030,len:6652
[19:57:43]ho 0 tail 12 room 4
[19:57:43]load:0x40078000,len:15052
[19:57:43]load:0x40080400,len:3836
[19:57:43]entry 0x4008069c

```

kuligs2, 2024-08-08:
> I'm having the same issue on a brand-new device. My settings are identical to https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml (except for api and wifi settings [...]

yeah i bought it too few days ago, At first it flashed somewhat ok, but when i unplug it, it reverts back to the default state (no lights, does nothing). Then i had to flash it several times to get it to work with voice assist, but still when i unplug it reverts back to default state.. and in the ESP home i always have multiple configurations showing up and the "adopt" button for the atom device with default ESP32 configuration/yaml. The adopt thing never go away even if i press it.

The ESP32 box 3 didnt had so many problems, flashed voice assistant on second try. The first one was blank white screen and voice assist didnt work. The screen turn on/off worked but whenever it is triggered the speaker would pop. This popping is still present even with correct voice assist on the box3 (with that smiley face, not the blank white screen).

Anyways.. is there a way to maybe not use these low powered devices but use android phone instead? I got many layin - e-wasting. It would be nice to put them to a good use.

My main gripe is that i want hands free experience with the assist cuz of disabilities.

rmeissn, 2024-08-12:
I flashed an Atom Echo yesterday through https://esphome.io/projects/ and it resulted in a boot loop problem - at least I guess so from the lights.
It wasn't possible to activate it by saying "ok nabu", nor through the button on the top and often lighted in red for half a second.

kuligs2, 2024-08-12:
> I flashed an Atom Echo yesterday through https://esphome.io/projects/ and it resulted in a boot loop problem - at least I guess so from the lights. It wasn't possible to activate it by saying "ok nabu", nor through the button on the top and often lighted in red for half a second.

I dont know exactly what i did but i got it working somehow.. Steps i took involved:

Install fw for first time use, 
COnnect to wifi
Create the new yaml for voice assistant, using the template from github
install it wirelessly from esphome page in homeassistant
If it fails repeat the process..

ndurchx, 2024-08-13:
Same happens for me. It occured, after the update which could not be installed ota, because of the new partition size. Then updated with "Plug into the computer running ESPHome Dashboard"
Config:
```
substitutions:
  name: echo1-238c04
  friendly_name: Atom Echo 1 238c04
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

ndurchx, 2024-08-13:
Turns out, it is already fixed in #227. Use the m5stack-atom-echo.adopted.yaml config

kuligs2, 2024-08-13:
> Turns out, it is already fixed in #227. Use the m5stack-atom-echo.adopted.yaml config

how can it be fixed if this happened to me last week?

ndurchx, 2024-08-14:
> how can it be fixed if this happened to me last week?

Just use the new config and write it via usb.

rmeissn, 2024-08-17:
I was able to get my m5stack working again by using the yaml file included in this repository (normal, not adopdted) and esphome docker container.

In contrast to what is written on https://esphome.io/projects/?type=voice I got the impression that the following firmware is used https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml , not the one from this repository. Because in this repository, there is no on-device wake-word engine on the latest main branch, but a set up device from https://esphome.io/projects/?type=voice contains the wake word engine -> This seems like a bug to me, even though having a local wake word on the device is desireable.
Is it possible to fix this somehow? So that https://esphome.io/projects/?type=voice installs the latest firmware from this repository, not some other repo.

closed 2025-05-09 by github-actions[bot]