esphome/firmware #258: Feature Request: Add the possibility to cast the tts response to media player devices

lajoshanko, 2024-08-26:
Since the M5Stack Echo and the ESP32 Box devices has pretty quiet speakers it would be nice to have a feature toggle to cast the voice response the media player devices.

Currently as a workaround the following config can be added to the esphome device config to cast the response to the device.  
Where the entity_id is optionally populated from a template select from home assistant.
```on_tts_end:
    - homeassistant.service:
        service: media_player.play_media
        data:
          entity_id: media_player.media_player_entity
          media_content_id: !lambda 'return x;'
          media_content_type: music
          announce: "true" 
```
Features on the esphome device config page:

1. A feature toggle to turn the casting on/off.
2. A select list of mediaplayers where the state is not off, unavailable, unknown.
3. Maybe a feature toggle to mute the esphome device built-in speaker.

igorsantos07, 2024-10-21:
While this is definitely an awesome idea (and I'm a complete newbie to ESP32 stuff), shouldn't this be pointed to the HA repository (as it's the one processing the request, generating the response and probably sending it back to the satellite) instead of the satellite firmware repo?

igorsantos07, 2025-05-01:
https://nostalebots.xyz