esphome/firmware #264: 2024.9.0 update breaks voice-assistant atom echo [mode] is an invalid option for [speaker.i2s_audio]. Did you mean [i2s_mode]?

carrierone, 2024-09-19:
Both m5stack-atom-echo.yaml and m5stack-atom-echo.adopted.yaml need to be updated for the new ESPHome 2024.9.0 which has a breaking change on i2s options.

INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-168750.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
Failed config

speaker.i2s_audio: [source /data/packages/c46f54c1/voice-assistant/m5stack-atom-echo.yaml:71]
  platform: i2s_audio
  id: echo_speaker
  i2s_dout_pin: GPIO22
  dac_type: external
  
  [mode] is an invalid option for [speaker.i2s_audio]. Did you mean [i2s_mode]?
  mode: mono

ninjaasok, 2024-09-19:
Can verify that, same for me.

ewaucq, 2024-09-19:
using a different source:
`source /data/packages/c46f54c1/voice-assistant/m5stack-atom-echo.adopted.yaml:45`

but same error

r0l01, 2024-09-19:
Seems like the problem still exists.

INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-a158b0.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
Failed config

speaker.i2s_audio: [source /data/packages/c46f54c1/voice-assistant/m5stack-atom-echo.adopted.yaml:45]
  platform: i2s_audio
  id: echo_speaker
  i2s_dout_pin: GPIO22
  dac_type: external
  
  [mode] is an invalid option for [speaker.i2s_audio]. Did you mean [i2s_mode]?
  mode: mono

jesserockz, 2024-09-19:
@r0l01 Packages by default have a 24 hour cache. Unless you delete the local cached folder or change the refresh time (its in the packages documentation) your local esphome install wont check this repo for new files for another 24 hours.

7542, 2024-09-19:
I am confused ... i still get the old problem! What do i have to do for solving? Pleeease help.

knemo20000, 2024-09-20:
@7542  Change mode: mono to channel: mono

gunarser, 2024-09-20:
I can confirm that now the issue is fixed for me. I was able to update my Atom Echo.

jaybe-dk, 2024-09-21:
I have the same issue, probably caused by mu local cache - where is the cache located?

c-def, 2024-09-21:
> @7542 Change mode: mono to channel: mono

HOW

jaybe-dk, 2024-09-23:
After two days and an ESPhome update, I still get the error (so guess it's not a 24-hour cache issue) - how can I fix this?

```
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/m5stack-atom-echo-11132c.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
Failed config

speaker.i2s_audio: [source /data/packages/43316764/voice-assistant/m5stack-atom-echo.yaml:57]
  platform: i2s_audio
  id: echo_speaker
  i2s_dout_pin: GPIO22
  dac_type: external
  
  [mode] is an invalid option for [speaker.i2s_audio]. Did you mean [i2s_mode]?
  mode: mono
```

carrierone, 2024-09-23:
Click the ... on your esphome item in homeassistant,  Clean Build Files, and then try to install again

c-def, 2024-09-23:
> Click the ... on your esphome item in homeassistant, Clean Build Files, and then try to install again

When I click on the device in question and Clean Build Files, I get the exact same error.

jaybe-dk, 2024-09-23:
> > Click the ... on your esphome item in homeassistant, Clean Build Files, and then try to install again
> 
> When I click on the device in question and Clean Build Files, I get the exact same error.

Tried  that too - same error, yes.

jpalbee, 2024-09-24:
This issue is not closed. I have the same error.

knemo20000, 2024-09-24:
@c-def @jaybe-dk @jpalbee
In ESPHome, click on "EDIT" for the device in question.  Look for mode: mono

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    **mode: mono**

change it to read channel: mono and click SAVE

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    **channel: mono**

Once saved, click on INSTALL and select Wirelessly

jpalbee, 2024-09-24:
There is no "Mode" to alter:

substitutions:
  name: museproto1
  friendly_name: Muse
packages:
  raspiaudio.muse-proto: github://esphome/firmware/media-player/raspiaudio-muse-proto.yaml@main    # not working
  #raspiaudio.muse-proto: github://esphome/firmware/blob/main/media-player/raspiaudio-muse-proto.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 7OH6FWonYQyMCGQeHfp2cQyjLtgMnMLT7Q8+Jt6185g=
ota:
  - platform: esphome
    password: 'pw'
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  manual_ip:
#    # Set this to the IP of the ESP
#    static_ip: 192.168.1.13
#    # Set this to the IP address of the router. Often ends with .1
#    gateway: 192.168.1.1
#    # The subnet of the network. 255.255.255.0 works for most home networks.
#    subnet: 255.255.255.0
sensor:
  - platform: adc
    pin: GPIO33
#    raw: true
    name: "Muse VCC"     #
    filters:
      - multiply: 3.3

jpalbee, 2024-09-24:
There is no "Mode"

`substitutions:
  name: museproto1
  friendly_name: Muse
packages:
  raspiaudio.muse-proto: github://esphome/firmware/media-player/raspiaudio-muse-proto.yaml@main    # not working
  #raspiaudio.muse-proto: github://esphome/firmware/blob/main/media-player/raspiaudio-muse-proto.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 7OH6FWonYQyMCGQeHfp2cQyjLtgMnMLT7Q8+Jt6185g=
ota:
  - platform: esphome
    password: !secret pw245
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: adc
    pin: GPIO33'

jaybe-dk, 2024-09-24:
This is mine - also no mode mono:

´´´
substitutions:
  name: m5stack-atom-echo-11132c
  friendly_name: M5Stack Atom Echo Køkken
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@cd57ca6f951d44cc5bf61de124a15b349ef1f9a4
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: MvUfPWMn5p6JUpcBOcXxDLq/5Gh04NfzUwZ8QNV4hXY=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

´´´

knemo20000, 2024-09-24:
@jpalbee @jaybe-dk  The channel: mono command is actually located in the yaml file being downloaded by the packages: raspiaudio.muse-proto: github://esphome/firmware/media-player/raspiaudio-muse-proto.yaml@main and packages: m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@cd57ca6f951d44cc5bf61de124a15b349ef1f9a4 commands respectively.  I would suggest adding the speaker: block to the code that you do have, save the file and try installing it again (the dot in front of the word platform should actually be a dash -).

speaker:

  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    channel: mono

jpalbee, 2024-09-25:
@knemo20000 If it is located in the online "package, " why is it not updated there? 
I tried moving that block into my code and received the same failure.

c-def, 2024-09-25:
@jpalbee they haven't come right out and said it yet, but essentially this broke the configuration on our devices and now I'm pretty certain we'll have to reinstall from scratch via USB on all impacted devices. This is not the first time this has happened and I doubt it will be the last.

helicopterjeff, 2024-09-25:
I've got this issue as well, i've updated the speaker block to match above. same thing.

Reinstalling via USB doesn't work because this won't compile. (preparation failed_

@knemo20000 if the yaml we need to modify is in those files, how do i modify those? my main config for the device has them already.

jpalbee, 2024-09-26:
@jesserockz I am not finding this fixed. What am I missing?

c-def, 2024-09-26:
> I've got this issue as well, i've updated the speaker block to match above. same thing.
>
> Reinstalling via USB doesn't work because this won't compile. (preparation failed)
>
> @knemo20000 if the yaml [...]

Try going to https://esphome.io/projects/ and installing via USB directly through the website.

Or you can try doing the same thing from this site: https://raspiaudio.github.io/

dstengele, 2024-09-26:
In my case, I was able to solve this by reinstalling the ESPHome addon. **Make sure to backup your configurations beforehand!**

facuzestoa, 2024-09-28:
Same issue, why this case is closed? I can see it is updated on GitHub but still looks espHome is not updating packages
How to get local packages refreshed ? How to locate folder where packages are stored ?

helicopterjeff, 2024-09-30:
> Try going to https://esphome.io/projects/ and installing via USB directly through the website.
>
> Or you can try doing the same thing from this site: https://raspiaudio.github.io/

The esphome.io/projects site worked perfectly!

Jonah1970, 2024-10-05:
I tried the the https://esphome.io/projects/ link twice and still get the same error:

```INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome/m5stack-atom-echo-80cbb4.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
Failed config

speaker.i2s_audio: [source /config/esphome/m5stack-atom-echo-80cbb4.yaml:51]
  platform: i2s_audio
  id: echo_speaker
  i2s_dout_pin: GPIO22
  dac_type: external
  
  [mode] is an invalid option for [speaker.i2s_audio]. Did you mean [i2s_mode]?
  mode: mono
```

PR #265 (2024-09-19): Updates for ESPHome 2024.9.0
Closes #264