esphome/firmware #272: in progress binary sensor...

RaceNJason, 2024-10-03:
With the latest home assistant core update, my ATOM Echo voice assistant now shows this notification:

This stops working in version 2025.4. Please address before upgrading.

The ESPHome in progress binary sensor binary_sensor.atom_echo_master_bedroom_assist_in_progress is deprecated.

Migrate your configuration to use the corresponding assist_satellite entity and then click SUBMIT to disable the in progress binary sensor and fix this issue.

daknightuk, 2024-10-03:
I have exactly the same issue, not sure whether to disable the in progress binary sensor or not?

[image]

RaceNJason, 2024-10-03:
I would not click the 'SUBMIT' button just yet. Prefer to wait to see this fixed (in which case the message may go away on it's own). Sounds like if it isn't fixed and you click on it, it could do more than just ignore the warning (like ignore in_progress messages thinking assist_satellite messages are now coming in)??

sanderlv, 2024-10-04:
I have the same error. Even after updating to the latest template.

https://github.com/esphome/media-players/blob/main/m5stack/m5stack-atom-echo.yaml

danerthompson, 2024-10-04:
I also have this error with a custom S3 config.

tduzzi, 2024-10-04:
I too have this error M5Atom Echo

Powerbless, 2024-10-04:
Same error

Menz01, 2024-10-05:
same issue here

jparchem, 2024-10-05:
Also same issue with both my Atom M3 Echo, VoIP, S3 Box and Wyoming satellites.   The error message itself is included in a release notes screenshot:  https://www.home-assistant.io/blog/2024/10/02/release-202410/#the-source-of-a-repair

Wummeke, 2024-10-05:
I had the same issue with my RaspiAudio Muse Luxe. I solved it by removing the `voice-assistant:` component from my ESPHome configuration. I don't really use it to be fair. after OTA-updating my ESPHome device and a repair in Home-Assistant, the warning was gone. If you do want to use the assistant functionality, you might wanna have alook at these configurations: https://github.com/esphome/firmware/tree/main/voice-assistant I tried that first, but I got all kinds of buffer errors in the ESPHome Logs and (for me the most important reason) the media-player functionality of my Muse Luxe was gone.

Nismonx, 2024-10-05:
Same here, 
Updating to 2024.9.2 via ESPhome seems to fix the issue.

jmcbri, 2024-10-05:
> Same here, Updating to 2024.9.2 via ESPhome seems to fix the issue.

Thanks.  How do you do thank?   I puttered around in various places in gui and didn't see an option.  Command line?

tduzzi, 2024-10-05:
That did not work for me as I am currently running 2024.9.2 and still have this warning:

Todd

kevvar, 2024-10-06:
When you click on 'Submit', it fixes the issue. Disables the binary sensor, and it automatically creates the assist satellite entity.

ericvb, 2024-10-06:
> When you click on 'Submit', it fixes the issue. It disables the binary sensor, and it automatically creates the assist satellite entity.

The explanatory text is then very unhappily worded : Migrate your configuration to use the corresponding assist_satellite entity and then click SUBMIT to disable the in progress binary sensor and fix this issue.
???

kevvar, 2024-10-06:
Yes you are right.

OzGav, 2024-10-07:
I tentatively clicked on SUBMIT and yes issue fixed for me. I also think the text should be `Migrate your configuration to use the corresponding assist_satellite entity and disable the in progress binary sensor by clicking SUBMIT to fix this issue.`

deejay1, 2024-10-08:
unfortunately there is no documentation on how to migrate the config

OzGav, 2024-10-08:
@deejay1 as explained above just click submit

weboide, 2024-10-08:
It's also tracked in https://github.com/esphome/wake-word-voice-assistants/issues/26

jlpouffier, 2024-10-09:
There is no config to migrate ;) 

The repair is just here because you could be using the `binary_sensor` `assist_in_progress` to automate or display something in your dashboards.

If you are **NOT** using the `binary_sensor` `assist_in_progress`, then simply click `SUBMIT` on the repair. We will disable  the `binary_sensor` `assist_in_progress` for you.

If you **ARE** using the `binary_sensor` `assist_in_progress`, then there is a new kid on the block called the `assist_satelitte` entity.

The `assist_satelitte` is great because it retains all the functionally of the old `binary_sensor`, and then add on top more data.

When the `binary_sensor` was simply 
- on (When something was happening)
- off.

The `assist_satelitte` has now 4 states
- idle
- listening
- processing
- responding

So you can do (Automate or display things) more, and more granularly.

By the way, if you still need a binary indication that something is happening (For example, I do - I display a very small badge in my dashboard when one of my voice assistant is active), then you can easily recreate the binary_sensor with a template.

Just go to Settings > Devices and Services > Helpers > Create Helper > Template > Binary Sensor.
Here is the template 
```
{{states('assist_satellite.s3box_chambre') in ['listening','responding','processing'] }}
```

✌🏻 JLo

jkosharek, 2025-02-04:
> There is no config to migrate ;)
>
> The repair is just here because you could be using the `binary_sensor` `assist_in_progress` to automate or display something in your dashboards.
>
> If you are **NOT** [...]

So would you let the 'fix' disable the binary_sensor.ha_voice_master_assist_in_progres sensor and then just created the template sensor like this in my case? 

{{states('assist_satellite.ha_voice_master') in ['listening','responding','processing'] }}

jesserockz, 2025-02-10:
I am closing this as it has nothing to do with the actual firmware and is something that is entirely on the Home Assistant side.

closed 2025-02-10 by jesserockz