esphome/firmware #276: Enable all sensors from the dock (ESP32-S3-BOX-3-SENSOR)

igorsantos07, 2024-10-21:
> It looks like the [more specific repo](https://github.com/esphome/wake-word-voice-assistants/issues/14) would be the correct place to report this, but given this repository has more people looking into it, and lots of reports which seem specific to the voice assistant in the ESP32-S3-BOX-3, I'm cross-posting. I apologize if this was definitely not the best idea!

The S3 can optionally come with a sensors dock, which includes temperature, humidity and what seems to be a mmWave radar. It also has an IR emitter/receiver, which I'm not sure could be included as well, although I bet it would be way more complex and thus, "out of scope".
[This is the version I bought @ AliExpress](https://www.aliexpress.com/item/1005006169541505.html).

It's sad to notice no tutorial explains this wouldn't work with the firmware we're so far able to install on it. Thus, I wonder if it would be possible to implement those sensors?

[image]

jeremypoulter, 2024-11-14:
FYI https://github.com/AlmostInteractive/ESP32-S3-Box-3-Voice-Assistant-Sensor-Dock looks to work well for me with this dock

igorsantos07, 2025-05-01:
https://nostalebots.xyz

jamesbiederbeck, 2025-05-17:
I normally hate +1 comments but I'd hate for this issue to go stale...

igorsantos07, 2025-05-17:
@jamesbiederbeck let this issue rot and move your S3 to one of the firmwares below:
- https://github.com/AlmostInteractive/ESP32-S3-Box-3-Voice-Assistant-Sensor-Dock, as suggested above
- https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome: the one I'm using, quite well-rounded and VERY well documented.

closed 2025-06-22 by github-actions[bot]