esphome/firmware #286: Unable to compile for M5Stack Atom Echo

Taomyn, 2024-12-18:
Upgraded my Home Assistant ESPHome to 2024.12.0 and I cannot compile the firmware for my M5Stack Atom Echo - I have other ESPHome devices that have all been successful so hopefully my environment is ok.

The messages during the compile are:

```
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/m5ae-31b338.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5ae-31b338 (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: 
Following dependencies have new versions available:
Dependency "espressif/esp-nn": "1.0.2" -> "1.1.0"
Consider running "idf.py update-dependencies" to update your lock file.
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/m5ae-31b338/.pioenvs/m5ae-31b338/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 3.40 seconds ==========================
```
My yaml config for the device, working fine until now, is:

```
substitutions:
  name: "m5ae-31b338"
  friendly_name: M5Stack Atom Echo 31b338
  micro_wake_word_model: hey_jarvis # okay_nabu, hey_jarvis and hey_mycroft are supported
packages:
#  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
#  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@cd57ca6f951d44cc5bf61de124a15b349ef1f9a4
#  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.adopted.yaml@main
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/wake-word-voice-assistant/m5stack-atom-echo.adopted.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <redacted>
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .mydomain.com
  on_connect:
    - delay: 5s  # Gives time for improv results to be transmitted
```

gunarser, 2024-12-18:
The same issue here

Bandit3112011, 2024-12-18:
The same issue here ESP32-S3-Box 3

5high, 2024-12-18:
I am M5Stack Atom Echo too, stack here:
```
INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/m5stack-atom-echo-a14fbc.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Cloning https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 213, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 231, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 341, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 307, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 716, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 67, in clone_or_update
    run_git_command(cmd)
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: fetch-pack: invalid index-pack output

```

tradosh, 2024-12-18:
I got this error:
raceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 213, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 231, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 341, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 307, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 811, in copy_files
    shutil.copytree(
  File "/usr/lib/python3.11/shutil.py", line 561, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/shutil.py", line 515, in _copytree
    raise Error(errors)
shutil.Error: [('/data/idf_components/b0ad5728/components/esp-sr/docs/myapp/lib64', '/data/build/esphome-web-0ae798/components/esp-sr/docs/myapp/lib64', "[Errno 17] File exists: 'lib' -> '/data/build/esphome-web-0ae798/components/esp-sr/docs/myapp/lib64'"), ('/data/idf_components/b0ad5728/components/esp-sr/docs/myapp/bin/python', '/data/build/esphome-web-0ae798/components/esp-sr/docs/myapp/bin/python', "[Errno 17] File exists: 'python3' -> '/data/build/esphome-web-0ae798/components/esp-sr/docs/myapp/bin/python'"), ('/data/idf_components/b0ad5728/components/esp-sr/docs/myapp/bin/python3', '/data/build/esphome-web-0ae798/components/esp-sr/docs/myapp/bin/python3', "[Errno 17] File exists: '/usr/bin/python3' -> '/data/build/esphome-web-0ae798/components/esp-sr/docs/myapp/bin/pyth

tradosh, 2024-12-18:
I think there are updates in the back end, as I now have a new error:
```
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/esphome-web-0ae798.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing esphome-web-0ae798 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-0ae798/.pioenvs/esphome-web-0ae798/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.62 seconds ==========================```

tradosh, 2024-12-18:
a workaround that seems to help me to unblock it is adding this inside the esp32 section
```
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

gunarser, 2024-12-18:
The workaround worked for me too.

> a workaround that seems to help me to unblock it is adding this inside the esp32 section
> 
> ```
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```

gfxdude-dus, 2024-12-19:
getting error:
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package
then 🥲

gfxdude-dus, 2024-12-19:
tried several versions, but getting an error again, with 5.4.0, 5.3.0

Platform Manager: Installing platformio/espressif32 @ 5.3.0
INFO Installing platformio/espressif32 @ 5.3.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.3.0 has been installed!
INFO espressif32@5.3.0 has been installed!
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

Processing m5stack-atom-echo (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

🥲

bsjouke, 2024-12-20:
> a workaround that seems to help me to unblock it is adding this inside the esp32 section
> 
> ```
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```

Where should I add this? Placing it in the devices yaml file from the esphome gui I get "Component not found: framework."

Taomyn, 2024-12-20:
> > a workaround that seems to help me to unblock it is adding this inside the esp32 section
> > ```
> >   framework:
> >     type: esp-idf
> >     version: 4.4.8
> >     platform_version: 5.4.0
> > ```
> 
> Where should I add this? Placing it in the devices yaml file from the esphome gui I get "Component not found: framework."

I added it at the end, it needs to go under the esp32 section as I didn't have one for my yaml:

```
esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

bsjouke, 2024-12-20:
> I added it at the end, it needs to go under the esp32 section as I didn't have one for my yaml:
>
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```

That worked like a charm. Thank you!

mroxso, 2024-12-20:
DId not work for my RaspiAudio Muse Luxe yet.
#287

chewara, 2024-12-20:
> I added it at the end, it needs to go under the esp32 section as I didn't have one for my yaml:
>
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```

worked for me too, but had this : 
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.

Clasper101, 2024-12-21:
Adding it to my M5Stack Atom Echo worked, but it didn't work for my ESP32 S3 Box3,
I got this error
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/voice-lounge/.pioenvs/voice-lounge/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

Has anyone got the S3 Box 3 to update successfully?

Bandit3112011, 2024-12-21:
@Clasper101  work fine for my  ESP32S3 box 3

darinlarimore, 2024-12-21:
> esp32:
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0

Thanks, this worked for my M5Stack Atom Echo too!

AndersRipa, 2024-12-21:
I have the same issue.

smith844, 2024-12-22:
I also have same issue.  The lines `esp32:
framework:
type: esp-idf` 
were already in the YAML, so I added
`version: 4.4.8
platform_version: 5.4.0`
but it will still not compile.
Will there be a fix to the source file coming ?

hum42, 2024-12-22:
...same issue here

GitHubxtsnl, 2024-12-23:
running ESPHome version 2024.12.2 inside HA 2024.12.5 with the same error.
Validation runs fine and configuration is valid.
added the following options to my .yaml:
```
    version: 4.4.8
    platform_version: 5.4.0
```

and luckily this helps, I can compile and flash now.

jaycollett, 2024-12-23:
Workaround fixed the issue for me as well, hoping this gets appropriately resolved soon.

GerardPolloRebozado, 2024-12-26:
This workaround fixed the issue but is it normal that it takes a long time to compile around 25 minutes

Clasper101, 2024-12-27:
That seems excessive, I've just moved mine to a new HA instance and the compiling took 319 seconds.

GerardPolloRebozado, 2024-12-27:
> That seems excessive, I've just moved mine to a new HA instance and the compiling took 319 seconds. 

Do you have a lot of resources assigned to the vm? I set 10gb of ram and 4cpu

Clasper101, 2024-12-27:
I'm using HAOS on a mini pc (N100) with 16gb of ram.

GerardPolloRebozado, 2024-12-28:
> I'm using HAOS on a mini pc (N100) with 16gb of ram. 

Then I don't understand my computer is more powerful than a mini PC. Could you post your yml?

Clasper101, 2024-12-28:
Here is my yml for an Atom Echo 
substitutions:
  name: "kitchen-voice"
  friendly_name: Kitchen Voice
packages:
  m5stack.atom-echo-wake-word-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  framework: 
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

and for an ESP32 S3 Box 3

substitutions:
  name: new-voice
  friendly_name: New Voice
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.xxx.xx.xxx

esp32:
  framework: 
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0 

Use address for when it doesn't automatically connect to wifi after compiling.
I hope that this helps.

adoucette, 2024-12-28:
So, if I wanted to change the wakeword to "hey jarvis", would I just edit the @Clasper101 .yaml as above to the following?

`substitutions:
  name: xxxxxxxxxxxxxxxxxxxxxxxx
  friendly_name: xxxxxxxxxxxxxxxxxxxxxxxx
  micro_wake_word_model: hey_jarvis  # alexa, okay_nabu, hey_jarvis, hey_mycroft are also supported
    
packages:
  m5stack.atom-echo-wake-word-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0`

Clasper101, 2024-12-29:
@adoucette, that's a good question. I haven't tried to change it yet.

GerardPolloRebozado, 2024-12-30:
> Here is my yml for an Atom Echo substitutions: name: "kitchen-voice" friendly_name: Kitchen Voice packages: m5stack.atom-echo-wake-word-voice-assistant: [...]

Even with you config it takes a lot to ocmpile

geiseri, 2025-01-02:
The change that @GitHubxtsnl  suggested fixed it for me.  Is it possible that the adf PR is not completely compatible with the esp-idf 5.1.5 version?  @jesserockz might PR5230 need to be tweaked?

jonthewise, 2025-01-03:
Adding this to the ESP32 section worked for me as well.

  version: 4.4.8
  platform_version: 5.4.0

szhorvath, 2025-01-04:
I'm experiencing the same issue, but I had to use the `m5stack-atom-echo.adopted.yaml` configuration file; otherwise, I encountered a memory maxed-out error.

```yaml
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.adopted.yaml@main

esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

dixonbg, 2025-01-04:
I can only assume, that whoever used to work on Echo is now working on the new Voice Assistant Box. And s3 box before that. I am waiting patiently for them to release the next ESPHome that should include the new ESP-IDF MediaPlayer that can use I2S Speaker as platform. This is what I see implemented in the code for Voice Assistant but it is missing in current, beta and dev version of ESPHome (but it is mentioned as "planned" in Music Assistant docs and in some docs in GitHub). Similarly, the incompatibility with ESP-IDF 5.1.5 is also fixed in Voice Assistant, but not in ESPHome. So someone should port those fixes. But I guess this can happen only when the commercial product is stable enough to start working on the next esphome.

devinhedge, 2025-01-10:
I'm having the same issue and trying to debug it.

It seems that the package is referring to a library that has been deprecated and no longer exists. I'm trying to find an old version of the library to see what it was and why some are able get it from the by overriding by referring to the expressif libraries.

mike5WA, 2025-01-14:
Running esp32 s3 and today failed to install code as it could not connect to a git hub site with error 500. Added the entries above viz version 4.4.8 and platform version 5.4.0  seems to have fixed things. Interestingly the issue only presented itself today.
Mike

YewTreeWeb, 2025-01-24:
I was having the same issue and by using:
```
esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```
I was able to prepare a download file to install on my Atom Echos

jcd420, 2025-01-24:
> I was having the same issue and by using:
> 
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```
> 
> I was able to prepare a download file to install on my Atom Echos

this worked for me

jesserockz, 2025-02-10:
Can you please let me know if the **original error this issue is showing** is still applying as the files on GitHub were updated 3 weeks ago with https://github.com/esphome/wake-word-voice-assistants/pull/46 and nobody should be seeing this error anymore. 

@dixonbg The work is almost all ported into ESPHome now and will be done before ESPHome 2025.2.0.

Any local yaml with packages for the echo with wake-word voice-assistant should be pointing to the updated repo source: [github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml).

The YAMLs will be updated in the coming weeks after the ESPHome release to take advantage of the new Media player code and speaker platforms with no reliance on esp-adf which has caused many issues.

I will be closing this issue at the end of this week unless it is shown to be not fixed.

Jesse

Taomyn, 2025-02-10:
@jesserockz not sure exactly what I needed to do, so I changed the repo to the one you mentioned and also remarked out the forced esp32 versions:

````substitutions:
  name: "m5ae-31b338"
  friendly_name: M5Stack Atom Echo 31b338
  micro_wake_word_model: hey_jarvis # okay_nabu, hey_jarvis and hey_mycroft are supported
packages:
  # m5stack.atom-echo-voice-assistant: github://esphome/firmware/wake-word-voice-assistant/m5stack-atom-echo.adopted.yaml@main
  m5stack.atom-echo-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret m5ae_key
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.39
    subnet: 255.255.255.0
    gateway: 192.168.1.1
    dns1: 192.168.1.11
    dns2: 192.168.1.71
  domain: !secret dns_suffix
  on_connect:
    - delay: 5s  # Gives time for improv results to be transmitted
#esp32:
#  framework:
#    type: esp-idf
#    version: 4.4.8
#    platform_version: 5.4.0
````

I then cleared out the build files and rebuilt it - there's still a warning at the top about versions but it succeeded and was able to upload to my device

[logs_m5ae-31b338_run.txt](https://github.com/user-attachments/files/18731011/logs_m5ae-31b338_run.txt)

jesserockz, 2025-02-10:
Yeah the newer yaml has the version overrides until the new stuff comes out with ESPHome 2025.2.

I see it got to detecting wake work, is it working correctly?

Taomyn, 2025-02-10:
> I see it got to detecting wake work, is it working correctly?

I'll let you know later as I'm not at home and did the change remotely, it was working this morning before I left.

Kalekulan, 2025-02-10:
I have tried it and everything works as expected. Thanks for the fix! :)

AndersRipa, 2025-02-10:
I have removed the fix and recompiled with ESPHome Device Builder version: 2024.12.4 
It compiles with no problems.
Thank you for the fix.

gunarser, 2025-02-10:
It works without workaround here as well. Thanks!

> Can you please let me know if the **original error this issue is showing** is still applying as the files on GitHub were updated 3 weeks ago with [esphome/wake-word-voice-assistants#46](https://github.com/esphome/wake-word-voice-assistants/pull/46) and nobody should be seeing this error anymore.
> 
> [@dixonbg](https://github.com/dixonbg) The work is almost all ported into ESPHome now and will be done before ESPHome 2025.2.0.
> 
> Any local yaml with packages for the echo with wake-word voice-assistant should be pointing to the updated repo source: [github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml).
> 
> The YAMLs will be updated in the coming weeks after the ESPHome release to take advantage of the new Media player code and speaker platforms with no reliance on esp-adf which has caused many issues.
> 
> I will be closing this issue at the end of this week unless it is shown to be not fixed.
> 
> Jesse

Clasper101, 2025-02-10:
Thanks Jesse, I have tried with one of mine and it updated successfully. Kind regards Andrew Sent from my Galaxy

HarlemSquirrel, 2025-02-10:
> Can you please let me know if the **original error this issue is showing** is still applying as the files on GitHub were updated 3 weeks ago with [...]

Worked for me on a Home Assistant Yellow

jesserockz, 2025-02-13:
Thanks for the confirmations 🙏

closed 2025-02-13 by jesserockz

Taomyn, 2025-02-19:
@jesserockz for 2025.2.0 do we need to change anything? It won't compile now or do I need to create a new issue?

```
Compiling .pioenvs/m5ae-31b338/src/esphome/components/audio/audio_decoder.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
src/esphome/core/helpers.h:85:59: note:   template argument deduction/substitution failed:
src/esphome/components/audio/audio_decoder.cpp:79:81: error: template argument 1 is invalid
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
src/esphome/components/audio/audio_decoder.cpp:80:25: error: base operand of '->' is not a pointer
       this->wav_decoder_->reset();
                         ^~
src/esphome/components/audio/audio_decoder.cpp: In member function 'esphome::audio::FileDecoderState esphome::audio::AudioDecoder::decode_wav_()':
src/esphome/components/audio/audio_decoder.cpp:314:5: error: 'esp_audio_libs' has not been declared
     esp_audio_libs::wav_decoder::WAVDecoderResult result = this->wav_decoder_->decode_header(
     ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:9: error: 'result' was not declared in this scope
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
src/esphome/components/audio/audio_decoder.cpp:321:29: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                             ^~
src/esphome/components/audio/audio_decoder.cpp:321:68: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                    ^~
src/esphome/components/audio/audio_decoder.cpp:321:104: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                                                        ^~
src/esphome/components/audio/audio_decoder.cpp:323:49: error: base operand of '->' is not a pointer
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
Compiling .pioenvs/m5ae-31b338/src/esphome/components/audio/audio_reader.o
*** [.pioenvs/m5ae-31b338/src/esphome/components/audio/audio_decoder.o] Error 1
========================= [FAILED] Took 15.01 seconds =========================
```

gunarser, 2025-02-19:
I am getting the same error with v2025.2.0.

Taomyn, 2025-02-20:
I just tried again as I was going to grab the logs, but it now works - tons of warnings, but it compiled, uploaded and so far looks ok. I'm remote so can't actually test it really works, but that was the same before.

FerNerd, 2025-02-20:
I had problems with updating too, so i Did a fresh install m5stack-atom-echo via the esphome.io website. That works, but if i try to update it now i get:

INFO ESPHome 2025.2.0
INFO Reading configuration config/m5stack-atom-echo-0d3670.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.5.1
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing m5stack-atom-echo-0d3670 (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208

Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/home/user/esphome/config/.esphome/build/m5stack-atom-echo-0d3670/.pioenvs/m5stack-atom-echo-0d3670/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /home/user/.platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /home/user/.platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /home/user/.platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /home/user/.platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

... what am i doing wrong?? I tried a clean build too, but that did not help

eric1467, 2025-02-20:
`INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/atom-echo-1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing atom-echo-1 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- esp-audio-libs @ 1.1.1
Compiling .pioenvs/atom-echo-1/src/esphome/components/audio/audio_decoder.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/button/button.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/esp32/core.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
src/esphome/core/helpers.h:85:59: note:   template argument deduction/substitution failed:
src/esphome/components/audio/audio_decoder.cpp:79:81: error: template argument 1 is invalid
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
src/esphome/components/audio/audio_decoder.cpp:80:25: error: base operand of '->' is not a pointer
       this->wav_decoder_->reset();
                         ^~
src/esphome/components/audio/audio_decoder.cpp: In member function 'esphome::audio::FileDecoderState esphome::audio::AudioDecoder::decode_wav_()':
src/esphome/components/audio/audio_decoder.cpp:314:5: error: 'esp_audio_libs' has not been declared
     esp_audio_libs::wav_decoder::WAVDecoderResult result = this->wav_decoder_->decode_header(
     ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:9: error: 'result' was not declared in this scope
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
[... cut ...]
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/atom-echo-1/src/esphome/components/audio/audio_decoder.o] Error 1
========================== [FAILED] Took 3.52 seconds ==========================

[logs_atom-echo-1_run.txt](https://github.com/user-attachments/files/18887364/logs_atom-echo-1_run.txt)

`

eric1467, 2025-02-20:
same issue again with 2025.2.0

`INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/atom-echo-1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing atom-echo-1 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- esp-audio-libs @ 1.1.1
Compiling .pioenvs/atom-echo-1/src/esphome/components/audio/audio_decoder.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/button/button.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/esp32/core.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
src/esphome/core/helpers.h:85:59: note:   template argument deduction/substitution failed:
src/esphome/components/audio/audio_decoder.cpp:79:81: error: template argument 1 is invalid
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
src/esphome/components/audio/audio_decoder.cpp:80:25: error: base operand of '->' is not a pointer
       this->wav_decoder_->reset();
                         ^~
src/esphome/components/audio/audio_decoder.cpp: In member function 'esphome::audio::FileDecoderState esphome::audio::AudioDecoder::decode_wav_()':
src/esphome/components/audio/audio_decoder.cpp:314:5: error: 'esp_audio_libs' has not been declared
     esp_audio_libs::wav_decoder::WAVDecoderResult result = this->wav_decoder_->decode_header(
     ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:9: error: 'result' was not declared in this scope
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
[... cut ...]
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/atom-echo-1/src/esphome/components/audio/audio_decoder.o] Error 1
========================== [FAILED] Took 3.52 seconds ==========================
`

gunarser, 2025-02-20:
I can confirm that today it compiled successfully without configuration change. Also, devices work as expected.

Taomyn, 2025-02-20:
> I just tried again as I was going to grab the logs, but it now works - tons of warnings, but it compiled, uploaded and so far looks ok. I'm remote so can't actually test it really works, but that was the same before.

I spoke too soon, yes the device works and hears my "Hey Jarvis", but he no longer speaks. The log shows:

```
[17:59:58][D][esp-idf:000][ann_read]: E (15588753) HTTP_CLIENT: Connection failed, sock < 0

[17:59:58][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
[17:59:58][E][speaker_media_player:334]: The announcement pipeline's file reader encountered an error.
[17:59:58][D][esp-idf:000][ann_read]: E (15588796) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780

[17:59:58][D][esp-idf:000][ann_read]: E (15588803) esp-tls: Failed to open new connection

[17:59:58][D][esp-idf:000][ann_read]: E (15588809) TRANSPORT_BASE: Failed to open a new connection

[17:59:58][D][esp-idf:000][ann_read]: E (15588819) HTTP_CLIENT: Connection failed, sock < 0

[17:59:58][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
```

michaelkleinhenz, 2025-02-20:
> I can confirm that today it compiled successfully without configuration change. Also, devices work as expected. Thanks

So you just did a retry on the build? Because it still does not work for me with ESPHome 2025.2.0, no matter how many time I retry it..

gunarser, 2025-02-20:
@michaelkleinhenz I cleaned build files on the device that failed previously. But I just hit Update on the other one without cleaning anything. In both cases it compiled fine.

michaelkleinhenz, 2025-02-20:
> [@michaelkleinhenz](https://github.com/michaelkleinhenz) I cleaned build files on the device that failed previously. But I just hit Update on the other one without cleaning anything. In both cases it compiled fine.

Interesting. I can't reproduce this. It is always breaking with the exact error above.

[2 more replies]